Koenkk/zigbee2mqtt #12319: Device Id TS110E Manufacturer name _TZ3210_ngqk6jia not supported error

Gibbz, 2022-04-30:
### What happened?

https://www.aliexpress.com/item/1005001937578720.html?spm=a2g0o.order_list.0.0.65131802WDLPWT
https://www.zigbee2mqtt.io/devices/TS110E_1gang.html

Ive added this device and it shows as not supported. How can i fix this?

### What did you expect to happen?

Device supported

### How to reproduce it (minimal and precise)

Add device

### Zigbee2MQTT version

1.25

### Adapter firmware version

20220219

### Adapter

zzh

Gibbz, 2022-04-30:
Other topics related
https://github.com/Koenkk/zigbee2mqtt/issues/8575

Koenkk, 2022-04-30:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-30 by Koenkk