Koenkk/zigbee2mqtt #12322: Zigbee2mqtt only connects when Sonoff USB dongle has the antenna removed

elporsche, 2022-04-30:
### What happened?

I bought a Sonoff USB zigbee dongle and I installed zigbee2mqtt on my RPi and I noticed that whenever I kept the antenna attached to the dongle, the zigbee-herdsman shows an error that there is an existing pan_id nearby, but as soon as I unscrewed the antenna, the error would no longer show and zigbee2mqtt would successfully start.

Even after the first configuration, the Sonoff Zigbee dongle can only recognize a ZIgbee peripheral when the antenna is removed; as soon as I attach the antenna, the dongle can no longer establish a connection to the peripheral.

### What did you expect to happen?

That the connection would be the same with and without the angenna

### How to reproduce it (minimal and precise)

I guess buy a Zigbee dongle and try

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee USB 3.0 Dongle

lux73, 2022-05-01:
that has nothing to do with your Sonoff Dongle or z2m - z2m probes for an existing pan_id, if so it throws this Error and refuse starting.

Either you have Main Powered Devices left from old Zigbee Network or other z2m User in Range uses the same pan_id as yours.

All you can do is change the pan_id in your Configuration - **but be aware, I'm not sure after you change it to reface your whole Zigbee Network!!**

elporsche, 2022-05-01:
> that has nothing to do with your Sonoff Dongle or z2m - z2m probes for an existing pan_id, if so it throws this Error and refuse starting.
>
> Either you have Main Powered Devices left from old Zigbee [...]

I have no other Zigbee devices (this was supposed to be the start of my Zigbee network). Changing the pan_id doesn't help, only detaching the antenna ensures a connection.

venditdevs, 2022-05-02:
It's impossible to detect an antenna or not for the ZigBee dongle and/or z2m 
If it's impossible to start z2m with this dongle WITH antenna i think there is a short in the antenna itself that's causing the problem.

I have the same Sonoff Dongle and it's working flawless. Used it as and drop in replacement for my CC2538

elporsche, 2022-05-04:
For future Googlers: the problem was indeed that I used a very short extension from the RPi to the Sonoff ZIgbee dongle. After buying a 3m USB extension and having the Dongle be as far away from the RPi/Wifi router as possible, I was able to connect to my Zigbee peripherals with the antenna connected so the problem disappeared.

closed 2022-05-04 by elporsche