Koenkk/zigbee2mqtt #12324: [New device support]:

goessensj, 2022-04-30:
### Link

https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html

### Database entry

{"id":90,"type":"Router","ieeeAddr":"0xa4c1385fbc2bad8b","nwkAddr":13222,"manufId":4417,"manufName":"_TZ3000_cehuw1lw","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","zclVersion":3}},"seMetering":{"attributes":{"currentSummDelivered":[0,4]}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2763076,"53251":121,"53252":10912,"53253":1302,"53264":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651349049030,"defaultSendRequestWhen":"immediate"}

### Comments

I bought some new TS011F (Tuya plugs) these are recognized as "TS011F", the ones i have in use at the moment are "TS011F_plug_1".

Koenkk, 2022-05-01:
Wast just added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4187

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-01 by Koenkk

goessensj, 2022-05-30:
Updated my Z2M docker container recently and the devices are not recognised anymore.