Koenkk/zigbee2mqtt #12326: [New device support]: TS0502B

dennis-bear, 2022-05-01:
### Link

https://item.taobao.com/item.htm?_u=1209dldgeg0c2c&id=660271936818

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0xa4c1387f71a67c90","nwkAddr":46942,"manufId":4417,"manufName":"_TZ3210_nvaik6gk","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502B","manufacturerName":"_TZ3210_nvaik6gk","powerSource":1,"zclVersion":3,"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651374600823,"defaultSendRequestWhen":"immediate"}

### Comments

i bought 4 of these LED dimmable & CCT driver lights. Did manged to pair but show unsupported device. i have been trying to create a custom new converter but unsuccessful.

[image]

Please help.
Do let me know if you need other information. i still learning :p

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

dennis-bear, 2022-05-01:
Hi, smulle 

Is there a way i can copy the file or what file should I copy and create a external converter to my zigbee2mqtt folder? 
I am still using ver 1.24 and don't dare to upgrade to 1.25 cos I afraid to break. I am not confident and don't have the knowledge to restore back if I do break. 
Thanks in advance

Koenkk, 2022-05-01:
Just merged https://github.com/Koenkk/zigbee-herdsman-converters/pull/4187

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-01 by Koenkk

dennis-bear, 2022-05-17:
Hi, 
i created a external converter:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {modelID: 'TS0502B', manufacturerName: '_TZ3210_nvaik6gk'},
    ],
    model: 'ZS-TY40-ZB-250-L',
    vendor: 'Tuya',
    description: 'Light controller',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 500]}),
};

module.exports = definition;

i noticed the colour temp is always opposite. Eg. i set 2700K my light will shine 6000K and when i set 6000K it will shine 2700K.
Sometimes, it not responsive and i also had this error in my log:

Zigbee2MQTT:error 2022-05-16 23:25:19: Publish 'set' 'state' to 'Living light front right ' failed: 'Error: Command 0xa4c138e9f320d6c3/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Zigbee2MQTT:error 2022-05-16 23:25:30: Publish 'set' 'brightness' to 'Living light back left' failed: 'Error: Command 0xa4c1387f71a67c90/1 genLevelCtrl.moveToLevelWithOnOff({"level":37,"transtime":20}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Zigbee2MQTT:error 2022-05-17 05:13:45: Publish 'set' 'state' to 'Living light front right ' failed: 'Error: Command 0xa4c138e9f320d6c3/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Do i need to add any other code to my external converter?
Hope you can help me. Thanks in advance