Koenkk/zigbee2mqtt #12328: [New device support]:

BashSer, 2022-05-01:
### Link

https://www.zigbee2mqtt.io/devices/ZTS-EU_3gang.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0024ccc71c","nwkAddr":23070,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":31,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1651398643790,"defaultSendRequestWhen":"immediate"} {"id":3,"type":"Router","ieeeAddr":"0xa4c13854bf989cf2","nwkAddr":39470,"manufId":4417,"manufName":"_TZE200_vm1gyrso","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":32,"65508":1,"appVersion":64}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651398625225,"defaultSendRequestWhen":"immediate"}

### Comments

Hello !
I has this switch(https://www.zigbee2mqtt.io/devices/ZTS-EU_3gang.html), but they cant be configure. I see many simillar switches and they has little difference in model TS0601 _TZE200_vm1gyrso\fjjbhx9d\oisqyl4o... Can be trouble in identify for all simillar switches ?
Can test and write feedback - what need for diagnose ?
Thank you !

Entry from db:
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0024ccc71c","nwkAddr":23070,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":31,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1651398643790,"defaultSendRequestWhen":"immediate"}

{"id":3,"type":"Router","ieeeAddr":"0xa4c13854bf989cf2","nwkAddr":39470,"manufId":4417,"manufName":"_TZE200_vm1gyrso","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":32,"65508":1,"appVersion":64}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651398625225,"defaultSendRequestWhen":"immediate"}

Link to product:
https://www.zigbee2mqtt.io/devices/ZTS-EU_3gang.html
https://a.aliexpress.com/_97OYsD

### External converter

```shell
Tried from  [New device support]: Moes ZS-EU, 1-3 gang wall switch dimmer #10813 , but dont work.
```

Koenkk, 2022-05-01:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-01 by Koenkk

BashSer, 2022-05-08:
Hello, Koenkk !
Possible add this device as light ? In this time i cant manipulate dimmer option =(

BashSer, 2022-05-09:
Good day, Koenkk !
I has many errors:
[image]

Logs with debug level:
```
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:14Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:14Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:14Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:14Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:14Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:14MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:15Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":3}],"seq":39424}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:20:15Received Zigbee message from 'Switch_3_Gang', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":39680}' from endpoint 1 with groupID 0
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":254,"level":1000,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:15MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
[... cut ...]
info 2022-05-09 19:20:17MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang', payload '{"brightness":253,"level":996,"linkquality":192,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2022-05-09 19:20:17MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l1', payload '{"state":"OFF"}'
info 2022-05-09 19:20:17MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l2', payload '{"state":"OFF"}'
info 2022-05-09 19:20:17MQTT publish: topic 'zigbee2mqtt/Switch_3_Gang/l3', payload '{"state":"OFF"}'
debug 2022-05-09 19:21:06Saving state to file /config/zigbee2mqtt/state.json
```

Works only turn on\off features without dimmer.
Thank you !

Koenkk, 2022-05-09:
Can you check with the code below? Errors will not be fixed but I'm wondering if the brightness will work:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    tuya_dimmer_level: {
        key: ['brightness_min', 'min_brightness', 'max_brightness', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            // upscale to 1000
            let newValue;
            let dp = tuya.dataPoints.dimmerLevel;
            if (['_TZE200_3p5ydos3', '_TZE200_9i9dt8is', '_TZE200_dfxkcots', '_TZE200_vm1gyrso'].includes(meta.device.manufacturerName)) {
                dp = tuya.dataPoints.eardaDimmerLevel;
            }
            if (key === 'brightness_min') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                    dp = tuya.dataPoints.dimmerLevel;
                } else {
                    throw new Error('Dimmer brightness_min is out of range 0..100');
                }
            } else if (key === 'min_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMinLevel;
                } else {
                    throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'max_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMaxLevel;
                } else {
                    throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'level') {
                if (value >= 0 && value <= 1000) {
                    newValue = Math.round(Number(value));
                } else {
                    throw new Error('Dimmer level is out of range 0..1000');
                }
            } else if (key === 'brightness_percent') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness_percent is out of range 0..100');
                }
            } else { // brightness
                if (value >= 0 && value <= 254) {
                    newValue = utils.mapNumberRange(value, 0, 254, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..254');
                }
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    }
}

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_vm1gyrso'}],
    model: 'ZS-EUD',
    vendor: 'Moes',
    description: '3 gang touch dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tzLocal.tuya_dimmer_state, tz.tuya_dimmer_level],
    exposes: [
        e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withEndpoint('l1'),
        e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withEndpoint('l2'),
        e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withEndpoint('l3'),
    ],
    onEvent: tuya.onEventSetLocalTime,
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;
```

BashSer, 2022-05-09:
After this zigbee2mqtt cant start with this repeats logs:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[19:32:58] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[19:33:00] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.1-dev start
> node index.js
(node:344) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:109:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:344) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:344) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[19:33:07] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.1-dev start
> node index.js
(node:413) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:109:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:413) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:413) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[19:33:15] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.1-dev start
> node index.js
```

After commented use external converter Zigbee2mqtt starts.

BashSer, 2022-05-09:
Next error and logs:
[image]
```

debug 2022-05-09 22:07:13Received MQTT message on 'zigbee2mqtt/Switch_3_Gang/set' with data '{"brightness_l1":11}'
debug 2022-05-09 22:07:13Publishing 'set' 'brightness' to 'Switch_3_Gang'
error 2022-05-09 22:07:13Publish 'set' 'brightness' to 'Switch_3_Gang' failed: 'ReferenceError: utils is not defined'
debug 2022-05-09 22:07:13ReferenceError: utils is not defined at Object.convertSet (/app/dist/util/externally-loaded.js:56:21) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:122:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (events.js:400:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1160:20) at Receiver.emit (events.js:400:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:528:14)
info 2022-05-09 22:07:13MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'Switch_3_Gang' failed: 'ReferenceError: utils is not defined'","meta":{"friendly_name":"Switch_3_Gang"},"type":"zigbee_publish_error"}'
debug 2022-05-09 22:07:38Received Zigbee message from 'Switch_3_Gang', type 'attributeReport', cluster 'genBasic', data '{"65506":32,"65508":1,"appVersion":64}' from endpoint 1 with groupID 0
debug 2022-05-09 22:07:39Received Zigbee message from 'Switch_3_Gang', type 'attributeReport', cluster 'genBasic', data '{"65506":32,"65508":1,"appVersion":64}' from endpoint 1 with groupID 0
debug 2022-05-09 22:07:39Received Zigbee message from 'Switch_3_Gang', type 'attributeReport', cluster 'genBasic', data '{"65506":32,"65508":1,"appVersion":64}' from endpoint 1 with groupID 0
debug 2022-05-09 22:07:39Received Zigbee message from 'Switch_3_Gang', type 'attributeReport', cluster 'genBasic', data '{"65506":32,"65508":1,"appVersion":64}' from endpoint 1 with groupID 0
debug 2022-05-09 22:07:39Received Zigbee message from 'Switch_3_Gang', type 'attributeReport', cluster 'genBasic', data '{"65506":32,"65508":1,"appVersion":64}' from endpoint 1 with groupID 0
```

Dimmer not working, but zigbee2mqtt start and turn on/off is works

Koenkk, 2022-05-10:
@BashSer updated https://github.com/Koenkk/zigbee2mqtt/issues/12328#issuecomment-1121409141 can you try again?

Koenkk, 2022-05-10:
Too bad it doesn't work, this means this device uses a different API than the other TuYa devices. You have to find out the data points to set the brightness, this can be done via the TuYa bridge. https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

BashSer, 2022-05-16:
Hi, @Koenkk  !

Can you watch this(https://github.com/zigpy/zha-device-handlers/issues/1302#issuecomment-1097291298 13 apr) and this (https://github.com/zigpy/zha-device-handlers/issues/1461#issuecomment-1090811884 07 apr)

May be in here has right endpoints ?

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaLevelControlManufCluster,
                    #TuyaOnOffMCU,
                    TuyaOnOffNM,
                    TuyaInWallLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    # TuyaOnOffMCU,
                    TuyaOnOffNM,
                    TuyaInWallLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    #TuyaOnOffMCU,
                    TuyaOnOffNM,
                    TuyaInWallLevelControlNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },

Sorry, i has not TuYa bridge for test. Only XIaomi Gateway 3

BashSer, 2022-05-28:
Sorry @Koenkk, but i cant turn on second and third gang. And i cant reset them to connect another zigbee bridge (sonoff usb zigbee). How i can reset them or connect to another device ?

Koenkk, 2022-05-29:
@BashSer I don't understand what you mean, can you clarify?

BashSer, 2022-05-30:
After many attempts delete dimmer from zigbee2mqtt - it finally went away and i can connect to new coordinator.
However, both on the old coordinator and on the new - only the first group works.
If I turn on the **first** group, then the **first** group is turned on.
If I turn on the **second** group, then the **first** group is turned on.
If I turn on the **third** group, then the **first** group is turned on.
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/8800041/logs.txt)

I tried different converters but the interface doesn't change. Everywhere displays the latest interface with adjustable brightness
I can't understand why only the first group works now. Of the changes - I reinstalled the zigbee2mqtt **edge** to a new version. Now is 1.25.1-dev commit: [502443a](https://github.com/Koenkk/zigbee2mqtt/commit/502443a)

Koenkk, 2022-05-31:
I see now, I have no clue why that happens. I hope someone else buys this device and can add proper support for it (I believe it needs some more reverse engineering via the TuYa gateway)

muxa, 2022-07-16:
I got this working. Not perfect, but does the job. Here's a custom converter: https://gist.github.com/muxa/94e29b8896c89da910e4e4a80fc44894 (code needs a cleanup)

It detects and controls state (on/off) and brightness for all 3 gangs.

Known issues:
1. If you change brightness while the light is off, then turn it on, the brightness will not change to the new value, and will remain the same as before switching off. It would be good to fix this, but I don't know how (I've tried a couple of things, I left the code it, commented out).

louismulder, 2022-08-28:
> I got this working. Not perfect, but does the job. Here's a custom converter: https://gist.github.com/muxa/94e29b8896c89da910e4e4a80fc44894 (code needs a cleanup)
> 
> It detects and controls state (on/off) and brightness for all 3 gangs.
> 
> Known issues:
> 
> 1. If you change brightness while the light is off, then turn it on, the brightness will not change to the new value, and will remain the same as before switching off. It would be good to fix this, but I don't know how (I've tried a couple of things, I left the code in, commented out).

Hi muxa,

I bought a Moes ZS-EUD 2 Gang Touch Dimmer and I have used your custom converter as base.
The issue I had was that ON/OFF was working.
But when I tried to use the brightness only and let it go on, it didn't.
So I have made some small changes in your code and now it is working perfect.
If the currectent State is OFF you should first give a state:ON before sending the brightness.
These are the changes I have made:

   tuya_dimmer_level: {
        key: ['brightness_min', 'min_brightness', 'max_brightness', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            //meta.logger.debug(`MOES TZ key=${key} value=${value} meta.endpoint_name=${meta.endpoint_name} meta.state=${JSON.stringify(meta.state)}`);

            // key=brightness value=128

            const lookup = {l1: 2, l2: 8};
            const dp = lookup[meta.endpoint_name];

            if (key == "brightness") {
                // upscale to 1000
                if (value >= 0 && value <= 254) {
                    newValue = utils.mapNumberRange(value, 0, 254, 0, 1000);
                    // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
		   if (meta.state[`state_${meta.endpoint_name}`] === 'ON') {                        
			await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
		  } else {
			await tuya.sendDataPoints(entity, [
				tuya.dpValueFromBool(dp-1, true),
				tuya.dpValueFromIntValue(dp, newValue),
			  ], 'dataRequest', 1);
		}
                } else {
                    throw new Error('Dimmer brightness is out of range 0..254');
                }
            } else {
                meta.logger.warn(`MOES TZ unsupported key=${key} value=${value}`);
            }            
        },
    },

helldog136, 2022-09-29:
Hi! I didn't like the "l1", "l2", "l3" designation so I Cleaned up a little the code and made this converter.

I suppose you could also control the backlight somehow but I haven't found how yet...

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ZSEUD: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: async (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case 1:
                    result.state_left = value ? 'ON' : 'OFF';
                    break;
                case 2:
                    result.brightness_left = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;
                case 7:
                    result.state_center = value ? 'ON' : 'OFF';
                    break;
                case 8:
                    result.brightness_center = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;
                case 15:
                    result.state_right = value ? "ON" : "OFF";
                    break;
                case 16:
                    result.brightness_right = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:ZSEUD: NOT RECOGNIZED DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
};

const tzLocal = {
    ZSEUD_state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            meta.logger.debug(`MOES TZ key=${key} value=${value} meta.endpoint_name=${meta.endpoint_name} meta.state=${JSON.stringify(meta.state)}`);
			
            const lookup = {left: 1, center: 7, right: 15};
            const dp = lookup[meta.endpoint_name];          
            await tuya.sendDataPointBool(entity, dp, value === 'ON');
        },
    },
    ZSEUD_brightness: {
        key: ['brightness_min', 'min_brightness', 'max_brightness', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            meta.logger.debug(`MOES TZ key=${key} value=${value} meta.endpoint_name=${meta.endpoint_name} meta.state=${JSON.stringify(meta.state)}`);
			
            const lookup = {left: 2, center: 8, right: 16};
            const dp = lookup[meta.endpoint_name];

            if (key == "brightness") {
                // upscale to 1000
                if (value >= 0 && value <= 254) {
                    newValue = utils.mapNumberRange(value, 0, 254, 0, 1000);
                    // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
                    await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..254');
                }
            } else {
                meta.logger.warn(`MOES TZ unsupported key=${key} value=${value}`);
            }            
        },
    },
};

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZS-EUD3', // Vendor model number, look on the device for a model number 
    vendor: 'Moes', // Vendor of the device (only used for documentation and startup logging)
    description: 'Moes 3 Gang Light Dimmer Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fzLocal.ZSEUD, fz.ignore_basic_report],
    toZigbee: [tzLocal.ZSEUD_brightness, tzLocal.ZSEUD_state],
    meta: {turnsOffAtBrightness1: true, multiEndpoint: true},
    onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
		await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
		if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
    exposes: [
        e.light_brightness().withEndpoint('left').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
        e.light_brightness().withEndpoint('center').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
        e.light_brightness().withEndpoint('right').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
    ],
    endpoint: (device) => {
        return {'left': 1, 'center': 1, 'right': 1};
    },
};

module.exports = definition;
```