Koenkk/zigbee2mqtt #12339: OTA updates hang and are not updated by Tuya TS0601_thermostat _TZE200_ckud7u2l

VikeDragon, 2022-05-01:
### What happened?

Greetings!
OTA updates hang and are not updated by Tuya TS0601_thermostat _TZE200_ckud7u2l.
Why are updates requested, but I get an error when updating
"Update of '0x....' failed (No new image available)"?
Please help me figure this out.
[image]

### What did you expect to happen?

OTA updates my thermostats

### How to reproduce it (minimal and precise)

Start Ota update

### Zigbee2MQTT version

1.25.0-1 

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (zStack3x0)

### Debug log

error 2022-05-01 23:28:34: Update of '0x....' failed (No new image available)

VikeDragon, 2022-06-04:
Help needed !

VikeDragon, 2022-08-05:
Not solved ...

VikeDragon, 2022-09-05:
Not solved ...

staastas, 2022-10-05:
Maybe this will help you https://github.com/Koenkk/zigbee2mqtt/issues/14345

VikeDragon, 2022-10-05:
Not solved ...

VikeDragon, 2022-11-05:
Not solved...

m0wlheld, 2022-12-09:
Still not addressed / solved. Columns Firmware Date and Firmware Version stay empty and requesting OTA check results in error (2 times):

> Failed to check if update available for '0x00....270c' (No image available for imageType '5634')

m0wlheld, 2023-01-09:
Still unresolved in 1.29.1 ... when checking for update, the error message is

> Failed to check if update available for '0x00....270c' (No image available for imageType '5634')

Koenkk, 2023-01-09:
We don't have an image for this device yet. To find out follow: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

m0wlheld, 2023-01-10:
Okay, to much effort for an update (for me). Unsubscribing from this issue. Thanks for the feedback.

closed 2023-02-18 by github-actions[bot]

florian-asche, 2023-09-05:
just for documentation: i checked tuya for updates, there is currently nothing new.