Koenkk/zigbee2mqtt #12340: "homeassistant: null" makes z2m fail to start

okastl, 2022-05-01:
### What happened?

After updating from z2m 1.25 to 1.25.1 it fails to start:

```
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY

Refusing to start because configuration is not valid, found the following errors:
- devices/0xb4e3f9fffe19f03b/homeassistant must be object
- devices/0x00124b00226babf1/homeassistant must be object

If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```

The offending setting is `homeassistant: null` as described here: https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html

`Set homeassistant: null to skip discovery for this device
`

This no longer works.

### What did you expect to happen?

"homeassistant: null" should continue to work.

### How to reproduce it (minimal and precise)

Stop z2m.
Add `homeassistant: null` to a device.
Start z2m.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

Slaesh

65698888, 2022-05-02:
Zigbee2mqtt configuration - Save, click OK and restart the add-in to return to normal.

okastl, 2022-05-02:
> Zigbee2mqtt configuration - Save, click OK and restart the add-in to return to normal.

Thanks, but this won't make z2m 1.25.1 recognize `homeassistant: null` parameter.
https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options

mdehaas, 2022-05-02:
Tried to replace 'null' with '{}' (an empty object) for the devices I do not want discovered in HA. This clears the error but the devices are discovered in HA.

Koenkk, 2022-05-02:
Can you provide your complete `configuration.yaml`?

okastl, 2022-05-02:
> Can you provide your complete `configuration.yaml`?

I made a simple example from my test installation.

[configuration.zip](https://github.com/Koenkk/zigbee2mqtt/files/8605192/configuration.zip)

Soyde, 2022-05-02:
Hi, same problem here.

I have fixed changing:
Before:   homeassistant: null
After:   homeassistant: {}

Complete device configuration after fix, like follows:

'0x00123c0001234d5f':
  friendly_name: Salón Árbol Navidad
  homeassistant: {}

Hope it helps anyone.
Regards!

okastl, 2022-05-02:
> 
> I have fixed changing: Before: homeassistant: null After: homeassistant: {}
> 

Unfortunately this isn't the same. `homeassistant: {}` will not skip adding the device to homeassistant discovery, as `homeassistant: null` did.

Soyde, 2022-05-02:
> > I have fixed changing: Before: homeassistant: null After: homeassistant: {}
> 
> Unfortunately this isn't the same. `homeassistant: {}` will not skip adding the device to homeassistant discovery, as `homeassistant: null` did.

But at least it solves the problem, Zigbetomqtt starts, and everything works again.
Then, you can edit the device, and disable it at HomeAssistant.

[image]

mdehaas, 2022-05-03:
> > I have fixed changing: Before: homeassistant: null After: homeassistant: {}
> 
> Unfortunately this isn't the same. `homeassistant: {}` will not skip adding the device to homeassistant discovery, as `homeassistant: null` did.

That is indeed what I stated in the next sentence. It is a workaround to get Z2M to start. I could also have commented the line. The effect would be the same.

mdehaas, 2022-05-03:
> Can you provide your complete `configuration.yaml`?

The issue is in the devices.yaml file as I think you have understood. Just to be sure. 

`external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  user: addons
  password: RemovedForPosting
  server: mqtt://core-mosquitto:1883
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_26_AB-if00-port0
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - removed
    - for
    - posting
  availability_blocklist: []
  availability_passlist: []
  ikea_ota_use_test_url: true
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: false
`

Koenkk, 2022-05-03:
Fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

okastl, 2022-05-03:
> Fixed now.
> 

Thank you!

closed 2022-05-03 by okastl

gado070707, 2022-12-12:
Fixed for me too !
Thank you.