Koenkk/zigbee2mqtt #12341: Aurora Lighting  AU-A1ZBDSS - Double Socket - Change Brightness setting false socket state

MJWMJW2, 2022-05-02:
### What happened?

Setting brightness to zero sets a false off state for the left socket (actual socket stays on if on)
Increasing brightness from zero to any value other than 1 sets a false on state for the left socket
Increasing brightness from any value >=1 sets a false on state for the left socket

The physical socket stays in the existing state - only the UI and home assistant report the new state incorrectly.

### What did you expect to happen?

The brightness should change, the left socket should not be getting a false change 

### How to reproduce it (minimal and precise)

With brightness set  100 and both sockets on, decrease brightness to zero in Zigbee2MQTT UI  - left socket will appear to turn off.

With brightness set to 0 and sockets turned off, increase brightness to 100, left socket will appear to turn on

With brightness set to 100, and sockets off, increase brightness to 200, left socket will appear to turn on

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20211217

### Adapter

Electrolama ZZH

### Debug log

#Brightness at zero in UI both sockets really off, UI both off

#refresh brightness
Debug 2022-05-02 00:18:52Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/get' with data '{"brightness":""}'
Debug 2022-05-02 00:18:52Publishing get 'get' 'brightness' to 'DoubleSocketTest'
Debug 2022-05-02 00:18:52Received Zigbee message from 'DoubleSocketTest', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0

#Refresh left
Debug 2022-05-02 00:19:29Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/get' with data '{"state_left":""}'
Debug 2022-05-02 00:19:29Publishing get 'get' 'state' to 'DoubleSocketTest'
Debug 2022-05-02 00:19:30Received Zigbee message from 'DoubleSocketTest', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-05-02 00:19:30MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":0,"brightness_left":1,"linkquality":58,"power_left":0,"power_right":0,"state":"OFF","state_left":"OFF","state_right":"OFF"}'

#Refresh right
Debug 2022-05-02 00:19:46Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/get' with data '{"state_right":""}'
Debug 2022-05-02 00:19:46Publishing get 'get' 'state' to 'DoubleSocketTest'
Debug 2022-05-02 00:19:47Received Zigbee message from 'DoubleSocketTest', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Info 2022-05-02 00:19:47MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":0,"brightness_left":1,"linkquality":58,"power_left":0,"power_right":0,"state":"OFF","state_left":"OFF","state_right":"OFF"}'

#Both Sockets really off

#Set brightness up from 0
Debug 2022-05-02 00:20:44Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/set' with data '{"brightness":88}'
Debug 2022-05-02 00:20:44Publishing 'set' 'brightness' to 'DoubleSocketTest'
Info 2022-05-02 00:20:44MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":88,"brightness_left":1,"linkquality":61,"power_left":0,"power_right":0,"state":"ON","state_left":"OFF","state_right":"OFF"}'
Debug 2022-05-02 00:20:45Received Zigbee message from 'DoubleSocketTest', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-05-02 00:20:45MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":88,"brightness_left":1,"linkquality":54,"power_left":0,"power_right":0,"state":"ON","state_left":"ON","state_right":"OFF"}'
Debug 2022-05-02 00:20:53Received Zigbee message from 'DoubleSocketTest', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
Info 2022-05-02 00:20:53MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":88,"brightness_left":1,"linkquality":65,"power_left":0,"power_right":0,"state":"ON","state_left":"ON","state_right":"OFF"}'

#Both are still actually off, UI changes to show left socket on.

#Set brightness to zero
Debug 2022-05-02 00:22:22Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/set' with data '{"brightness":0}'
Debug 2022-05-02 00:22:22Publishing 'set' 'brightness' to 'DoubleSocketTest'
Info 2022-05-02 00:22:22MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":0,"brightness_left":1,"linkquality":54,"power_left":0,"power_right":0,"state":"OFF","state_left":"ON","state_right":"OFF"}'
[... 63 lines cut ...]
#Result - false set of left socket to on (socket is  actually off)

#Decrease brightness
Debug 2022-05-02 00:46:19Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/set' with data '{"brightness":100}'
Debug 2022-05-02 00:46:19Publishing 'set' 'brightness' to 'DoubleSocketTest'
Info 2022-05-02 00:46:19MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":100,"brightness_left":1,"linkquality":69,"power_left":0,"power_right":0,"state":"ON","state_left":"ON","state_right":"OFF"}'

#Refresh left
Debug 2022-05-02 00:51:57Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/get' with data '{"state_left":""}'
Debug 2022-05-02 00:51:57Publishing get 'get' 'state' to 'DoubleSocketTest'
Debug 2022-05-02 00:51:57Received Zigbee message from 'DoubleSocketTest', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-05-02 00:51:57MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":108,"brightness_left":1,"linkquality":58,"power_left":0,"power_right":0,"state":"ON","state_left":"ON","state_right":"OFF"}'

#Correct left by setting to off
Debug 2022-05-02 00:52:30Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/set' with data '{"state_left":"OFF"}'
Debug 2022-05-02 00:52:30Publishing 'set' 'state' to 'DoubleSocketTest'
Info 2022-05-02 00:52:30MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":108,"brightness_left":1,"linkquality":58,"power_left":0,"power_right":0,"state":"ON","state_left":"OFF","state_right":"OFF"}'
Debug 2022-05-02 00:52:30Received Zigbee message from 'DoubleSocketTest', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-05-02 00:52:30MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":108,"brightness_left":1,"linkquality":58,"power_left":0,"power_right":0,"state":"ON","state_left":"OFF","state_right":"OFF"}'

#Refresh left
Debug 2022-05-02 00:52:58Received MQTT message on 'zigbee2mqtt/DoubleSocketTest/get' with data '{"state_left":""}'
Debug 2022-05-02 00:52:58Publishing get 'get' 'state' to 'DoubleSocketTest'
Debug 2022-05-02 00:52:58Received Zigbee message from 'DoubleSocketTest', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-05-02 00:52:58MQTT publish: topic 'zigbee2mqtt/DoubleSocketTest', payload '{"brightness":108,"brightness_left":1,"linkquality":54,"power_left":0,"power_right":0,"state":"ON","state_left":"OFF","state_right":"OFF"}'

MJWMJW2, 2022-05-02:
@karlc28
Are you able to replicate this?

MJWMJW2, 2022-05-02:
Having written all the above and having a really good read through the debug, my feeling is this is a device issue.  I've got aone hub, I'll test on that and report back.

MJWMJW2, 2022-05-02:
Connected devices to manufacturers hub and behaviour is the same - increasing brightness, then reports left socket is turned on (when it stays off).  There is no obvious way to set to off in their app, so I can only test in one direction.  I am contacting the manufacturer and will mark this as resolved.

karlc28, 2022-05-02:
I can confirm the same happens with mine

MJWMJW2, 2022-05-04:
I've had a reply from Aurora who confirmed the issue,

Denis (Aurora Lighting) May 4, 2022, 11:31 AM GMT+1 Hello, We really appreciate your feedback on this issue and the work yourself and the wider Home Assistant community have undertaken to provide us with this information. 

We believe that the Home Assistant platform is using the movetolevelwithonoff Zigbee Command.If you try and switch to the movetolevel Zigbee Command that should resolve this issue. 

We have tested this fix on our platform and can confirm it has resolved the issue.
--

@Koenkk as you recently added this feature at the end of  #11381 is it something you know about?

Koenkk, 2022-05-04:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    backlight_brightness: {
        key: ['brightness'],
        options: [exposes.options.transition()],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('genLevelCtrl', 'moveToLevel', {level: value, transtime: 0}, utils.getOptions(meta.mapped, entity));
            return {state: {brightness: value}}
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genLevelCtrl', ['currentLevel']);
        },
    },
}

const definition = {
    zigbeeModel: ['DoubleSocket50AU'],
    model: 'AU-A1ZBDSS',
    vendor: 'Aurora Lighting',
    description: 'Double smart socket UK',
    fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement, fz.brightness],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'),
        e.power().withEndpoint('left'), e.power().withEndpoint('right'),
        exposes.numeric('brightness', ea.ALL).withValueMin(0).withValueMax(254)
            .withDescription('Brightness of this backlight LED')],
    toZigbee: [tzLocal.backlight_brightness],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint2);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

MJWMJW2, 2022-05-04:
I cannot say for sure,
With external converter the brightness will change, and the switch does not move with any of the previous tests

However attempting to change the state of the switch within z2m gives 
```
2022-05-04 22:12:41No converter available for 'state' (undefined)
```
Changing state on the physical device is updated correctly.

Koenkk, 2022-05-05:
Can you try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    backlight_brightness: {
        key: ['brightness'],
        options: [exposes.options.transition()],
        convertSet: async (entity, key, value, meta) => {
            await entity.command('genLevelCtrl', 'moveToLevel', {level: value, transtime: 0}, utils.getOptions(meta.mapped, entity));
            return {state: {brightness: value}}
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genLevelCtrl', ['currentLevel']);
        },
    },
}

const definition = {
    zigbeeModel: ['DoubleSocket50AU'],
    model: 'AU-A1ZBDSS',
    vendor: 'Aurora Lighting',
    description: 'Double smart socket UK',
    fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement, fz.brightness],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'),
        e.power().withEndpoint('left'), e.power().withEndpoint('right'),
        exposes.numeric('brightness', ea.ALL).withValueMin(0).withValueMax(254)
            .withDescription('Brightness of this backlight LED')],
    toZigbee: [tzLocal.backlight_brightness, tz.on_off],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint2);
    },
};

module.exports = definition;
```

Koenkk, 2022-05-06:
Cool, integrated the changes.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-06 by Koenkk

karlc28, 2022-06-17:
@MJWMJW2 after the last changes is your power monitering working correctly

MJWMJW2, 2022-06-17:
> @MJWMJW2 after the last changes is your power monitering working correctly

Yes, all working here, I have just added another that one of the ACs plugs in to as well (so 4 in live service).  

I have found though a about 3-4 times that they have stopped reporting power at all for some reason (one individual device each time).  Now I did all sorts of things trying to getting it working, and I can't remember finally if I, force removed and re-added, hit to repair without removing, or just did a re-interview, either way I managed to get them all to work again.

karlc28, 2022-06-17:
> Yes, all working here, I have just added another that one of the ACs plugs in to as well (so 4 in live service).
>
>
>
> I have found though a about 3-4 times that they have stopped reporting power at [...]

Had to force remove and repair but all working now

MJWMJW2, 2023-11-30:
> > @MJWMJW2 after the last changes is your power monitering working correctly
> 
> Can anyone else with these sockets please confirm their power accuracy? I've seen information online suggesting they can do accuracy to 1 decimal place but I'm only seeing round numbers. The settings under specific settings don't seems to change anything.

Accurate enough, but by no means exact so would be daft to be down to 1/10th watt.