Koenkk/zigbee2mqtt #12344: Aqara plug SP-EUC01 doesn't report current and voltage

mrskycriper, 2022-05-02:
### What happened?

[image]
After the newest update i saw that SP-EUC01 should now expose current and voltage characteristics. However it doesn't report their values at all.

### What did you expect to happen?

I expected to see current and voltage values being reported.

### How to reproduce it (minimal and precise)

Just open device properties and see.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee 3.0 usb dongle plus

Koenkk, 2022-05-02:
Might be fixed by updating your coordinator to the latest firmware and repairing the device.

reyhard, 2022-05-07:
I had same issue and I can confirm that after repairing devices, voltage and current are properly reported. I'm using newer coordinator firmware though ( 20220219 )

mrskycriper, 2022-05-09:
I've checked for the coordinator firmware updates and it seems that mine has way too old version right now. I'll have to figure out how to update it without loosing any of my devices. But I hope this will fix the issue.

mrskycriper, 2022-05-17:
> I had same issue and I can confirm that after repairing devices, voltage and current are properly reported. I'm using newer coordinator firmware though ( 20220219 )
Yeah, that worked for me as well. Had to re-pair both of my plugs after flashing newer firmware. But now everything seems to be working as intended.

closed 2022-05-17 by mrskycriper

shrx, 2024-11-03:
Same problem with Z2M 1.41.0 and CC2531 with firmware revision 20211116.