Koenkk/zigbee2mqtt #12349: Supported device initially recognized as unsupported

michalk-k, 2022-05-02:
### What happened?

When adding a new device it very often happens that it's initially recognized as unsupported.
Then after some time (from seconds to minutes) it turns to 'supported' on it's own. Because this operation may take minutes, it lefts an operator in doubt if the operation will ever finish with success.

### What did you expect to happen?

Is it expected behavior?

Is it possible to change the wording of this transitive state to something more informative? For example: 'pending' or something like this

with regards

### How to reproduce it (minimal and precise)

Add any new device (must not behave the same in every case)

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

 0x26700700

### Adapter

 ConBee2

Koenkk, 2022-05-02:
It depends on the device, some TuYa device require pairing multiple times (bug on the device side)

Koenkk, 2022-05-06:
You can see if the interview of a device failed in the frontend (if interview fails you should try pairing it again).

[image]

michalk-k, 2022-05-27:
Yes, you are right.
But as a user, I get the first information about whether a device is supported or not watching device list. And this list does not provide information about interview state. Just about device is not supported

[image]

Koenkk, 2022-05-27:
The warning sign on the left indicates the interview failed.

michalk-k, 2022-05-27:
Ok, it looks like indicating any problem, not only

Koenkk, 2022-05-28:
When the device is being interviewed (which always happens after joining) the device icon will show a round progress indicator.

closed 2022-07-06 by github-actions[bot]