Koenkk/zigbee2mqtt #12350: ZLinky configuration error

Tolriq, 2022-05-02:
### What happened?

ZLinky fails at configuration and does not return data.

Firmware build date: 20220217
Firmware version: 4000-0005
Manufacturer: [LiXee](https://www.zigbee2mqtt.io/supported-devices/#v=LiXee)
Model: [ZLinky_TIC](https://www.zigbee2mqtt.io/devices/ZLinky_TIC.html#lixee-zlinky_tic)

Failed to configure 'Linky', attempt 3 (Error: Bind 0x00158d0005ce7a37/1 seMetering from '0x00212effff06d3e7/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/lixee.js:598:9) at Configure.configure (/app/lib/extension/configure.ts:115:13))

### What did you expect to happen?

No error and working.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x266b0700

### Adapter

ConBee2/RaspBee2

### Debug log

Failed to configure 'Linky', attempt 3 (Error: Bind 0x00158d0005ce7a37/1 seMetering from '0x00212effff06d3e7/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/lixee.js:598:9) at Configure.configure (/app/lib/extension/configure.ts:115:13))

Tolriq, 2022-05-02:
Restart all fixed it sorry for noise.

closed 2022-05-02 by Tolriq

Phyks, 2023-01-14:
I got the exact same issue upon upgrading from 1.24.0 to 1.29.2 just now.

Traceback is:

```
janv. 14 10:27:38 domotique npm[687]: Zigbee2MQTT:error 2023-01-14 10:27:38: Failed to configure 'Linky', attempt 1 (Error: Bind 0x00158d0005d28561/1 seMetering from '0x00212effff07e62e/1' failed (waiting for response TIMEOUT)
janv. 14 10:27:38 domotique npm[687]:     at DeconzAdapter.bind (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19)
janv. 14 10:27:38 domotique npm[687]:     at Endpoint.bind (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
janv. 14 10:27:38 domotique npm[687]:     at Object.bind (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
janv. 14 10:27:38 domotique npm[687]:     at Object.configure (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/lixee.js:598:9)
janv. 14 10:27:38 domotique npm[687]:     at Configure.configure (/home/pi/zigbee2mqtt/lib/extension/configure.ts:117:13)
janv. 14 10:27:38 domotique npm[687]:     at Immediate.<anonymous> (/home/pi/zigbee2mqtt/lib/extension/configure.ts:69:17))
``` 

I'm running zigbee2mqtt with a ConnBee 2. Restarting/rebooting does not change the result, I consistently get this error traceback in the logs. This being said, it seems it does not affect normal operation since the exposed values by the ZLinky are getting updated.

Sigri44, 2023-03-18:
Same error for me :
```
error 2023-03-18 12:09:50: Failed to configure 'ZLinky', attempt 2 (TypeError: Promise.allSettled is not a function                   │    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/lixee.js:617:24))     
```

traylenator, 2024-11-17:
> Same error for me :
> 
> ```
> error 2023-03-18 12:09:50: Failed to configure 'ZLinky', attempt 2 (TypeError: Promise.allSettled is not a function                   │    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/lixee.js:617:24))     
> ```

Also seeing this also with a conbee 2