Koenkk/zigbee2mqtt #12352: unstable work and delay from devices

KaportsevIA, 2022-05-02:
### What happened?

After the latest update 1.25.1.1, there were delays when sending commands and statuses from devices via ZigBee. Delays can be up to 5-6 seconds, or the device will simply not respond.

> Error 2022-05-02 17:39:56Publish 'set' 'state' to '0x04cf8cdf3c8e0d39' failed: 'Error: Command 0x04cf8cdf3c8e0d39/1 genOnOff.off({}, {"sendWhen":"immediate","timeout": 10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null, "writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
> Error 2022-05-02 17:40:03Publish 'set' 'state' to '0x54ef4410001c649b' failed: 'Error: Command 0x54ef4410001c649b/2 genOnOff.off({}, {"sendWhen":"immediate","timeout": 10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null, "writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'

### Zigbee2MQTT version

1.25.1.1

### Adapter firmware version

20220219

### Adapter

Sonoff ZStack CC2652P

KaportsevIA, 2022-05-02:
Things got even worse. A lot of errors, and the well-known Aqara Wall Switch devices (H1 EU, QBKG04LM, SP-EUC01) began to fall off the network.

> Error 2022-05-02 19:05:42Failed to read state of '0x00158d00045f20ab' after reconnect (Read 0x00158d00045f20ab/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:05:45Failed to interview '0x54ef4410001c649b', device has not successfully been paired
> Error 2022-05-02 19:05:51Failed to read state of '0x00158d00053eef8f' after reconnect (Read 0x00158d00053eef8f/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:06:34Failed to read state of '0x00158d00053f2af0' after reconnect (Read 0x00158d00053f2af0/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:07:24Failed to read state of '0x00158d00045e30ff' after reconnect (Read 0x00158d00045e30ff/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:07:29Failed to read state of '0x00158d00053f2a59' after reconnect (Read 0x00158d00053f2a59/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:08:11Failed to read state of '0x00158d00053eee28' after reconnect (Read 0x00158d00053eee28/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:11:00Failed to read state of '0x00158d00053eef8f' after reconnect (Read 0x00158d00053eef8f/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:11:15Failed to read state of '0x00158d00045f20ab' after reconnect (Read 0x00158d00045f20ab/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:11:43Failed to read state of '0x00158d00053f2af0' after reconnect (Read 0x00158d00053f2af0/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
> Error 2022-05-02 19:12:10Failed to read state of '0x54ef4410001972d9' after reconnect (Read 0x54ef4410001972d9/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
> Error 2022-05-02 19:12:33Failed to read state of '0x54ef4410001d0b06' after reconnect (Read 0x54ef4410001d0b06/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
> Error 2022-05-02 19:12:38Failed to read state of '0x00158d00053f2a59' after reconnect (Read 0x00158d00053f2a59/2 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultRe

KaportsevIA, 2022-05-02:
See in pics
[image]

swests, 2022-05-03:
Seeing the same here. Since the upgrade devices simply timeout

[image]

swests, 2022-05-03:
Is there a process to roll back to a previous version?

Tediore, 2022-05-06:
I had a similar issue randomly pop up after a week or so of being on 1.25. No network route sometimes, but most frequently `SRSP - AF - DataRequest after 6000ms` error. It happened with both my ZZH and Sonoff Zigbee 3.0 stick (the new one with the black case) on various firmware versions, including the February 2022 revision. 

From what I've read, that DataRequest error typically means the adapter is locked up. Restarting z2m and/or unplugging the adapter and plugging it back in would fix the issue temporarily, but usually within an hour it would happen again. Rolled back to 1.24 and no longer had any issues.

I commissioned a test zigbee network on a separate zigbee2mqtt 1.25 instance and connected some devices to it. Just waiting to see if the same thing happens. If it does I'll provide some logs.

avdgaag1984, 2022-05-07:
I have also issues after the update 1.25.1 
Have already had 2 evenings in a row that suddenly my lamps no longer turn on, motion detectors no longer work and door sensors that no longer work. I then have to turn off my virtual machine running on my synology and turn it on again. Then everything works again.

I have a zigbee stick: cc2652rb with last coordinator firmware. And i don't use a extension cable. It is directly connected to my synology nas

This says the log: 
error 2022-05-06 21:39:16: Publish 'set' 'state' to 'Kantoor' failed: 'Error: Command 3 genOnOff.on({}) failed (SRSP - AF - dataRequestExt after 6000ms)'
info  2022-05-06 21:39:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Kantoor' failed: 'Error: Command 3 genOnOff.on({}) failed (SRSP - AF - dataRequestExt after 6000ms)'","meta":{"friendly_name":"Kantoor"},"type":"zigbee_publish_error"}'
info  2022-05-06 21:39:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'

Maybe i am going put a backup from last weekend, than i have version 1.25.0

netweaver1970, 2022-05-07:
Same issue here, happened last night. It was when a bunch of zigbee commands was send. Whole network dead with same error. Restart of the add-on fixed it. My sonoff zigbee plus usb adapter is connected with extention lead

3skelion, 2022-05-26:
The same problem, Zigbee devices work with a large

closed 2022-07-03 by github-actions[bot]

Armageddit, 2022-07-12:
same problem

Zigbee2MQTT Version [1.26.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.26.0) commit: [unknown]
Coordinator Typ
zStack3x0
Coordinator Version
20220219

Sonoff Zigbee3.0 stick (Black Alu case)

32 divices of 56 have the problem!
the other (endpoints) works normal 

LOG:
```
Warning 2022-07-12 23:32:14Failed to ping 'Licht Bad' (attempt 1/1, Read 0x842e14fffe13cd6d/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
Warning 2022-07-12 23:32:19Failed to ping 'Power Strip' (attempt 1/1, Read 0x60a423fffe802c3f/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
Warning 2022-07-12 23:32:24Failed to ping 'Blitzwolf Steckdose Spülmaschine' (attempt 1/1, Read 0xa4c1384447f9d808/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
Warning 2022-07-12 23:32:34Failed to ping 'Steckdose außen ' (attempt 1/1, Read 0xa4c13850ac591485/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
```

sk8er000, 2022-09-04:
I'm having the same random problem from Moes MS-104BZ and Lonsonho QS-Zigbee-S05-LN. What I noticed is that when this error occurs the device is not directly connected with the coordinator (they don't have any end device connected since they are connected with IKEA and Aqara routers). If I repair the module again it start working again for a random amount of time, it can be hours or days. Unfortunately it's not easy to find any other 2 gang switch with the same dimesions (or smaller) that are know to always work

Actually I'm using the SONOFF USB 3.0 plus with the lastes zstack firmware and Z2M is the 1.27.2-1

aharoncg, 2022-12-05:
Hi there,

I have the same problem since I moved from deconz/phoscon to z2m.

Devices are loosing connection randomly, house residents are losing patience... Once the device is suddenly popping up its state changes. It causes the light to suddenly turn on or off randomly and without sender any command. The whole network/technology is unstable and unreliable.

It happened with both conbee2 and sonoff 3 (cc2652p). I tried with different types of extension cables, of different ports (usb2/usb2)

My setup is bare metal installation on a standard x86 mini pc. It is a fresh setup after it happens with both docker on ubuntu ans bare metal on rpi4.

Zigbee is using channel 11, wifi also 11 - veey fat from eachother.
-----
Zigbee2MQTT version
1.28.4 commit: unknown

Coordinator type
ConBee2/RaspBee2

Coordinator revision
0x26580700

Frontend version
0.6.119

Total 76 devices ate connected (I have 112 but removed some to try reducing the load - it didn't help)

Two dedicated ikea router, rest are sensors and lights from sonoff, aqara, tuya, lidl, ledvance, osram, awox, eglo, and more.

Zigbee2MQTT version
[1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
ConBee2/RaspBee2
Coordinator revision
0x26580700
Coordinator IEEE Address
0x00212effff0973d9
Frontend version
0.6.119
Stats
Total 76
By device type
End devices: 47
Router: 29
By power source
Battery: 47
Mains (single phase): 29
By vendor
LUMI: 32
_TZ1800_fcdjzz3s: 10
_TZ3000_tqlv4ug4: 8
AwoX: 7
LEDVANCE: 4
OSRAM: 4
_TZE200_ntcy3xu1: 3
IKEA of Sweden: 2
_TZ3210_p9ao60da: 2
eWeLink: 2
_TZ3210_umi6vbsz: 1
_TZ3000_zmy4lslw: 1
By model
TY0202: 10
lumi.sensor_magnet: 9
TS0001: 8
lumi.weather: 6
EGLO_ZM_TW: 5
A60 DIM Z3: 4
lumi.remote.b1acn01: 4
lumi.sensor_magnet.aq2: 3
TS0601: 3
lumi.sensor_motion.aq2: 3
TS0505B: 3
Plug 01: 2
lumi.sensor_ht: 2
lumi.sensor_wleak.aq1: 2
TRADFRI signal repeater: 2
MS01: 2
lumi.remote.b286acn01: 2
lumi.vibration.aq1: 1
CLA60 TW OSRAM: 1
TLSR82xx: 1
EBF_RGB_Zm: 1
CLA60 RGBW OSRAM: 1
TS0002: 1

I am lost.
Please help.