Koenkk/zigbee2mqtt #12357: `Missing command '104'` error when trying to pair an `EnOcean PTM 215Z` device

Grandma-Betty, 2022-05-03:
### What happened?

I have three different switches, all based on the `EnOcean PTM 215Z` Green Power switch.
When trying to pair the device according to the [official Z2M instructions](https://www.zigbee2mqtt.io/devices/PTM_215Z.html), the device is indeed being paired but whenever I try to confirm the channel afterwards as described on the following screenshot...

[image]

... I get the following error:

[image]

I have double checked: The device is positioned the right way up so it's not that I press the wrong buttons.
Any idea what the problem could be? 

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

0x26720700

### Adapter

Conbee II

Koenkk, 2022-05-03:
Besides this error, does the device work correctly? The only thing this error means is that this won't be published as an `action` to MQTT.

Grandma-Betty, 2022-05-08:
@Koenkk Sorry for being late with my anser but as I was planning to jump over from the Conbee II to the Sonoff V3 USB gateway, which I did now, I wanted to wait to see if this error still exists, which it indeed does.
To answer your question: Yes, the device could be paired successfully with the Conbee II and automations with those devices did indeed work initially, but somehow all of the three Green Power switches did not work anymore after an arbitrary amount of time (I was not able to turn on/off the lights by pressing the Green Power switch buttons anymore).

As I did read [here](https://www.reddit.com/r/homeassistant/comments/q7eekk/conbee_ii_zigbee_woes_im_really_getting_desperate/) that the Conbee II has a somewhat crappy behaviour with Z2M compared to the Sonoff V3, I decided to give it a shot and ordered one.
So what can I say? Oh boy, what a much(!) better experience I made with the Sonoff V3 according to Z2M too!

Anyway, I guess this is out of topic.
To conclude: I thought the error message I opened this issue ticket for may also be the reason the Green Power switches not being reachable after an amount of time, but your answer that the error message is only `action` related made it clear to me that it has nothing to do with it. After jumping to the Sonoff V3 since yesterday, the Green Power Switches obviously are still getting the same error message but they still work reliable within my automations so I guess it's just the difference between the Conbee II and the Sonoff V3.

Thank you anyway for your investigation mate, much appreciated!

closed 2022-05-08 by Grandma-Betty