Koenkk/zigbee2mqtt #12361: tuya ts130f -  _TZ3000_4uuaja4a shutter curtain zigbee

pieroprofita, 2022-05-03:
### What happened?

Hi all
i have two of these https://www.aliexpress.com/item/1005003012298844.html . they are tuya and are known by Z2M as ts130f . 

From scratch they works fine, but after few hours they began not managble
I can see it in Z2M but when i push ( on off etc ) nothing happen . I have to re-pairing it to take the control of them.. but after a few hours ( more of 10 ) they began not manageable . 

It is not a problem of signal, i have tested it near the coordinator also ... 

regards 
Luigi 

### What did you expect to happen?

it works always 

### How to reproduce it (minimal and precise)

I can see it in Z2M but when i push ( on off etc ) nothing happen . I have to re-pairing it to take the control of them.. but after a few hours they began not manageable .  I have tested it many times in many days. 

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

electrolama zig-zag

### Debug log

Zigbee2MQTT:debug 2022-05-03 13:32:08: Received MQTT message on 'zigbee2mqtt/tapparella cucina /set' with data '{"state":"CLOSE"}'
Zigbee2MQTT:debug 2022-05-03 13:32:08: Publishing 'set' 'state' to 'tapparella cucina '
Zigbee2MQTT:debug 2022-05-03 13:32:09: Received MQTT message on 'zigbee2mqtt/tapparella cucina /set' with data '{"state":"OPEN"}'
Zigbee2MQTT:debug 2022-05-03 13:32:09: Publishing 'set' 'state' to 'tapparella cucina '
Zigbee2MQTT:debug 2022-05-03 13:32:10: Saving state to file /share/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2022-05-03 13:32:10: Received MQTT message on 'zigbee2mqtt/tapparella cucina /set' with data '{"state":"CLOSE"}'
Zigbee2MQTT:debug 2022-05-03 13:32:10: Publishing 'set' 'state' to 'tapparella cucina '
Zigbee2MQTT:debug 2022-05-03 13:32:11: Received MQTT message on 'zigbee2mqtt/tapparella cucina /set' with data '{"state":"STOP"}'
Zigbee2MQTT:debug 2022-05-03 13:32:11: Publishing 'set' 'state' to 'tapparella cucina '
Zigbee2MQTT:error 2022-05-03 13:32:19: Publish 'set' 'state' to 'tapparella cucina ' failed: 'Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.downClose({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2022-05-03 13:32:19: Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.downClose({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:390:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Zigbee2MQTT:error 2022-05-03 13:32:31: Publish 'set' 'state' to 'tapparella cucina ' failed: 'Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.upOpen({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2022-05-03 13:32:31: Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.upOpen({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:390:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2022-05-03 13:32:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'tapparella cucina ' failed: 'Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.upOpen({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"tapparella cucina "},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2022-05-03 13:32:43: Publish 'set' 'state' to 'tapparella cucina ' failed: 'Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.downClose({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2022-05-03 13:32:43: Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.downClose({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:390:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:info  2022-05-03 13:32:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'tapparella cucina ' failed: 'Error: Command 0xa4c138ac24f9fb32/1 closuresWindowCovering.downClose({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"tapparella cucina "},"type":"zigbee_publish_error"}'

Zigbee2MQTT:debug 2022-05-03 13:54:17: Received MQTT message on 'zigbee2mqtt/tapparella cucina /set' with data '{"position":80}'
Zigbee2MQTT:debug 2022-05-03 13:54:17: Publishing 'set' 'position' to 'tapparella cucina '
Zigbee2MQTT:debug 2022-05-03 13:54:22: Received MQTT message on 'zigbee2mqtt/tapparella cucina /set' with data '{"position":100}'
Zigbee2MQTT:debug 2022-05-03 13:54:22: Publishing 'set' 'position' to 'tapparella cucina '

pieroprofita, 2022-05-08:
i updated z2m ver 1.25 but always same problem

closed 2022-06-15 by github-actions[bot]

galambert75, 2022-06-19:
[pieroprofita](https://github.com/pieroprofita) did you manage to solve this? Seems to be the same thing reported in #12774