Koenkk/zigbee2mqtt #12364: [New device support]: ZG-204ZL / TS0601 _TZE200_3towulqd / PIR sensor with illuminance

Impact123, 2022-05-03:
### Link

https://www.aliexpress.com/item/1005004018051598.html

### Database entry

{     "id":45,     "type":"EndDevice",     "ieeeAddr":"0xa4c138fa3dbe72e5",     "nwkAddr":19681,     "manufId":4417,     "manufName":"_TZE200_3towulqd",     "powerSource":"Battery",     "modelId":"TS0601",     "epList":[         1     ],     "endpoints":{         "1":{             "profId":260,             "epId":1,             "devId":1026,             "inClusterList":[                 1,                 1280,                 0             ],             "outClusterList":[                 25,                 10             ],             "clusters":{                 "genBasic":{                     "attributes":{                         "stackVersion":0,                         "dateCode":""                     }                 },                 "ssIasZone":{                     "attributes":{                         "iasCieAddr":"0x00124b00258de0d9",                         "zoneState":0                     }                 }             },             "binds":[                 {                     "cluster":0,                     "type":"endpoint",                     "deviceIeeeAddress":"0x00124b00258de0d9",                     "endpointID":1                 }             ],             "configuredReportings":[                              ],             "meta":{                              }         }     },     "appVersion":82,     "stackVersion":0,     "hwVersion":1,     "dateCode":"",     "zclVersion":3,     "interviewCompleted":true,     "meta":{         "configured":821693351     },     "lastSeen":1651589167942,     "defaultSendRequestWhen":"immediate" }

### Comments

Device reports `Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_3towulqd'`
Tested with version `1.25.1-dev commit` [b949961](https://github.com/Koenkk/zigbee2mqtt/commit/b949961)

Tuya data points:
```
PIR state: 1
Battery level: 4
Sensitivity: 9
Keep Time: 10
Illuminance Value: 12
```

Maybe this is useful too. The possible values in SmartLife are as following:
 - Max Illuminance: `1000` Lux
 - Selectable Sensitivity Value: `Low`, `Medium`, `High`, 
 - Selectable Keep Time: `10s`, `30s`, `60s`, `120s`

Some log excerpts were posted in #12033
Please let me know if you need further information.

Impact123, 2022-05-04:
I wrote this converter which partially works. The battery, lux value and the occupancy all work but the lux value is inaccurate. I have a UA-962 light meter and tried to find a formula but depending on the brightness the formula has to change to stay accurate. See comment in converter for values.
I'm not sure how to properly write the `toZigbee` part.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
// const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_3towulqd',
        },
    ],
    model: 'ZG-204ZL',
    vendor: 'Tuya',
    description: 'PIR sensor with illumination',
    fromZigbee: [
        // fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump,
        fz.ias_occupancy_alarm_1,
        {
            cluster: 'manuSpecificTuya',
            type: ['commandDataResponse', 'commandDataReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = {};
                for (const dpValue of msg.data.dpValues) {
                    const dp = dpValue.dp;
                    const value = tuya.getDataValue(dpValue);
                    switch (dp) {
                    case 1:
                        result.occupancy = !value;
                        break;
                    case 4:
                        result.battery = value;
                        break;
                    case 9:
                        result.sensitivity = value;
                        break;
                    case 10:
                        result.keep_time = value;
                        break;
                    case 12:
                        result.illuminance = value;
                        // Real lux - Measured value
                        // 0          10
                        // 30.1       778
                        // 43.3       826
                        // 57.4       850
                        // 75.7       898
                        // 98.6       922
                        // 127.6      1000
                        result.illuminance_lux = Math.round(value / 25);
                        break;
                    default:
                        meta.logger.warn(`ZG-204ZL debug: UNRECOGNIZED DP #${dp} with data ${JSON.stringify(dpValue)}`);
                    }
                }
                return result;
            },
        },
    ],
    toZigbee: [
        tz.tuya_data_point_test,
        {
            key: [
                'sensitivity', 'keep_time',
            ],
            convertGet: async (entity, key, meta) => {
                switch (key) {
                case 'sensitivity':
                    meta.logger.warn(`ZG-204ZL debug: Attempting to read key ${key}`);
                    // To be implemented
                    break;
                case 'keep_time':
                    meta.logger.warn(`ZG-204ZL debug: Attempting to read key ${key}`);
                    // To be implemented
                    break;
                default:
                    meta.logger.warn(`ZG-204ZL debug: Attempting to read key ${key} but it's not defined`);
                }
            },
            convertSet: async (entity, key, value, meta) => {
                meta.logger.warn(`ZG-204ZL debug: Attempting to write ${value} to key ${key}`);
                switch (key) {
                case 'sensitivity':
                    meta.logger.warn(`ZG-204ZL debug: Attempting to write ${value} to key ${key}`);
                    // To be implemented
                    break;
                case 'keep_time':
                    meta.logger.warn(`ZG-204ZL debug: Attempting to write ${value} to key ${key}`);
                    // To be implemented
                    break;
                default:
                    meta.logger.warn(`ZG-204ZL debug: Attempting to write key ${key} but it's not defined`);
                }
            },
        },
    ],
    // onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(), e.illuminance(), e.illuminance_lux(), e.occupancy(),
        exposes.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
        exposes.enum('keep_time', ea.ALL, [10, 30, 60, 120]).withDescription('PIR keep time in seconds'),
    ],
};

module.exports = definition;
```

jaspernlp, 2022-05-06:
I've bought the same sensor. The path for the software I made was a bit different than above. The changes are in three different files: devices/tuya.js, converters/toZigbee.js and converters/fromZigbee.js. Below you will find the code that goes into these files. 
Difficult to figure out was when the config can be changed and read. It only works if the sensor is active, so it detects motion. If the green popups show with the debug info then the request to update/read will work. The read of the config will always return 3 messages, for the battery, sensitivity and the keep_time.

In the tuya.js file
module.exports = [
   {    fingerprint: [
                {modelID: 'TS0601', manufacturerName: '_TZE200_3towulqd'}
            ],
        model: 'ZG-204ZL',
        vendor: 'TuYa',
        description: 'Luminance Motion Sensor',
        fromZigbee: [
                fz.ignore_basic_report,
                fz.tuya_data_point_dump,
                fz.ZG204ZL_lms],
        toZigbee: [
                tz.tuya_data_point_test,
                tz.ZG204ZL_lms,
                ],
        exposes: [
                e.occupancy(), e.illuminance(), e.battery(),
                exposes.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
                exposes.enum('keep_time', ea.ALL, ['10', '30', '60', '120']).withDescription('PIR keep time in seconds'),
        ],
    },
... // the rest of the file

In the fromZigbee.js file
const converters = {
        ZG204ZL_lms: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
//meta.logger.info(JSON.stringify(msg));
                const result = {};
                for (const dpValue of msg.data.dpValues) {
                        const dp = dpValue.dp; // First we get the data point ID
                        const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                        switch (dp) {
                        case 1: // DPID that we added to common
                                result.occupancy = (value === 0);
                                break;
                        case 4:
                                result.battery = value;
                                break;
                        case 9:
                                const senslookup = {'0': 'low', '1': 'medium', '2': 'high'};
                                result.sensitivity = senslookup[value];
                                break;
                        case 10:
                                const keeplookup = {'0': '10', '1': '30', '2': '60', '3': '120'};
                                result.keep_time = keeplookup[value];
                                break;
                        case 12:
                                result.illuminance = value;
                                break;
                        default:
                                meta.logger.warn(`zigbee-herdsman-converters:ZG204ZL_lms: NOT RECOGNIZED DP #${
                                        dp} with data ${JSON.stringify(dpValue)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
            }
        }
        return result;
    },
},

    // #region Generic/recommended converters ... and the rest of the file

And the addition to the toZigbee.js file
const converters = {
    ZG204ZL_lms: {
        key: [
            'sensitivity', 'keep_time',
        ],

       convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'sensitivity':
                await tuya.sendDataPointEnum(entity, 9, {'low': 0, 'medium': 1, 'high': 2 }[value]);
                break;
            case 'keep_time':
                await tuya.sendDataPointEnum(entity, 10, {'10': 0, '30': 1, '60': 2, '120': 3}[value]);
                break;
            default: // Unknown key
                throw new Error(`tz.ZG204ZL_lms: Unhandled key ${key}`);
            }
        },

        convertGet: async (entity, key, meta) => {
            switch (key) {
            case 'sensitivity':
                await tuya.sendDataPointEnum(entity, 9, 0, 'dataQuery' );
                break;
            case 'keep_time':
                await tuya.sendDataPointEnum(entity, 10, 0, 'dataQuery' );
                break;
             default: // Unknown key
                throw new Error(`Unhandled key toZigbee.ZG204ZL_lms.convertGet ${key}`);
            }
        }
    },

// #region Generic converters

Impact123, 2022-05-06:
Thanks, now everything seems to work. I deviated from the instructions and put it all into one file because it's easier to test when everything is contained. Do you plan to create a PR for this?

jaspernlp, 2022-05-06:
That is fine, I'll create a PR for the change, after I remove the debug part and will place the code not as the first (that was the easiest for editing now).

jaspernlp, 2022-05-07:
Created the PRs https://github.com/Koenkk/zigbee-herdsman-converters/pull/4219 and https://github.com/Koenkk/zigbee2mqtt.io/pull/1372

Koenkk, 2022-05-08:
Merged, thanks @jaspernlp 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-08 by Koenkk

Pferdebockwurst, 2022-09-28:
@Koenkk I have received these sensors from AliExpress today. It's definitely the same device, the model number is ZG-204ZL. But the internal ZigBee manufacturer name is _TZE200_1ibpyhdc, which why it's recognized as an unsupported device. I don't know why they use a different manufacturer name here, but please add this to the database.

Koenkk, 2022-09-28:
@Pferdebockwurst added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

FragMenthor, 2022-09-30:
I also received the "_TZE200_1ibpyhdc". I don't use dev version, can someone give the full external converter, please? In one file? Thanks all!

FragMenthor, 2022-09-30:
> Thanks, now everything seems to work. I deviated from the instructions and put it all into one file because it's easier to test when everything is contained. Do you plan to create a PR for this?

One external converter file to use in the non dev version? Please, I need that 😊! Thanks in advance!

Koenkk, 2022-09-30:
@FragMenthor it will be supported in tomorrows new release

FragMenthor, 2022-09-30:
> @FragMenthor it will be supported in tomorrows new release

Thanks!!!😊

FragMenthor, 2022-10-01:
> @FragMenthor it will be supported in tomorrows new release

It's ON 😊💪 thanks!!!

lippertmarkus, 2022-11-02:
I found that the sensor updates the illuminance (line 2) 3s after it detected the motion (line 1). After the `keep_time` it first updates illuminance (line 3) before sending the `occupancy=false` message (line 4):
```
Info 2022-11-02 20:33:39MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":124,"keep_time":"10","linkquality":123,"occupancy":true,"sensitivity":"high"}'
Info 2022-11-02 20:33:42MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":136,"keep_time":"10","linkquality":127,"occupancy":true,"sensitivity":"high"}'
Info 2022-11-02 20:33:51MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":124,"keep_time":"10","linkquality":123,"occupancy":true,"sensitivity":"high"}'
Info 2022-11-02 20:33:51MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":124,"keep_time":"10","linkquality":131,"occupancy":false,"sensitivity":"high"}'
```

Can someone confirm that it's the same for their sensor?

I want to only switch lights on only if the room is dark, but I can't switch on the lights when motion is detected (line 1) because the illuminance value is not updated yet. I would need to wait 3s after the motion is detected for the updated illuminance value (line 2) to determine if the room is dark enough. It's not very user-friendly to wait for 3s before switching the light on.

Pferdebockwurst, 2022-11-02:
> 

Here is one cycle of occupancy from my log:
`2022-11-02 21:00:17MQTT publish: topic 'zigbee2mqtt/Bewegungsmelder Foyer', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":72,"occupancy":true,"sensitivity":"high"}'
2022-11-02 21:00:17MQTT publish: topic 'zigbee2mqtt/Foyer Decke', payload '{"brightness":228,"color_mode":"color_temp","color_temp":500,"color_temp_startup":65535,"linkquality":83,"state":"ON"}'
2022-11-02 21:00:20MQTT publish: topic 'zigbee2mqtt/Bewegungsmelder Foyer', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":76,"occupancy":true,"sensitivity":"high"}'
2022-11-02 21:00:33MQTT publish: topic 'zigbee2mqtt/Bewegungsmelder Foyer', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":76,"occupancy":true,"sensitivity":"high"}'
2022-11-02 21:00:35MQTT publish: topic 'zigbee2mqtt/Bewegungsmelder Foyer', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":76,"occupancy":false,"sensitivity":"high"}'`

lippertmarkus, 2022-11-03:
Thanks! yeah, the illuminance doesn't change (or only slightly) because the illuminance is read after the light was switched on..

pratahsinnetamby, 2023-03-29:
Pardon my ignorance, but does this work with Skyconnect (which I believe is ZHA)

FragMenthor, 2023-08-01:
Did anyone find a way to force update when changes are only in lux? The device only updates lux on occupancy change...

fherreror, 2023-08-29:
> I found that the sensor updates the illuminance (line 2) 3s after it detected the motion (line 1). After the `keep_time` it first updates illuminance (line 3) before sending the `occupancy=false` message (line 4):
> 
> ```
> Info 2022-11-02 20:33:39MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":124,"keep_time":"10","linkquality":123,"occupancy":true,"sensitivity":"high"}'
> Info 2022-11-02 20:33:42MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":136,"keep_time":"10","linkquality":127,"occupancy":true,"sensitivity":"high"}'
> Info 2022-11-02 20:33:51MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":124,"keep_time":"10","linkquality":123,"occupancy":true,"sensitivity":"high"}'
> Info 2022-11-02 20:33:51MQTT publish: topic 'zigbee2mqtt/0xa4c138f51023e8d5', payload '{"battery":100,"illuminance":124,"keep_time":"10","linkquality":131,"occupancy":false,"sensitivity":"high"}'
> ```
> 
> Can someone confirm that it's the same for their sensor?
> 
> I want to switch lights on only if the room is dark, but I can't switch on the lights when motion is detected (line 1) because the illuminance value is not updated yet. I would need to wait 3s after the motion is detected for the updated illuminance value (line 2) to determine if the room is dark enough. It's not very user-friendly to wait for 3s before switching the light on.

Same problem here. I have 4 of these sensors. Two of them (the ones I paired many months ago), have this behaviour: on motion, occupancy is updated, but not lux. 3 o 4 seconds later, a new mqtt message is send with the lux update.
However, the other two sensors work as expected: first they update illumnance, and a fraction of a second later, they update occupancy.
The faulty sensors, however do work as expected on occupancy=false, i.e. when no motion is detected, they update first illuminance and a fraction of a second later, they update occupancy.

Can be this a zigbee2mqtt problem? or may it be a hardware problem (is 50% of my sensors... the ones I paired some months ago).

All of them are TS0601 - ZG-204ZL with manufacterer _TZE200_3towulqd

dany547, 2024-07-12:
i just got 2 of this sensors from Aliexpress, integrated them with Z2MQTT but the Illuminance value is way of. In a room where it should be a maximum value of 50 lux, it reports something like 2800 lux
`{
    "battery": 100,
    "illuminance": 2856,
    "illuminance_interval": null,
    "keep_time": null,
    "linkquality": 142,
    "occupancy": false,
    "sensitivity": null
}`

Scoff123, 2024-07-12:
That's correct, it's reporting 'raw illuminance' as opposed to 'illuminance_lux' like other lux sensors report. That is what is stated in the Z2M device support page for this device. I'm not sure if anyone has been able to figure out a change request / converter for this device yet, to allow it to report 'illuminance_lux' but it would be helpful.

dany547, 2024-07-13:
> That's correct, it's reporting 'raw illuminance' as opposed to 'illuminance_lux' like other lux sensors report. That is what is stated in the Z2M device support page for this device. I'm not sure if [...]

ow...that makes sence. Another thing that ive noticed is the fact that the Illuminance is not exposed to Home Assistant, thus i cannot use the sensors output for automations. But the sensor clearly outputs Raw Illuminance values in the log: `2024-07-13 07:31:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Motion Sensor', payload '{"battery":100,"illuminance":1941,"illuminance_interval":1,"keep_time":"60","linkquality":116,"occupancy":false,"sensitivity":"high"}'`
[image]

dany547, 2024-07-18:
managed to make it work! Now it reports also Illuminance in HASS.

- remove the device (force remove)
- repair the device
- restart Home Assistant
- go to Integration/Devices and enable Illuminance