Koenkk/zigbee2mqtt #12365: Xiaomi Aqara WS-EUK03 not reporting Action in Decoupled Mode

mtwilliams5, 2022-05-03:
### What happened?

I have set up the wall switch in Z2M, and can control the various switches correctly - in control_relay mode, it turns on and off the light, and when in decoupled mode, pressing the switch does not alter the light's power state.

However, the Action exposed entity (and thus the action sensor in HA) is not updating. I've monitored the logs and listened for state change events as I press the button, I've tried re-pairing the device, but Action never seems to get updated. I'm aware it gets updated only momentarily and then reset back to an empty string, but the logs are showing that that initial update never happens.

### What did you expect to happen?

When I press the switch, Action should change to `single`, and then change back to `""` a moment later

### How to reproduce it (minimal and precise)

1. Configure Z2M with homeassistant integration (to enable the action property)
2. Pair switch in Z2M
3. Set switch to decoupled mode
4. Open (and monitor) logs in Z2M
5. In HA->Developer Tools->Events, Start Listening on `state_changed`
6. Press switch, double-press switch, etc, and check the log and event outputs

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x26580700

### Adapter

ConBee II

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8611631/log.txt)

closed 2022-06-10 by github-actions[bot]

Woutoud, 2022-08-06:
I'm running into the same issue. Home Assistant is reporting the state of the Action in de Sensors section as "none".

mtwilliams5, 2022-08-14:
I've finally managed to get this issue to go away, but I don't know how useful the solution may be to others.

I recently bought the two-gang version of this switch and fitted it - and it was able to register actions straight away, and I was able to set up the automations that I wanted.

So, I turned my attention back to this one; it still wasn't publishing any action information, but going into the Zigbee2MQTT management page and hitting the reconfigure button seems to have done the trick this time.

The change is (by design) momentary; action will come through as either "single" or "double", and then be immediately reset to an empty string. But I was able to see this happening by sticking a debug node on the state change within NodeRED (if you don't use NodeRED, you should be able to see it within the HA Logbook). I've now been able to turn the switch into something useful, rather than just a dud button that at least stopped people cutting power to my smart bulb unintentionally.

j0hann3s, 2023-02-17:
Got the same issue but no matter how many times I pair again or use the reconfigure button, nothing gets registered when clicking the button in decoupled mode.

oliverholliday, 2023-04-27:
This is still a bug, please could it be reopened.