Koenkk/zigbee2mqtt #12370: [New device support]: Zemismart TB26-6

rodrigogbs, 2022-05-03:
### Link

https://www.zemismart.com/products/tb26-6

### Database entry

{"id":51,"type":"EndDevice","ieeeAddr":"0xa4c1387c4028ac15","nwkAddr":51060,"manufId":4417,"manufName":"_TZE200_9mahtqtg","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":1,"65534":0,"appVersion":70,"modelId":"TS0601","manufacturerName":"_TZE200_9mahtqtg","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd4ade8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651595324059,"defaultSendRequestWhen":"immediate"}

### Comments

I bought the brand new Zemismart TB26-6 6 gang switch (https://pt.aliexpress.com/item/1005003973677496.html?spm=a2g0o.productlist.0.0.4cc53716IEY7ek&algo_pvid=51bc722f-a5be-44eb-9a05-f95606d0faaa&algo_exp_id=51bc722f-a5be-44eb-9a05-f95606d0faaa-2&pdp_ext_f=%7B%22sku_id%22%3A%2212000027609734570%22%7D&pdp_npi=1%40dis%7CBRL%7C%7C176.06%7C%7C%7C%7C%7C%40210318cf16515992358918782e0163%7C12000027609734570%7Csea) but it doesn't work on Zigbee2mqtt.

I tried to configure an external converter with the code below, without success.

And the logs are:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing...
[13:25:02] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[13:25:05] INFO: Handing over control to Zigbee2mqtt Core ...

zigbee2mqtt@1.25.1 start
node index.js
Zigbee2MQTT:error 2022-05-03 13:26:53: Failed to configure '0xa4c1387c4028ac15', attempt 1 (Error: Bind 0xa4c1387c4028ac15/1 genOnOff from '0x00124b001cd4ade8/1' failed (AREQ - ZDO - bindRsp after 10000ms)
at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
at listOnTimeout (internal/timers.js:557:17)
at processTimers (internal/timers.js:500:7))
Zigbee2MQTT:error 2022-05-03 13:26:59: Device '0xa4c1387c4028ac15' has no endpoint 'center_left'
Zigbee2MQTT:error 2022-05-03 13:27:02: Device '0xa4c1387c4028ac15' has no endpoint 'bottom_left'
Zigbee2MQTT:error 2022-05-03 13:27:03: Device '0xa4c1387c4028ac15' has no endpoint 'top_right'
Zigbee2MQTT:error 2022-05-03 13:27:03: Device '0xa4c1387c4028ac15' has no endpoint 'center_right'
Zigbee2MQTT:error 2022-05-03 13:27:04: Device '0xa4c1387c4028ac15' has no endpoint 'bottom_right'
Zigbee2MQTT:error 2022-05-03 13:27:07: Device '0xa4c1387c4028ac15' has no endpoint 'center_left'
Zigbee2MQTT:error 2022-05-03 13:27:08: Device '0xa4c1387c4028ac15' has no endpoint 'bottom_left'
Zigbee2MQTT:error 2022-05-03 13:27:09: Device '0xa4c1387c4028ac15' has no endpoint 'top_right'
Zigbee2MQTT:error 2022-05-03 13:27:10: Device '0xa4c1387c4028ac15' has no endpoint 'center_right'
Zigbee2MQTT:error 2022-05-03 13:27:11: Device '0xa4c1387c4028ac15' has no endpoint 'bottom_right'
Zigbee2MQTT:error 2022-05-03 13:27:16: Publish 'get' 'state' to '0xa4c1387c4028ac15' failed: 'Error: Read 0xa4c1387c4028ac15/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51060 - 1 - 29 - 6 - 1 after 10000ms)'
Zigbee2MQTT:error 2022-05-03 13:27:42: Publish 'get' 'state' to '0xa4c1387c4028ac15' failed: 'Error: Read 0xa4c1387c4028ac15/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51060 - 1 - 35 - 6 - 1 after 10000ms)'
Zigbee2MQTT:error 2022-05-03 13:27:44: Failed to configure '0xa4c1387c4028ac15', attempt 2 (TypeError: Cannot read property 'bind' of undefined
at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
at Object.configure (/app/dist/util/externally-loaded.js:27:25)
at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-05-03 13:28:42: Failed to configure '0xa4c1387c4028ac15', attempt 3 (TypeError: Cannot read property 'bind' of undefined
at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
at Object.configure (/app/dist/util/externally-loaded.js:27:25)
at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2022-05-03 13:35:23: Publish 'set' 'state' to 'Tomada Fita Led TV Suíte' failed: 'Error: Command 0xa4c13830d4c9ec18/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43597 - 1 - 59 - 6 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-05-03 13:40:04: Failed to configure 'Sensor Movimento Sacada', attempt 1 (Error: Bind 0x00124b0023b987a3/1 genPowerCfg from '0x00124b001cd4ade8/1' failed (AREQ - ZDO - bindRsp after 10000ms)
at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
at listOnTimeout (internal/timers.js:557:17)
at processTimers (internal/timers.js:500:7))
Zigbee2MQTT:error 2022-05-03 13:41:08: Failed to configure 'Sensor Movimento Sacada', attempt 2 (Error: Bind 0x00124b0023b987a3/1 genPowerCfg from '0x00124b001cd4ade8/1' failed (AREQ - ZDO - bindRsp after 10000ms)
at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
at listOnTimeout (internal/timers.js:557:17)
at processTimers (internal/timers.js:500:7))
Zigbee2MQTT:error 2022-05-03 13:50:45: Failed to configure 'Sensor Movimento Sacada', attempt 3 (Error: Bind 0x00124b0023b987a3/1 genPowerCfg from '0x00124b001cd4ade8/1' failed (AREQ - ZDO - bindRsp after 10000ms)
at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
at listOnTimeout (internal/timers.js:557:17)
at processTimers (internal/timers.js:500:7))
[... 20 lines cut ...]
zigbeeModel: ['TS0601'],
model: 'ZM-TB26-6',
vendor: 'Zemismart',
description: 'Smart light switch - 6 gang with neutral wire CUSTOM',
extend: extend.switch(),
exposes: [e.switch().withEndpoint('top_left'), e.switch().withEndpoint('center_left'), e.switch().withEndpoint('bottom_left'), e.switch().withEndpoint('top_right'), e.switch().withEndpoint('center_right'), e.switch().withEndpoint('bottom_right')],
endpoint: (device) => {
return {'top_left': 1, 'center_left': 2, 'bottom_left': 3, 'top_right': 4, 'center_right': 5, 'bottom_right': 6};
},
whiteLabel: [{vendor: 'ShinaSystem', model: 'SBM300Z6', description: 'Zigbee switch'}],
meta: {multiEndpoint: true, disableDefaultResponse: true},
configure: async (device, coordinatorEndpoint, logger) => {
const endpoint = device.getEndpoint(1);
await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(5), coordinatorEndpoint, ['genOnOff']);
await reporting.bind(device.getEndpoint(6), coordinatorEndpoint, ['genOnOff']);
},
};

module.exports = definition;
```

rodrigogbs, 2022-05-14:
> Hi, have you had any success pairing this since your last post?
> About to get mine this weekend

Unfortunately not yet…

rafaelsiqueira11, 2022-05-30:
Hi, have you had any success pairing this since your last post?

rafaelsiqueira11, 2022-05-30:
partial success

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601_switch_6_gang', 
    vendor: '_TZE200_9mahtqtg',
    description: 'Interruptor Zigbee 6 Gang', 

    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l5').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l6').setAccess('state', ea.STATE_SET)],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch],
    toZigbee: [tz.tuya_switch_state],
    meta: {multiEndpoint: true},
    whiteLabel: [
        {vendor: 'Norklmes', model: 'MKS-CM-W5'},
        {vendor: 'Somgoms', model: 'ZSQB-SMB-ZB'},
        {vendor: 'Moes', model: 'WS-EUB1-ZG'},
        {vendor: 'AVATTO', model: 'ZGB-WS-EU'},
    ],
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1};
    }

};

module.exports = definition;`

rodrigogbs, 2022-05-31:
> partial success
>
> `const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = [...]

Only 4 buttons are working and with a strange slowness...same there?

rodrigogbs, 2022-06-03:
@rafaelsiqueira11 any news? Your progress was great until here...

Koenkk, 2022-06-04:
This device has just been added: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4315

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-04 by Koenkk

rodrigogbs, 2022-06-06:
Tested in dev branch and now it's working, but with a big slowness, 3 to 5 secs between the command and execution...and I'm 30cm far from the coordinator...

Does anyone know what could be?

platbr, 2022-06-06:
@rodrigogbs I haven't this problem but i didn't like this device. It isn't  router.

@Koenkk I'm noob at zigbee world. I guess zigbee2mqtt only deal with raw data coming from and going to devices and it uses the `zigbee-herdsman-converters` to adapt this messages (maybe too shallow). Issues related to response time generally are originated at coordinator firmware level. Am i wrong?

rodrigogbs, 2022-06-07:
Tks guys, I found the problem, the device is defective, I bought it on the pre-sales and this delay is an issue of the first production batch, according to Zemismart, they will change for a new one, that is router too!

douglas-srs, 2022-06-22:
> Tks guys, I found the problem, the device is defective, I bought it on the pre-sales and this delay is an issue of the first production batch, according to Zemismart, they will change for a new one, that is router too!

Hey man, I bought the same product and I'm already using the new converter but only the first 4 buttons works as well, I don't have the 3 to 5 secs delay as you had tho, did you manage to make the last 2 buttons of your device work? If not, what did you tell to Zemismart to make them replace your product? Mine is an enddevice too =(

douglas-srs, 2022-06-23:
> 

thank you for answering me, yes, from AliExpress, did you have to send him the old device?

rodrigogbs, 2022-06-23:
> > 
> 
> thank you for answering me, yes, from AliExpress, did you have to send him the old device?

@douglas-srs nope! they just sent me a new one...but I had the delay issue, not only the enddevice spec...

douglas-srs, 2022-06-23:
> > > 
> > 
> > 
> > thank you for answering me, yes, from AliExpress, did you have to send him the old device?
> 
> @douglas-srs nope! they just sent me a new one...but I had the delay issue, not only the enddevice spec...

I see, thank you very much, the zigbee2mqtt edge version fixed my button issues!

alexh8555, 2024-02-08:
Has anyone tried TB26 series switch with different numbers of gangs?(TB26-1, TB26-2, TB26-3)

I've checked only 4 and 6 are listed as supported device.
I wonder others will work.

Thanks

rodrigogbs, 2024-02-08:
> Has anyone tried TB26 series switch with different numbers of gangs?(TB26-1, TB26-2, TB26-3)
> 
> I've checked only 4 and 6 are listed as supported device. I wonder others will work.
> 
> Thanks

all of them are currently supported by native Z2M, just pair and be happy.

alexh8555, 2024-02-08:
Thank you for the quick reply!
I'll get them soon.