Koenkk/zigbee2mqtt #12372: Aqara E1 (MCCGQ14LM) doesn't display the batterylevel

Jens-Wymeersch, 2022-05-03:
### What happened?

I bought the new Aqara door sensors. After pairing, I've noticed that it doesn't display any battery level. Although if I look at https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html, it should. Can someone please have a look ?

### What did you expect to happen?

To see the battery level of these devices.

### How to reproduce it (minimal and precise)

I have 2 devices with the same problem.

### Zigbee2MQTT version

1.25.1 

### Adapter firmware version

20220219

### Adapter

ZZH!

Otnow, 2022-05-04:
Try using this [external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters):

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.magnet.acn001'],
        model: 'MCCGQ14LM',
        vendor: 'Xiaomi',
        description: 'Aqara E1 door & window contact sensor',
        fromZigbee: [fz.ias_contact_alarm_1, fz.aqara_opple, fz.battery],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2850_3000_log'}},
        exposes: [e.contact(), e.battery(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
        },
        ota: ota.zigbeeOTA,
};

module.exports = definition;
```

To further improve support for this sensor, please [enable debug mode](https://www.zigbee2mqtt.io/guide/configuration/logging.html) (`log_level: debug`) and write a log that will contain the following actions:

- sensor pairing
- press button on sensor once
- open the sensor contact
- press button on sensor once
- close the sensor contact
- press button on sensor once
- check OTA update (_in z2m web interface on OTA tab in sensor line, click "Check for new updates" button_)
- press button on sensor once

Jens-Wymeersch, 2022-05-04:
@Otnow thank you for your quick reply.
I've installed Z2M on HA. Not sure how to get this converter in there. In the OTA, it only indicates IKEA/HUE devices that can be updated. Not sure how I can get the rest. Any ideas ?

For the logs, as the aqara sensors are a drag to pair, I'll wait for the next batch to send the log data you requested.

Otnow, 2022-05-04:
> I've installed Z2M on HA. Not sure how to get this converter in there.

As far as I understand, _z2m_ is installed as an addon in _HA_.

If this is the case, then the external converter file can be placed in the _z2m_ configuration folder by the following methods:

- using the **File editor** addon, create a file manually in the `/config/zigbee2mqtt` directory

- using the **Samba share** addon to open network access and copy the file to the `\config\zigbee2mqtt` directory

After placing the file, you need to [specify it in the settings](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) and restart _z2m_.
***

> In the OTA, it only indicates IKEA/HUE devices that can be updated. Not sure how I can get the rest. Any ideas ?

On the OTA tab, the sensor will appear only when using an external converter.
***

> For the logs, as the aqara sensors are a drag to pair, I'll wait for the next batch to send the log data you requested.

For a better understanding of the operation of the sensor, you need to write a log _already when the external converter is used, with all the indicated actions and in the indicated sequence_.

If writing a log with the indicated actions is difficult for you, then do not do it.

Jens-Wymeersch, 2022-05-04:
@Otnow Thank you for the explanation, and I'll handle it next weekend.

Jens-Wymeersch, 2022-05-08:
@Otnow I was wondering if we could do something similar for the following problem I have https://github.com/Koenkk/zigbee2mqtt/issues/12418

Otnow, 2022-05-11:
> Thank you for the explanation, and I'll handle it next weekend.

[This PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4230) has fixed the current issue.

***
> I was wondering if we could do something similar for the following problem I have #12418

I do not have such a sensor and I am not familiar with the features of _Tuya_ devices, so I cannot help.

Otnow, 2022-05-11:
@Noltari, thanks, but the information is not complete, because you didn't use the external converter I suggested.

Can you use an external converter and follow all the suggested steps?

Noltari, 2022-05-11:
> @Noltari, thanks, but the information is not complete, because you didn't use the external converter I suggested.
> 
> Can you use an external converter and follow all the suggested steps?

Sorry, I missed that part...
```
debug 2022-05-11 20:48:42: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"q9feg-2","value":true}'
info  2022-05-11 20:48:42: Zigbee: allowing new devices to join.
info  2022-05-11 20:48:42: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"q9feg-2"}'
info  2022-05-11 20:48:56: Device '0x54ef441000290748' joined
info  2022-05-11 20:48:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000290748","ieee_address":"0x54ef441000290748"},"type":"device_joined"}'
info  2022-05-11 20:48:57: MQTT publish: topic 'homeassistant/binary_sensor/0x54ef441000290748/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x54ef441000290748/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://zigbee2mqtt.noltari.es/#/device/0x54ef441000290748/info","identifiers":["zigbee2mqtt_0x54ef441000290748"],"manufacturer":"Xiaomi","model":"Aqara E1 door & window contact sensor (MCCGQ14LM)","name":"0x54ef441000290748"},"device_class":"door","name":"0x54ef441000290748 contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x54ef441000290748","unique_id":"0x54ef441000290748_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
info  2022-05-11 20:48:57: MQTT publish: topic 'homeassistant/sensor/0x54ef441000290748/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x54ef441000290748/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://zigbee2mqtt.noltari.es/#/device/0x54ef441000290748/info","identifiers":["zigbee2mqtt_0x54ef441000290748"],"manufacturer":"Xiaomi","model":"Aqara E1 door & window contact sensor (MCCGQ14LM)","name":"0x54ef441000290748"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x54ef441000290748 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000290748","unique_id":"0x54ef441000290748_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-05-11 20:48:57: MQTT publish: topic 'homeassistant/sensor/0x54ef441000290748/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x54ef441000290748/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://zigbee2mqtt.noltari.es/#/device/0x54ef441000290748/info","identifiers":["zigbee2mqtt_0x54ef441000290748"],"manufacturer":"Xiaomi","model":"Aqara E1 door & window contact sensor (MCCGQ14LM)","name":"0x54ef441000290748"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"0x54ef441000290748 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000290748","unique_id":"0x54ef441000290748_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info  2022-05-11 20:48:57: MQTT publish: topic 'homeassistant/sensor/0x54ef441000290748/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x54ef441000290748/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://zigbee2mqtt.noltari.es/#/device/0x54ef441000290748/info","identifiers":["zigbee2mqtt_0x54ef441000290748"],"manufacturer":"Xiaomi","model":"Aqara E1 door & window contact sensor (MCCGQ14LM)","name":"0x54ef441000290748"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x54ef441000290748 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef441000290748","unique_id":"0x54ef441000290748_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2022-05-11 20:48:57: MQTT publish: topic 'homeassistant/sensor/0x54ef441000290748/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x54ef441000290748/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://zigbee2mqtt.noltari.es/#/device/0x54ef441000290748/info","identifiers":["zigbee2mqtt_0x54ef441000290748"],"manufacturer":"Xiaomi","model":"Aqara E1 door & window contact sensor (MCCGQ14LM)","name":"0x54ef441000290748"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x54ef441000290748 update state","state_topic":"zigbee2mqtt/0x54ef441000290748","unique_id":"0x54ef441000290748_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
[... cut ...]
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:194:39)
debug 2022-05-11 20:51:31: Received Zigbee message from '0x54ef441000290748', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.magnet.acn001"}' from endpoint 1 with groupID 0
```

Otnow, 2022-05-11:
@Noltari, great, thanks, just what you need 👍

Created a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4237) with improvements based on your log.

Otnow, 2022-05-12:
@Jens-Wymeersch, changes is now available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), and will appear in the _master branch_ at the beginning of next month.

Otnow, 2022-05-13:
> Thank you for your efforts. I assume I don't need to update the firmware to make this work, right ?

You are welcome 🙂

If you mean the firmware of the coordinator, then it does not need to be updated.

To take advantage of the changes right now, you can [switch to the dev version](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) of _z2m_, or wait for the release early next month.

closed 2022-06-20 by github-actions[bot]

ViperScull, 2023-11-18:
I have the same problem with the MCCGQ11LM model of the E1 door sensor. Could you patch the xiaomi.js for this model as you did for the MCCGC14LM?