Koenkk/zigbee2mqtt #12373: Newly added devices missing output clusters

zone31, 2022-05-03:
### What happened?

When i add new devices to the network, there are missing output clusters. 
For  example, with an ikea bulb LED1624G9, there are missing clusters when adding the device.
[image]
[image]
This bulb has been reporting and bound before, and i have 2 identical that are currently behaving.

I am not entirely sure this happened on 1.25.1, since 3 other bulbs are having the same issue, and those where added to the network before i updated.

### What did you expect to happen?

The newly added device to have the same configuration as the old ones.

### How to reproduce it (minimal and precise)

Pair a new device to the network.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

zzh!

### Debug log

[log_anon.txt](https://github.com/Koenkk/zigbee2mqtt/files/8614094/log_anon.txt)

RitaSghe, 2022-05-04:
Same problem with Vimar switch ...

[image]

zone31, 2022-05-08:
Tried to pair and unpair the bulb in 1.24.0, and the problem consists. Could this have anything to do with the adapter firmware?

zone31, 2022-05-08:
Re flashing the adapter did not help

zone31, 2022-05-08:
Deleting everything, generating a new neworkkey and panid gave the same result.

zone31, 2022-05-08:
I tried to reset the network with a new key, and recorded the herdsman logs.
[herdsman logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/8647097/herdsman.logs.txt)

MattWestb, 2022-05-08:
The IKEA CWS1 is / was one ZLL light but have getting one updated firmware in the winter that is Zigbee 3 so most cluster is changed on the endpoints.
Read the firmware version and if its 1.X its the old ZLL and if its 2.X its Zigbee 3.
I recommending deleting and adding the device after updating it fro ver 1.x to 2.x for getting the cluster setting being OK in the system.

zone31, 2022-05-08:
@MattWestb the current installed firmware on LED1624G9 is 20211025 2.3.086. 
however, if i press "check for new updates" i get  "Failed to check if update available for '0xbc33acfffe2bfd8c' (No image available for imageType '10243')"

MattWestb, 2022-05-08:
Its on the latest firmware and very likely its not getting any more updated then its one of the oldest lights IKEA (600 lm) is having and its sold out then the new 800 lm is the current version.
If you was installing it before 13th December 2021 it was added with ZLL firmware (ver 1.X) and is having changed endpoint / cluster configuration then the update was made.

zone31, 2022-05-08:
Even after deleting the bulb and adding it again the issue persist. i also have the same problem on 3 LED1923R5 bulbs.

zone31, 2022-05-09:
Also tried lowering the firmware version on my zzh! stick to 20220103, but i still see the same behaviour

zone31, 2022-05-09:
Having a look at the database.db, and looking at a working and non working LED1923R5, i see that the genbasic is different

The working one:
```
{
    "id": 5,
    "type": "Router",
    "ieeeAddr": "0x2c1165fffe94b819",
    "nwkAddr": 39288,
    "manufId": 4476,
    "manufName": "IKEA of Sweden",
    "powerSource": "Mains (single phase)",
    "modelId": "TRADFRI bulb GU10 CWS 345lm",
    "epList": [
        1,
        242
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 269,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                8,
                768,
                2821,
                4096,
                64636
            ],
            "outClusterList": [
                25
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TRADFRI bulb GU10 CWS 345lm",
                        "manufacturerName": "IKEA of Sweden",
                        "powerSource": 1,
                        "zclVersion": 3,
                        "appVersion": 80,
                        "stackVersion": 101,
                        "hwVersion": 1,
                        "dateCode": "2019062886LEL08N",
                        "swBuildId": "1.0.021"
                    }
                },
                "lightingColorCtrl": {
                    "attributes": {
                        "colorCapabilities": 31,
                        "colorTempPhysicalMin": 250,
                        "colorTempPhysicalMax": 454,
                        "colorMode": 2,
                        "currentX": 28607,
                        "currentY": 26479,
                        "enhancedCurrentHue": 0,
                        "currentSaturation": 183,
                        "colorTemperature": 337,
                        "startUpColorTemperature": 65535
                    }
                },
                "genLevelCtrl": {
                    "attributes": {
                        "currentLevel": 211
                    }
                },
[... 138 lines cut ...]
            "devId": 97,
            "inClusterList": [],
            "outClusterList": [
                33
            ],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 80,
    "stackVersion": 101,
    "hwVersion": 1,
    "dateCode": "2019062886LEL08N",
    "swBuildId": "1.0.021",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 2
    },
    "lastSeen": 1652127717128,
    "defaultSendRequestWhen": "immediate"
}
```

zone31, 2022-05-15:
Tried with a sonoff adapter, and the issue still persists.
Changed the channel, and the same result.
Here is the new log, with the paring process on channel 25 with the sonoff adapter, with herdsman logging enabled:
[log_new_channel.txt](https://github.com/Koenkk/zigbee2mqtt/files/8694779/log_new_channel.txt)
I also reflashed the adapter, the paring process starts at: 7480

I noticed some stuff:
```
2022-05-15T10:58:46.054Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '61851' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
```
---
```
2022-05-15T10:58:49.356Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"TRADFRI bulb E27 CWS opal 600lm"}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":61851,"endpoint":1,"linkquality":229,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-05-15 12:58:49: Received Zigbee message from '0xbc33acfffe2bfd8c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI bulb E27 CWS opal 600lm"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-05-15 12:58:49: Skipping message, definition is undefined and still interviewing
2022-05-15T10:58:49.369Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Also, it seems like all information from the genbasic attributes are ignored. this is also seen in the database.db file. This pattern repeats for all genbasic attributes.

---
```
Zigbee2MQTT:debug 2022-05-15 12:58:50: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0xbc33acfffe2bfd8c!
```
---

So, to summarise, i have done:
* Downgrade of zigbee2mqtt, new zzh firmware 
* Downgrade of zzh stick. new zigbee2mqtt
* Downgrade both zzg and zigbee2mqtt
* Repaired multiple times
* Tried on a new adapter (sonoff)
* Tried with and without an usb extension cord
* Redid the config fully, and cleared all in the data folder(Using /app/data in a folder from docker-compose)
* Tried on channel 25

@Koenkk Sorry for tagging you, but i am out of ideas. Do you have any idea what i am doing wrong?

zone31, 2022-05-15:
well i'm dumb..... It was not clear that the bind and reporting is automatically applied when binding a remote.

closed 2022-05-15 by zone31