Koenkk/zigbee2mqtt #12375: Heiman SMHM-I1 Motion Sensor tamper entity

migor7, 2022-05-04:
### What happened?

After updating z2m to version 1.25.1-1, the tamper entity disappeared from the Heiman SMHM-I1 Motion Sensor

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20201010

### Adapter

zStack30x

Koenkk, 2022-05-04:
This is not supported by the device, see https://github.com/Koenkk/zigbee2mqtt/issues/12076 (CC @unpack8866)

migor7, 2022-05-04:
> This is not supported by the device, see #12076 (CC @unpack8866)

If I restore version 1.25.1-0, tamper works fine and responds to contact opening

unpack8866, 2022-05-05:
> > This is not supported by the device, see #12076 (CC @unpack8866)
> 
> If I restore version 1.25.1-0, tamper works fine and responds to contact opening

This is my device pcb board; no tamper on it
[image]
[image]

migor7, 2022-05-05:
> > > This is not supported by the device, see #12076 (CC @unpack8866)
> > 
> > 
> > If I restore version 1.25.1-0, tamper works fine and responds to contact opening
> 
> This is my device pcb board; no tamper on it [image] [image]

[image]
I have this version. there is a tamper button

unpack8866, 2022-05-05:
maybe two hardware versions；Does your version have a voltage display?

migor7, 2022-05-05:
> maybe two hardware versions；Does your version have a voltage display?

my version without voltage display

Koenkk, 2022-05-05:
Can you both provide the data/database.db of this device?

migor7, 2022-05-05:
> Can you both provide the data/database.db of this device?

{"id":28,"type":"EndDevice","ieeeAddr":"0x000d6f0010dceb04","nwkAddr":24429,"manufId":4619,"manufName":"GS","powerSource":"Battery","modelId":"SMHM-I1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,65526],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SMHM-I1","manufacturerName":"GS","powerSource":3,"zclVersion":1,"appVersion":22,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.7","swBuildId":"1.1.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00194b4faa","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194b4faa","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":22,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.7","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1045631717},"lastSeen":1651693804349,"defaultSendRequestWhen":"immediate"}

unpack8866, 2022-05-05:
{"id":5,"type":"EndDevice","ieeeAddr":"0x00124b0014afcd96","nwkAddr":61936,"manufId":0,"manufName":"Feibit","powerSource":"Battery","modelId":"SMHM-I1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,1280],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"SMHM-I1","manufacturerName":"Feibit","powerSource":3,"zclVersion":1,"appVersion":3,"stackVersion":4,"hwVersion":1,"dateCode":"08/09/2021 19:57","swBuildId":"08/09/2021 19:57"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0022ab7114","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":31}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7114","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":3,"stackVersion":4,"hwVersion":1,"dateCode":"08/09/2021 19:57","swBuildId":"08/09/2021 19:57","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1045631717},"lastSeen":1651737303857,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-05-05:
So seems some version of this device have the tamper switch and some don't (you can see the `dateCode` differs). I would actually prefer to add the tamper back and make a note on the device page (this keeps things simple).

unpack8866, 2022-05-06:
> So seems some version of this device have the tamper switch and some don't (you can see the `dateCode` differs). I would actually prefer to add the tamper back and make a note on the device page (this keeps things simple).

Yes, you're right!

Koenkk, 2022-05-06:
Added it back

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-06 by Koenkk