Koenkk/zigbee2mqtt #12378: upgrade to version 1.25.1 changed all TS011F_plug_1 to TS011F_plug_3

pgolis, 2022-05-04:
### What happened?

after upgrade to version 1.25.1, all my devices TS011F_plug_1 have been changed to TS011F_plug_3 and they do not show the power usage, pooling is also not working. 

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20211217

### Adapter

zStack3x0

oncetemp, 2022-05-11:
why changed "TZ3000_gjnozsaz" is plug_1 to plug_3???
ts011f_plug_1 is properly works with power monitoring (not polling mode)
How do I downgrade haos addon z2m 1.25.1 to 1.25 and How to set it to plug_1?

zaludjakub, 2022-05-11:
Still not working for me with the edge version HA addon. I have orange plugs TS011F_TZ3000_gjnozsaz which worked great before 1.25.1, but now they show as TS011F_plug_3 and the power reporting is broken. Reverting to 1.25.0 fixed the issue.

Koenkk, 2022-05-11:
Pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-11 by Koenkk