Koenkk/zigbee2mqtt #12379: [Feature request]: Don't allow friendly names to include control characters

Koenkk, 2022-05-04:
### Is your feature request related to a problem? Please describe

> The MQTT specification specifies that topics "SHOULD NOT" include control- or non-characters and that a server or client MAY close the connection if such characters are present.

### Describe the solution you'd like

Don't allow z2m to start when control characters are in the friendly name

### Describe alternatives you've considered

-

### Additional context

See https://github.com/home-assistant/core/pull/71263

Koenkk, 2022-05-04:
@[emontnemery](https://github.com/emontnemery) implemented

closed 2022-05-04 by Koenkk

emontnemery, 2022-05-05:
That's awesome @Koenkk!
Which release of zigbee2mqtt will this be in?

Koenkk, 2022-05-05:
@emontnemery the one that will be released on 1 June