Koenkk/zigbee2mqtt #12381: undefined symbol: _ZN2v812api_internal12ToLocalEmptyEv

sdfg2, 2022-05-04:
### What happened?

```
May 04 21:26:02 heracles systemd[1]: Started zigbee2mqtt.
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:info  2022-05-04 21:26:03: Logging to console and directory: '/var/lib/zigbee2mqtt/log/2022-05-04.21-26-02' filename: log.txt
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:info  2022-05-04 21:26:03: Starting Zigbee2MQTT version 1.25.1-dev (commit #unknown)
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:info  2022-05-04 21:26:03: Starting zigbee-herdsman (0.14.27)
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:error 2022-05-04 21:26:03: Error while starting zigbee-herdsman
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:error 2022-05-04 21:26:03: Failed to start zigbee
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:error 2022-05-04 21:26:03: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:error 2022-05-04 21:26:03: Exiting...
May 04 21:26:03 heracles node[2609433]: Zigbee2MQTT:error 2022-05-04 21:26:03: Error: /usr/lib/node_modules/zigbee2mqtt/node_modules/@serialport/bindings/build/Release/bindings.node: undefined symbol: _ZN2v812api_internal12ToLocalEmptyEv
May 04 21:26:03 heracles node[2609433]:     at Object.Module._extensions..node (internal/modules/cjs/loader.js:1057:18) 
May 04 21:26:03 heracles node[2609433]:     at Module.load (internal/modules/cjs/loader.js:863:32)
May 04 21:26:03 heracles node[2609433]:     at Function.Module._load (internal/modules/cjs/loader.js:708:14)
May 04 21:26:03 heracles node[2609433]:     at Module.require (internal/modules/cjs/loader.js:887:19)
May 04 21:26:03 heracles node[2609433]:     at require (internal/modules/cjs/helpers.js:74:18)
May 04 21:26:03 heracles node[2609433]:     at bindings (/usr/lib/node_modules/zigbee2mqtt/node_modules/bindings/bindings.js:112:48)
May 04 21:26:03 heracles node[2609433]:     at Object.<anonymous> (/usr/lib/node_modules/zigbee2mqtt/node_modules/@serialport/bindings/lib/linux.js:2:36)
May 04 21:26:03 heracles node[2609433]:     at Module._compile (internal/modules/cjs/loader.js:999:30)
May 04 21:26:03 heracles node[2609433]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
May 04 21:26:03 heracles node[2609433]:     at Module.load (internal/modules/cjs/loader.js:863:32)
May 04 21:26:03 heracles systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
May 04 21:26:03 heracles systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May 04 21:26:03 heracles systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
May 04 21:26:03 heracles systemd[1]: Stopped zigbee2mqtt.
May 04 21:26:03 heracles systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
May 04 21:26:03 heracles systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May 04 21:26:03 heracles systemd[1]: Failed to start zigbee2mqtt.
```

### What did you expect to happen?

zigbee2mqtt starts

### How to reproduce it (minimal and precise)

Just try and start the service.

### Zigbee2MQTT version

1.25.1.r10.g9b594a4c-1

### Adapter firmware version

Can't get to the front end

### Adapter

CC2652R1

### Debug log

```
May 04 21:33:07 heracles systemd[1]: Started zigbee2mqtt.
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:debug 2022-05-04 21:33:08: Loaded state from file /var/lib/zigbee2mqtt/state.json
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:info  2022-05-04 21:33:08: Logging to console and directory: '/var/lib/zigbee2mqtt/log/2022-05-04.21-33-08' filename: log.txt
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:debug 2022-05-04 21:33:08: Removing old log directory '/var/lib/zigbee2mqtt/log/2022-05-04.21-16-06'
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:info  2022-05-04 21:33:08: Starting Zigbee2MQTT version 1.25.1-dev (commit #unknown)
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:info  2022-05-04 21:33:08: Starting zigbee-herdsman (0.14.27)
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:debug 2022-05-04 21:33:08: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":16,"delay":null,"disableLED":false},"backupPath":"/var/lib/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/var/lib/zigbee2mqtt/database.db.backup","databasePath":"/var/lib/zigbee2mqtt/database.db","network":{"channelList":[26],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0","rtscts":false}}'
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:error 2022-05-04 21:33:08: Error while starting zigbee-herdsman
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:error 2022-05-04 21:33:08: Failed to start zigbee
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:error 2022-05-04 21:33:08: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:error 2022-05-04 21:33:08: Exiting...
May 04 21:33:08 heracles node[2636598]: Zigbee2MQTT:error 2022-05-04 21:33:08: Error: /usr/lib/node_modules/zigbee2mqtt/node_modules/@serialport/bindings/build/Release/bindings.node: undefined symbol: _ZN2v812api_internal12ToLocalEmptyEv
May 04 21:33:08 heracles node[2636598]:     at Object.Module._extensions..node (internal/modules/cjs/loader.js:1057:18)
May 04 21:33:08 heracles node[2636598]:     at Module.load (internal/modules/cjs/loader.js:863:32)
May 04 21:33:08 heracles node[2636598]:     at Function.Module._load (internal/modules/cjs/loader.js:708:14)
May 04 21:33:08 heracles node[2636598]:     at Module.require (internal/modules/cjs/loader.js:887:19)
May 04 21:33:08 heracles node[2636598]:     at require (internal/modules/cjs/helpers.js:74:18)
May 04 21:33:08 heracles node[2636598]:     at bindings (/usr/lib/node_modules/zigbee2mqtt/node_modules/bindings/bindings.js:112:48)
May 04 21:33:08 heracles node[2636598]:     at Object.<anonymous> (/usr/lib/node_modules/zigbee2mqtt/node_modules/@serialport/bindings/lib/linux.js:2:36)
May 04 21:33:08 heracles node[2636598]:     at Module._compile (internal/modules/cjs/loader.js:999:30)
May 04 21:33:08 heracles node[2636598]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
May 04 21:33:08 heracles node[2636598]:     at Module.load (internal/modules/cjs/loader.js:863:32)
May 04 21:33:08 heracles systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
May 04 21:33:08 heracles systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May 04 21:33:08 heracles systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
May 04 21:33:08 heracles systemd[1]: Stopped zigbee2mqtt.
May 04 21:33:08 heracles systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
May 04 21:33:08 heracles systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May 04 21:33:08 heracles systemd[1]: Failed to start zigbee2mqtt.
```

Koenkk, 2022-05-05:
Try updating nodejs on your system, execute `npm ci` in the z2m directory and start again.

sdfg2, 2022-05-05:
Ok, problem solved.  It was that nodejs was actually up-to-date, I had to get nvm sorted with lts-erbium.

closed 2022-05-05 by sdfg2