Koenkk/zigbee2mqtt #12382: Hue Dimmer Switch Drains Battery Badly

JRJSmith, 2022-05-04:
### What happened?

I have a Hue Dimmer Switch, which I'm using to test using Philips Hue devices through Zigbee2MQTT and Mosquitto on Home Assistant rather than using the Philips Hue hub.

I have finally managed to get an HA Automation working to use the buttons on the Philips Hue Dimmer Switch as a Trigger to cause an Action on a Philips Hue B22 Lamp without using the Philips Hue hub. Great except that the battery on the Philips Hue Dimmer Switch has dropped from 65% to 1.2% in less than 24 hours.

This issue appears to be the same as Issue #7509, which has been Closed - supposedly, a firmware upgrade to the Conbee II Zigbee USB stick. I'm using a Sonoff Zigbee 3.0 USB Dongle Plus device as a Zigbee Coordinator and four Sonoff Zigbee 3.0 USB Dongle Plus devices as Zigbee Routers around the house.

Is anyone aware of a firmware upgrade for the Sonoff devices that is similar to that for the Conbee II device that resolves the issue with the extreme battery drain of the Philips Hue Dimmer Switch, please?

I am guessing that the issue is with the frequency of the polls made from the Zigbee Coordinator/Routers to the Philips Hue Dimmer Switch. Can anyone confirm that, please?

Many thanks and best regards
Joe.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

JRJSmith, 2022-06-04:
Well, I have to confess that it is extremely disappointing to have my issue completely ignored!

It seems only right that I should completely ignore the request to sponsor Koenkk.

KoalaWerewolf, 2022-09-21:
I have five Hue Dimmer Switches, two gen 1 and three gen 2. Very recently one of the gen 1 dimmer switches started consuming the battery quickly. This makes me think that it's an issue with the switch itself. 
I will try to make a new pairing of the switch to my controller and see if that might help.

JonasHess, 2022-11-14:
I have the same problem with my v1 Hue Dimmer Switches. It burns through a fesh battery in just a few days. I am using multiple Sonoff Zigbee 3.0 in the network.

Lapeno89, 2023-02-20:
I have the same problem

RevLaw, 2023-02-27:
I've got the same issue. I have three v1 Switches and one of them is having the issue describe here. 
I've tried to change the Zigbee Channel, Location of the Switch and Battery manufacturer. The issue persist. My last resort was buying a new one. Seems like a hardware issue.

ThomasA94, 2023-03-05:
I have the same issue with the same hardware. The battery lost 70% battery in < 1 week.

DocSneider, 2023-03-11:
Same here,

I think it may could be related to an OTA request? But thats just a guess. It definitely started **after** switching from the direct connection with the lamp to zigbee2mqtt.

I run zigbee2mqtt inside a Archlinux-LXC-container, the CC2531 is passed through from the proxmox host.

I think the problems were starting, when I first tried to update the FW over the air. Right after this first try I had to completely reset it, with the battery disconnected over night. 
Then I somehow managed to update it.

But take a look inside my logs - it seems that at some point it constantly requested for an OTA update:

```
Mar 04 06:13:11 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:11: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 06:13:11 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:11: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 06:13:20 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:20: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 06:13:20 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:20: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 06:13:30 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:30: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 06:13:30 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:30: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 06:13:40 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:40: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 06:13:40 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:40: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 06:13:50 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:50: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 06:13:50 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 06:13:50: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 09:17:45 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:17:45: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 09:17:45 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:17:45: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 09:17:46 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:17:46: Device 'hue-dimmer-switch-v1' announced itself
Mar 04 09:17:46 zigbee2mqtt node[81]: Zigbee2MQTT:info  2023-03-04 09:17:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"hue-dimmer-switch-v1","ieee_address":"xxx"},"type":"device_ann>
Mar 04 09:17:48 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:17:48: Retrieving state of 'hue-dimmer-switch-v1' after reconnect
Mar 04 09:17:54 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:17:54: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 09:17:54 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:17:54: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 09:18:04 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:18:04: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 09:18:04 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:18:04: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 09:18:14 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:18:14: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 09:18:14 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:18:14: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 09:18:24 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:18:24: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 09:18:24 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:18:24: Device 'hue-dimmer-switch-v1' requested OTA
Mar 04 09:42:13 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 09:42:13: Received Zigbee message from 'hue-dimmer-switch-v1', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 2 wi>
Mar 04 09:42:13 zigbee2mqtt node[81]: Zigbee2MQTT:info  2023-03-04 09:42:13: MQTT publish: topic 'zigbee2mqtt/hue-dimmer-switch-v1', payload '{"battery":100,"brightness":223,"counter":1,"linkquality":0,"update":{"installed_version":11073>
Mar 04 11:53:12 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 11:53:12: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
[... cut ...]
Mar 04 12:54:58 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 12:54:58: Received Zigbee message from 'hue-dimmer-switch-v1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"im>
Mar 04 12:54:58 zigbee2mqtt node[81]: Zigbee2MQTT:debug 2023-03-04 12:54:58: Device 'hue-dimmer-switch-v1' requested OTA
```

Maybe this kind of constant traffic is draining the battery ???

Maybe the other ones could also check their logs, to see if they may have the same problem?

Jupp1008, 2023-04-17:
I have the same issue since a few weeks, with 5 of my 10 Hue Dimmer Switches. 
New Batteries drain in just one day. 
I am using the native Hue App and iConnect Hue. 
I'd like to share the zigbee log, but have no possibilities to read out the log.

septeven, 2023-05-03:
Same here, it's been almost one or two years that I have noticed this issue, initially I thought it was due to the quality of the batteries, but no. All my V2 US dimmer switches (RWL020) - I used to live in Canada before moving to Europe - are experiencing the severe battery drain (from 100% to 1-2% in only a few days, may be a month). No problem with the V2 EU dimmer switches (RWL021).

Any chance to know what the heck is going on with those switches ? And to those who's been experiencing the issue, do you all live in US/CA (or do you use a RWL020 dimmer switch)

closed 2023-06-10 by github-actions[bot]

bjornarfjelldal, 2023-07-06:
Not stale - same issue here. And other places, like #12545 and https://community.home-assistant.io/t/battery-drain-philips-hue-wall-switch-modules/518501/4

djashjones, 2023-07-07:
I've come from Hubitat and back to HA. In the logs, my V1 dimmers keep asking for a update every 2 hours, which will have an impact on battery life considering they are obsolete now.

On hubitat, they last less than a year (depending on usage) but can easily last 2-3 years on Hue.

gzeel, 2023-09-07:
I have the same issue suddenly with all of my Hue Dimmer switches. Putting in a new battery helps for a day or 2, but after a few days there dead again.
Hue dimmer switch v1 and v2.

deadrubberboy, 2023-10-03:
same issue here.

paarak, 2023-10-13:
Hue dimmer switch RWL021 - 1.1.28573 
Radio type deconz
from 100% to 70% 24 hours

popy2k14, 2023-11-03:
Exact same issue on one of 24x RWL021 hue dimmer switches.
At first i also thought the batterys are bad.
But it's the same on this particular switch with any battery.

Would be nice if there is an solution/workaround.

EnricoGiordano, 2023-12-07:
Same issue with "Philips 324131092621" (Hue Dimmer Switch).

machielkreiken, 2023-12-27:
Same here. On one of the 5 dimmers I have at home, battery wens down from 100 to 0 in less then a month. All other dimmers are fine.

simoncraddock, 2024-04-09:
Still ongoing. Fitted new batteries to 3 switches a month ago today they are almost dead and these are very very low use

simoncraddock, 2024-04-09:
> Same to me **regards only 1 of my in total 4 devices** of model "324131092621" (aka Philips Hue dimmer switch)!
> 
> Funny is, that - thanks OTA via Zigbee2MQTT - they all have the same firmware (regards date and version). Also I don't think that it is about the connected device as I joined it to different repeating devices (also once directly to the coordinator).
> 
> Easiest option would be, that there is a short circuit what drains the battery much quicker than usually. But since the device is still fully functional, I would be very surprised if this option would come true!
> 
> (Off-topic: Think that I fully move to Ikea 4-Way switches - as named STYRBAR - who are really reliable and better stay connected than Philips switches)

I've decided to move all my Hue Remotes to ZHA.
Work just as well.
Hopefully no more battery drain

renslinden, 2024-07-04:
Experience the same issue. All my Hue Switches (4 pc) runs out of battery within 1 month.
No wonder that these devices disappeared from the Devices list.... (interview failed)

```
Warning 2024-07-04 09:46:30z2m: Device 'Hue Switch - Bedroom' left the network
Info 2024-07-04 09:46:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue Switch - Bedroom","ieee_address":"0x0017880108f71c05"},"type":"device_leave"}'
Info 2024-07-04 09:46:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Switch - Bedroom', payload '{"action":null,"action_duration":null,"battery":2,"last_seen":"2024-07-04T09:46:29+02:00","linkquality":255,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
Info 2024-07-04 09:46:39z2m: Device 'Hue Switch - Bedroom' joined
Info 2024-07-04 09:46:39z2m: Starting interview of 'Hue Switch - Bedroom'
Info 2024-07-04 09:46:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue Switch - Bedroom","ieee_address":"0x0017880108f71c05"},"type":"device_joined"}'
Info 2024-07-04 09:46:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue Switch - Bedroom","ieee_address":"0x0017880108f71c05","status":"started"},"type":"device_interview"}'
Info 2024-07-04 09:46:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Switch - Bedroom', payload '{"action":null,"action_duration":null,"battery":2,"last_seen":"2024-07-04T09:46:29+02:00","linkquality":255,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
Info 2024-07-04 09:46:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Switch - Bedroom', payload '{"action":null,"action_duration":null,"battery":2,"last_seen":"2024-07-04T09:46:30+02:00","linkquality":255,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
Info 2024-07-04 09:46:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue Switch - Bedroom","ieee_address":"0x0017880108f71c05"},"type":"device_announce"}'
Error 2024-07-04 09:46:55zh:controller: Interview failed for '0x0017880108f71c05 with error 'Error: {"target":26065,"apsFrame":{"profileId":0,"clusterId":4,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":23},"responseClusterId":32772} timed out after 15000ms'
Error 2024-07-04 09:46:55z2m: Failed to interview 'Hue Switch - Bedroom', device has not successfully been paired
Info 2024-07-04 09:46:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue Switch - Bedroom","ieee_address":"0x0017880108f71c05","status":"failed"},"type":"device_interview"}'
```

Using the SkyConnect (Connect ZBT-1) coordinator. 

And by the way, why is this topic closed again? Is no one able to do a deep dive?

popy2k14, 2024-07-04:
In my case i replaced the battery to an duracell branded (not from amazon) but from an local distributor and re-paired the remote near the coordinator. Since then the issue is gone on this single remote.

PS.: dont know if the battery swap or re-pair solved the issue for me

Hope this helps.

KoalaWerewolf, 2024-07-04:
For me the actual drain is gone. Batteries last months to a year. However, the reporting is all over the place. The reported battery level can jump between a few percent and 80% and can stay at 2% for months while still operating properly.

rusitschka, 2024-07-11:
> For me the actual drain is gone. Batteries last months to a year. However, the reporting is all over the place. The reported battery level can jump between a few percent and 80% and can stay at 2% for months while still operating properly. 

Same here

fti7, 2024-10-19:
Same here. Why is this issue closed?

daath, 2024-12-07:
Z2M here - I had one dimmer switch drain it's battery VERY quickly. Today 3 of my other dimmer switches seem to do the same :/

popy2k14, 2024-12-07:
I have now sold my 26x V1 switches and replaced them with V2. Also had this annoying issue sometimes. 
Will see how it goes with the V2 ones...

daath, 2024-12-08:
> Z2M here - I had one dimmer switch (324131092621) drain it's battery VERY quickly. Today 3 of my other dimmer switches seem to do the same :/

It jumps a bit. One of the dimmer switches went from 4% battery to 22% during the night ;P

nicopeet911, 2025-02-01:
Same here, still solution anyone?

popy2k14, 2025-02-01:
Ironically. I swapped all my 26x dimmers from V1 (which drained sometimes) to V2 in the hope that V2 won't drain.
Sadly, one V2 is also draining badly.

I also have Z2M with SONOFF Zigbee 3.0 USB Dongle Plus-P stick.

nicopeet911, 2025-02-02:
I can say I solved it for my situation. I changed at zigbee2mqtt under settings --> OTA updates tab --> OTA update interval from 60 to 1440 and activated the "disable automatic update check" at the top. 

My v1 and v2 hue dimmers that were 19% and 55% after 1 day from 100% now both fully recovered to 100% battery. Problem seems resolved by these change of settings

simoncraddock, 2025-02-02:
Just to add to this discussion.

Over a year ago I migrated all my Hue dimmers to Z2MQTT and for weeks struggled with reported battery life.

I then moved them to ZHA and whilst battery reporting seemed to stabilise they kept going offline, so eventually I plugged the Hue hub back in and paired them all back and it's been like that for months and I kinda forgot about them.

About 4 days ago I noticed 3 out of 10 were reporting very low battery and thought I'll do them at the weekend, no immediate rush.

So this morning I've gone into the Hue app to remind myself which needed doing and they are all showing 65-100% battery life.

So if anything I'd say the issue could be hardware related not software.

popy2k14, 2025-02-02:
I will try to replace the one V2 which already drained two cells in short time. Will report back.

simoncraddock, 2025-02-02:
And to prove my theory, and hour later 1 of the 3 dimmers is reporting critical.

Even on the Hue Hub/Hue app they are unpredictable

popy2k14, 2025-02-02:
Mine drained a cell in about ~21days.
I'll replaced it now again, repaired the device and see how it goes.
If it wil drain again in such a short time, i will let i replace by amazon (where i have bougth it).
Keep you updated.

paarak, 2025-02-02:
I switched from the Philips Remote v1 RWL021 about a year ago when v1 started draining the battery within 1 to 3 days, to the Philips Remote v2 RWL022. After about a year of daily use, it's still at 100% battery. It's probably not a solution, just some information that there's little chance a solution will come since this thread was started (May 4, 2022). I paid around 18 euros on sale here in Norway in 2024, and I see it's still around the same price today :) Wishing everyone a great day!

[image]

simoncraddock, 2025-02-02:
Bit expensive to start replacing almost a dozen switches. They can stay on the Hue hub

popy2k14, 2025-02-05:
Good point, so it seems hardware (hue switch or battery) related. As pointed out above. I had Duracell batterys from Amazon which where fake and not good quality. 
Bought than from an official distributer near me, and they lasted much longer.

ayanguas77, 2025-03-22:
Is TEMPERATURE causing the erratic level behaviour?

I have several V1 RWL021, connected to the HUE Bridge. I have recently deleted them and re-added for an undergoing migration to HUE API V2.

I’m seeing exactly the same issue (erratic battery level from 80% to 15%, or so) with Home Assistant. And Today I have seen a very strange pattern.

Surprisingly, most of my HUE dimmer switches battery level curve **correlate with the daily variation of each room temperature**. This correlation may suggest that **there could be some temperature cause** for the software/hardware measuring of the battery level data. The temperature is measured by a HUE motion sensor in each room, so it would be too strange that the correlation would be “inside” the same platform HUE, but different devices dimmer switches and motion sensors.

I have to say that for some other Dimmer Switches the correlation is not as obvious, but still daily cicles. I could guess that in the others the battery is newer and behaves more powerful.

Below you have my last week's graph of the battery level of a HUE dimmer switch, starting Monday evening when I added it,  and the corresponding temperature in the same room from a HUE motion sensor, both taken from Home Assistant. The correlation seems obvious.

Other Dimmer Switch, with a less closed correlation, had a very low battery this afternoon, down to 5%, I disabled the Automatic Software Update from the HUE App. As mentioned in other posts, the continuous polling for checking software updates or whatever, could drain the battery.  The thing is that after one hour or so, the battery level rose up to 70%.

In the end, it looks that the measured level is the erratic malfunction, but not the real battery level, as the Dimmer Switch continues working without problems, even if its led light is orange/red sometimes when pressing buttons, instead of green, but the button pressed action always works.

[image]

[image]