Koenkk/zigbee2mqtt #12385: Problem during the installation of my Zigbee2MQTT server

karimtaz5962, 2022-05-05:
### What happened?

Hello,

I contact you because I have a problem during the installation
Following a handling problem when setting DomoticZ, I corrupted the operation of my Zigbee2mqtt server, so I had to start over with a clean installation
I use a Rpi 3B+ with raspbian 10 buster, my mqtt server is well installed and functional
the port of my sonoff 3.0 dongle is well fixed and entered in my configuration.yml file, and it is flashed with the latest firmware
in the previous installations I was running with mqtt without user or password and on this installation I wanted to start a user and password
my concern comes after the launch of npm start
this is what i end up with in the terminal

`pi@raspberrypi:~ $ sudo apt-get install -y nodejs npm git make g++ gcc
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
g++ est déjà la version la plus récente (4:8.3.0-1+rpi2).
g++ passé en « installé manuellement ».
gcc est déjà la version la plus récente (4:8.3.0-1+rpi2).
gcc passé en « installé manuellement ».
git est déjà la version la plus récente (1:2.20.1-2+deb10u3).
make est déjà la version la plus récente (4.2.1-1.2).
make passé en « installé manuellement ».
nodejs est déjà la version la plus récente (14.19.2-deb-1nodesource1).
Certains paquets ne peuvent être installés. Ceci peut signifier
que vous avez demandé l'impossible, ou bien, si vous utilisez
la distribution unstable, que certains paquets n'ont pas encore
été créés ou ne sont pas sortis d'Incoming.
L'information suivante devrait vous aider à résoudre la situation :

Les paquets suivants contiennent des dépendances non satisfaites :
 nodejs : Est en conflit avec: npm
 npm : Dépend: node-gyp (>= 3.6.2~) mais ne sera pas installé
E: Impossible de corriger les problèmes, des paquets défectueux sont en mode « garder en l'état ».
pi@raspberrypi:~ $ node --version
v14.19.2
pi@raspberrypi:~ $ npm --version
6.14.17
pi@raspberrypi:~ $ git clone https://github.com/Koenkk/zigbee2mqtt.git
Clonage dans 'zigbee2mqtt'...
remote: Enumerating objects: 21711, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 21711 (delta 5), reused 11 (delta 5), pack-reused 21693
Réception d'objets: 100% (21711/21711), 101.75 MiB | 4.14 MiB/s, fait.
Résolution des deltas: 100% (16266/16266), fait.
pi@raspberrypi:~ $ sudo mv zigbee2mqtt /opt/zigbee2mqtt
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm ci

> @serialport/bindings@9.2.8 install /opt/zigbee2mqtt/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=14.19.2 runtime=node arch=arm libc= platform=linux)
make : on entre dans le répertoire « /opt/zigbee2mqtt/node_modules/@serialport/bindings/build »
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:78:69: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to                                                                                                  ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
                                                                     ^~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:135:71: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to                                                                                                  ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
   uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
                                                                       ^~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:175:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to                                                                                                  ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
   uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
                                                                      ^~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:215:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to                                                                                                  ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
[... 111 lines cut ...]
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:21)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1153) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promi                                                                                                 se rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (reject                                                                                                 ion id: 1)
(node:1153) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`

I don't see where it could come from
could you help me
thank you

### Zigbee2MQTT version

?

### Adapter firmware version

?

### Adapter

sonoff zstack 3.0

pierrepinon, 2022-05-26:
I have the same problem with fresh installation

closed 2022-07-03 by github-actions[bot]

rbswift, 2022-09-02:
Were you able to solve this? I think I have the same problem. Thanks