Koenkk/zigbee2mqtt #12386: Zigbee Groups defined in Z2M not reapplied if light is re-added in Z2M

ahd71, 2022-05-05:
### What happened?

Zigbee group shows members that isn't members according to the device - only according to z2m.

Occasional, but rare, some of my lights gets unavailable and when they don't reconnect via a power cycle I reset them and they gracefully rejoins the network and as Z2M remembers the device the name which is the most important for me is kept and they smoothly works again in HA [much better than it works in ZHA when i used that in the past]

However, what's not working and I believe should be easy to resolve in Z2M is that if they were part of a Z2M/Zigbee group that group configuration should be reapplied after the device is rejoining the network.

### What did you expect to happen?

I exect that if Z2M has a group where the light is member that group configuration is reapplied on the light in a [re]configuration cycle.

(Or if it is intentional that it doesn't keep the group configuration that the groups member list gets updated in z2m to reflect that (light is removed. From a UX point of view it's better if they are kept and reconfigured though)

### How to reproduce it (minimal and precise)

1. Add a device to the network
2. Add that device to a group
3. Reset that device and let it rejoin the network
4. Verify that the group still shows the light as a member [it does now]
5. Verify that you can toggle that light group [it does work to that extent that the UI shows the light as controlled, but in reallity the light is not turning on/off - this is the issue!]
6. Add the light again [it's already there but after doing it again everything works] - my assumption is that this step could be automatically done when a light joins the network if Z2M has a group with that member in the db.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

Koenkk, 2022-05-05:
Could you provide the debug log when you reset your bulbs? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

ahd71, 2022-05-05:
First of all, thanks for quick reply!

1. Created a new group 'office_ceiling_zg'
2. Added three devices (office_ceiling_1, office_ceiling_2, office_ceiling_3)
3. Tested that the groups worked to turn on/off the lights successfully
4. Hardware reset IKEA Tradfri light 'office_ceiling_3' at 19:00:47: Device 'office_ceiling_3' announced itself
5. Tested the group again at 19:01:36: Publishing 'set' 'state' to 'office_ceiling_zg' (not successful, light does not turn off/on)

`debug 2022-05-05 19:00:47: Device 'office_ceiling_3' announced itself
info  2022-05-05 19:00:47: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"office_ceiling_3","ieee_address":"0x50325ffffebc67dc"},"type":"device_announce"}'
debug 2022-05-05 19:00:49: Retrieving state of 'office_ceiling_3' after reconnect
info  2022-05-05 19:00:51: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:00:51+02:00","linkquality":72,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
info  2022-05-05 19:00:54: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:00:54+02:00","linkquality":72,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
info  2022-05-05 19:00:54: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:00:54+02:00","linkquality":72,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
debug 2022-05-05 19:00:54: Device 'office_ceiling_3' announced itself
info  2022-05-05 19:00:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"office_ceiling_3","ieee_address":"0x50325ffffebc67dc"},"type":"device_announce"}'
info  2022-05-05 19:00:55: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:00:55+02:00","linkquality":72,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
debug 2022-05-05 19:00:55: Device 'office_ceiling_3' announced itself
info  2022-05-05 19:00:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"office_ceiling_3","ieee_address":"0x50325ffffebc67dc"},"type":"device_announce"}'
debug 2022-05-05 19:00:57: Retrieving state of 'office_ceiling_3' after reconnect
debug 2022-05-05 19:00:57: Received Zigbee message from 'office_ceiling_3', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-05-05 19:00:57: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:00:57+02:00","linkquality":72,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
info  2022-05-05 19:00:57: MQTT publish: topic 'zigbee2mqtt/office_ceiling_zg', payload '{"state":"ON"}'
info  2022-05-05 19:00:58: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:00:57+02:00","linkquality":72,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
debug 2022-05-05 19:01:01: Received Zigbee message from 'office_ceiling_3', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-05-05 19:01:01: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:01:01+02:00","linkquality":102,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
debug 2022-05-05 19:01:36: Received MQTT message on 'zigbee2mqtt/office_ceiling_zg/set' with data '{"state": "OFF"}'
debug 2022-05-05 19:01:36: Publishing 'set' 'state' to 'office_ceiling_zg'
info  2022-05-05 19:01:37: MQTT publish: topic 'zigbee2mqtt/office_ceiling_1', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"last_seen":"2022-05-05T18:57:36+02:00","linkquality":72,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":null}'
info  2022-05-05 19:01:37: MQTT publish: topic 'zigbee2mqtt/office_ceiling_zg', payload '{"state":"OFF"}'
info  2022-05-05 19:01:37: MQTT publish: topic 'zigbee2mqtt/office_ceiling_2', payload '{"brightness":216,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"last_seen":"2022-05-05T18:57:39+02:00","linkquality":108,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":null}'
info  2022-05-05 19:01:37: MQTT publish: topic 'zigbee2mqtt/office_ceiling_3', payload '{"brightness":234,"color":{"x":0.4657,"y":0.4118},"color_mode":"color_temp","color_temp":380,"last_seen":"2022-05-05T19:01:01+02:00","linkquality":102,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":null}'
debug 2022-05-05 19:01:43: Received MQTT message on 'zigbee2mqtt/office_ceiling_zg/set' with data '{"state": "ON"}'
`

ahd71, 2022-05-05:
Also captured the other log (which is a bit more verbose but i think i selected the most interesting part

Interesting parts?

`2022-05-05T17:00:51.112Z zigbee-herdsman:controller:log Device '0x50325ffffebc67dc' joined
2022-05-05T17:00:51.113Z zigbee-herdsman:controller:log Device '0x50325ffffebc67dc' accepted by handler
2022-05-05T17:00:51.113Z zigbee-herdsman:controller:log Device '0x50325ffffebc67dc' is already in database with different networkAddress, updating networkAddress
2022-05-05T17:00:51.123Z zigbee-herdsman:controller:log Not interviewing '0x50325ffffebc67dc', completed 'true', in progress 'false'
2022-05-05T17:00:53.734Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x50325ffffebc67dc:63532,205,1)
2022-05-05T17:00:53.735Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 63532
`

Full log

`2022-05-05T17:00:44.968Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-05-05T17:00:47.160Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,44,248,44,248,220,103,188,254,255,95,50,80,142,60]
2022-05-05T17:00:47.160Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,44,248,44,248,220,103,188,254,255,95,50,80,142,60]
2022-05-05T17:00:47.161Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [44,248,44,248,220,103,188,254,255,95,50,80,142] - 60
2022-05-05T17:00:47.161Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":63532,"nwkaddr":63532,"ieeeaddr":"0x50325ffffebc67dc","capabilities":142}
2022-05-05T17:00:47.161Z zigbee-herdsman:controller:log Device announce '0x50325ffffebc67dc'
State changed noted by my custom extention {
  entity: Device {
    zh: Device {
      _events: [Object: null prototype] {},
      _eventsCount: 0,
      _maxListeners: undefined,
      ID: 243,
      _type: 'Router',
      _ieeeAddr: '0x50325ffffebc67dc',
      _networkAddress: 63532,
      _manufacturerID: 4476,
      _endpoints: [Array],
      _manufacturerName: 'IKEA of Sweden',
      _powerSource: 'Mains (single phase)',
      _modelID: 'TRADFRI bulb GU10 CWS 345lm',
      _applicationVersion: 80,
      _stackVersion: 101,
      _zclVersion: 3,
      _hardwareVersion: 1,
      _dateCode: '2019062886LEL08N',
      _softwareBuildID: '1.0.021',
      _interviewCompleted: true,
      _interviewing: false,
      _skipDefaultResponse: false,
      _skipTimeResponse: false,
      meta: [Object],
      _lastSeen: 1651770047161,
      _defaultSendRequestWhen: 'immediate',
      _linkquality: 72,
      [Symbol(kCapture)]: true
    },
    _definition: {
      exposes: [Array],
      fromZigbee: [Array],
      toZigbee: [Array],
      configure: [AsyncFunction (anonymous)],
      ota: [Object],
      onEvent: [AsyncFunction: bulbOnEvent],
      zigbeeModel: [Array],
      model: 'LED1923R5',
      vendor: 'IKEA',
      description: 'TRADFRI LED bulb GU10 345 lumen, dimmable, white spectrum, color spectrum',
      meta: undefined,
      options: [Array]
    },
    _definitionModelID: 'TRADFRI bulb GU10 CWS 345lm'
  },
  from: {
    color_mode: 'color_temp',
    state: 'ON',
[... 549 lines cut ...]
      zigbeeModel: [Array],
      model: 'BASICZBR3',
      vendor: 'SONOFF',
      description: 'Zigbee smart switch',
      meta: undefined,
      options: []
    },
    _definitionModelID: 'BASICZBR3'
  },
  from: {
    state: 'OFF',
    color_mode: 'xy',
    color: { x: 0.61, y: 0.376, hue: 13, saturation: 100, h: 13, s: 100 },
    color_temp: 454
  },
  to: {
    state: 'OFF',
    color_mode: 'xy',
    color: { x: 0.61, y: 0.376, hue: 13, saturation: 100, h: 13, s: 100 },
    color_temp: 454
  },
  reason: null,
  update: { state: 'OFF' }
}
`

ahd71, 2022-05-05:
I hope that gave you some details about the issue, let me know what more to share, test or do to help you!

Koenkk, 2022-05-07:
With a hardware reset, you just meaning removing the bulbs from the mains net for some time right? I cannot reproduce this issue with my Ikea bulb (I have many and already removed them from mains a couple of time).

The bulbs should definitely not forgot their group configuration when rebooted; is there an OTA update available for your bulbs?

ahd71, 2022-05-07:
Sorry if I was not clear. I meant the "factory reset" (ikea power cycle five times in a couple of seconds).
Its rare I have to use it but sometimes it was the only way to get the lights back in the network - but then groups missing, everything else works supersmooth)

ahd71, 2022-05-07:
No OTA available for them. For this one it is on 1.0.021 but I think the behaviour has been consistent with all my bulbs that gone thru factoring reset cycle and had groups. I can verify other brands too if you want.

ahd71, 2022-05-07:
Perhaps z2m not even realize they are being reset (5 quick power cycles in short time) and just welcome them back as from a regular power outage (1 cycle of arbitrary length)

Koenkk, 2022-05-08:
@ahd71 a `deviceAnnounce` does not mean a reset, this is a `deviceLeave` + `deviceJoined`. It is also strange that you have to factory reset the bulb to get it working again, a simple power cycle should do it in case the bulb has crashed. What do you see in the debug log when sending commands to it when it has crashed?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

ahd71, 2022-05-08:
Hi. I will capture if/when it happends again. That is very rare for me but still wanted to report that the group config is missing in physical device but present in configuration as that part is easily reproducible (and probably easy to fix).