Koenkk/zigbee2mqtt #12387: [New device support]:

daantjuh44, 2022-05-05:
### Link

https://www.philips-hue.com/en-au/p/hue-smart-plug/8719514342361#specifications

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0x001788010be0cf47","nwkAddr":34174,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LOM011","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":266,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211119","swBuildId":"1.93.6","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1651739357205,"defaultSendRequestWhen":"immediate"}

### Comments

Having trouble following the guide as I have never done anything like this before.
This appears to be the Australian version of the device listed in issue [#9815](https://github.com/Koenkk/zigbee2mqtt/issues/9815)

Koenkk, 2022-05-05:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-05 by Koenkk

thetravellor, 2022-05-18:
I am also having this issue, however dev branch yields an unresponsive web UI when I switch to it, no errors in log file and the zigbee2mqtt service is polling all the devices in the log file.

Whats the timeframe for including it in the master?

Or is there a .js file I can use with the master to make it work?