Koenkk/zigbee2mqtt #12393: Xiaomi RTCGQ13LM presence detection inconsistent

nielsfaber, 2022-05-05:
### What happened?

The Xiaomi Aqara RTCGQ13LM (lumi.motion.agl04) is not reporting occupancy reliably with latest z2m firmware.
It seems to report occupancy = detected state only every ~30 mins, rather than after every movement.

### How to reproduce it (minimal and precise)

Issue occurred after updating from v1.21.
Reverting to this version also solves the problem.
There seems to be a breaking change in between.

### Zigbee2MQTT version

1.25.1-dev commit: 5c7e968

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

McGiverGim, 2022-05-05:
Can you post a debug log with movement that is not working to see if there is some message that is not being processed correctly?

Otnow, 2022-05-05:
Did you accidentally update the coordinator's firmware along with _z2m_?

If yes, then you need to re-pair the sensor.

nielsfaber, 2022-05-06:
@Otnow 
> Did you accidentally update the coordinator's firmware along with _z2m_?
> 
> If yes, then you need to re-pair the sensor.

This did the trick!
I had no idea it could be related to this.
All (other) sensors seemed to work fine after updating the firmware of the stick. This is the only sensor that failed, but only fails sometimes (most of the times). 
I will keep it in mind for next time, thanks a lot! 👍

closed 2022-05-06 by nielsfaber

Otnow, 2022-05-06:
@nielsfaber, fine 👍

Unfortunately, Aqara does not like to comply with the zigbee-standard, therefore, for the normal operation of them updated devices (_T1/E1/P1 series and others_), [special changes](https://github.com/Koenkk/zigbee2mqtt/issues/9274#issuecomment-957251683) were made to the coordinator's firmware, due to which the devices think that they are connected to their native hub Aqara and change their work logic.

In this regard, the _RTCGQ13LM_ converter was [changed](https://github.com/Koenkk/zigbee2mqtt/issues/5203#issuecomment-1006955549) (and other devices too), and information was [added](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html#adapter-firmware) to the documentation about the need to update the coordinator firmware and re-pair the sensor if it was already connected to the network before the update, because the device makes the choice of the variant of the logic of its work at the moment of pairing and remembers it until it leaves the network.

nielsfaber, 2022-05-07:
@Otnow Reading back the topic you linked and the remark in the documentation of the sensor, I realise I should have managed to figure this out without creating this issue 😅 
Clearly all was done to create awareness for the change 👍 
Thanks for clearing things up for me, now I also know why only these devices are affected (in my case) by the update.