Koenkk/zigbee2mqtt #12397: Devices seem bound in Z2M

formatBCE, 2022-05-06:
### What happened?

Device 1 (knob): https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html
Device 2 (socket): https://www.zigbee2mqtt.io/devices/ZNCZ12LM.html

Today i noticed, that knob, apart from triggering Home Assistant automation, does toggle the socket.
It happens quietly, without sending anything to MQTT /set subtopic.

Happens with several different knobs (same model).

I deleted both devices from Z2M using "Force Delete", and this behaviour was preserved up until hard devices reset.

Also, i noticed, that there's some binding for knob i  Z2M, which i cannot delete. Some other device also have bindings, so i assume, this is normal - however, here we are.

### What did you expect to happen?

I expect this not to happen. :)

### How to reproduce it (minimal and precise)

- Connect both devices to coordinator.
- Set knob to "action" mode (3-click too switch).
- Perform single click on knob.
...
- Socket does toggle.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220103

### Adapter

Slaeh's CC2652RB

nonedark2008, 2022-05-07:
@formatBCE, I had exactly the same problem. It seams knob and socket are by default placed into group ID=0 by their manufacturers. To bypass it, you can manually create a group in Z2M with different ID (ID=1, for example) and add your knob inside.

formatBCE, 2022-05-07:
@nonedark2008 thank you, will try it today!

formatBCE, 2022-05-08:
@nonedark2008 it worked, seems like

closed 2022-05-08 by formatBCE

pcmunro, 2024-02-11:
This dimmer switch has caused nothing but problems for me. Any recommendations on how to get it integrated? I can not even add it to the group.

formatBCE, 2024-02-11:
> This dimmer switch has caused nothing but problems for me. Any recommendations on how to get it integrated? I can not even add it to the group.

It works for me since then. Works good.
Direct binding to IKEA bulbs works too - the only thing that for state reporting to work I had to add it to bindi g via MQTT command.

pcmunro, 2024-02-11:
I’m going to try and flash the device and see if I have any more success.
Nothing has worked since moving from ZHA.

formatBCE, 2024-02-11:
> I’m going to try and flash the device and see if I have any more success.
> Nothing has worked since moving from ZHA.
> 
> 
> > This dimmer switch has caused nothing but problems for me. Any
> > recommendations on how to get it integrated? I can not even add it to the
> > group.
> >
> > It works for me since then. Works good.
> > Direct binding to IKEA bulbs works too - the only thing that for state
> > reporting to work I had to add it to bindi g via MQTT command.

Ahhh, ZHA...
Sorry, I don't use it. Zigbee2MQTT is better IMO.

pcmunro, 2024-02-12:
That’s what I’ve moved to and I can’t get anything to work.

formatBCE, 2024-02-12:
> That’s what I’ve moved to and I can’t get anything to work.

Huh, probably you're doing it wrong? Zigbee2MQTT is used by tens of thousands...

danonb, 2024-08-13:
Did anyone work out how to configure the Tuya smart knob to not be a member of any group?
I ask because the Tuya smart knob transmits broadcast messages on the groupID it is configured for. That being either the default groupID, or one that we configure.
I have a Zigbee network with many routers (powered devices) and these knobs flood the network with their broadcast messages. 
Each router retransmits the broadcast message, and I end up with periods of seconds where the network won't respond to any new messages from the knob or any other device.
Do you think we could write a '-1' into the groupID property of the knob? 
Or any other ideas on how to make the knob a unicast transmitter?

raveit65, 2025-07-06:
Same for me. Adding this device to a group doesn't work.
```
[2025-07-06 11:53:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{},"error":"Failed to add from group (ZCL command 0xa4c13865c95f473d/1 genGroups.add({\"groupid\":1,\"groupname\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Timeout - 10945 - 1 - 105 - 4 - 0 after 10000ms))","status":"error","transaction":"atedw-2"}'
[2025-07-06 11:53:29] error: 	z2m: Failed to add from group (ZCL command 0xa4c13865c95f473d/1 genGroups.add({"groupid":1,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 10945 - 1 - 105 - 4 - 0 after 10000ms))
```
What did i wrong?

feenes, 2025-07-15:
adding to group fails for me as well, but changing the binds does work.
Just don't forget to press a botton on the osram device to wake it up and receive the bind command