Koenkk/zigbee2mqtt #12404: Cannot pair EnOcean PTM 215Z

DennisJohnsen, 2022-05-06:
### What happened?

I had a EnOcean PTM 215Z that stopped responding, assuming i would just have to repair it and try again.
I deleted it from Zigbee2MQTT and hit the pairing sequence (top-left for +8 seconds, bottom-left + top-right right after to confirm). But nothing happened. 

I tried then replacing the battery, but the same issue. There is no mention of it in the logs or Zigbee2MQTT frontend logs.

Assuming the switch had died, I had it replaced by the store without any issues - but I have the same problem with it.

I then tried to take a few random Ikea devices, and they paired up just fine.

I'm currently lost, and I cannot just start over completely as I have 83 devices I would then have to pair again. I have currently 16 other GreenPower devices hooked up (also EnOcean PTM215Z)

This is my config in json https://pastebin.com/kvYDa6Z8

### Zigbee2MQTT version

1.25.1 commit: 3f6a1378

### Adapter firmware version

0x26690700

### Adapter

ConBee2/RaspBee2

DennisJohnsen, 2022-05-06:
Ever since i updated to 1.25.1  my EnOcean PTM 215Z are behaving so strange. In node red and Zigbee2Mqtt im getting an empty string as action.

Its causing all sorts of issues currently. Maybe its related to why i cannot pair "new" switches either.

DennisJohnsen, 2022-05-06:
Currently no one of my switches of this type is working. I have no idea whats going on currently.

Koenkk, 2022-05-07:
The action will always be empty when the Home Assistant integration is enabled: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

DennisJohnsen, 2022-05-07:
> The action will always be empty when the Home Assistant integration is enabled: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

Alright. That fixes that issue then. The option must have been enabled when i updated Zigbee2MQTT in order to try and see if that fixed my pairing issue.

DennisJohnsen, 2022-05-07:
> The action will always be empty when the Home Assistant integration is enabled: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

**Sorry, another unrelated thing i noticed after updating to the latest Zigbee2MQTT:**
I have disabled OTA automatic checks, but i'm seeing these messages in the log:
```
May 07 11:52:47 homebridge npm[7877]: Zigbee2MQTT:debug 2022-05-07 11:52:47: Device 'Motion sensor @carport' requested OTA
```

This is in the bottom of my config file.
```
ota:
  disable_automatic_update_check: true
```

DennisJohnsen, 2022-05-07:
Alright, the main issue still persists. I cannot get my EnOcean PTM 215Z (both the old and the new from the store). I don't dare to unpair another in the off chance I cannot get that paired again either.

I then just tried pairing a different device, and instantly it joined. But I did get this bug screen on the frontend:

[image]
[initialState.json (1).zip](https://github.com/Koenkk/zigbee2mqtt/files/8644864/initialState.json.1.zip)

But still, after holding all 4 buttons on the switch to reset it (10 seconds, i also tried + 20seconds just to be safe) when hitting the top left button and the two others to confirm, nothing happens.

Koenkk, 2022-05-07:
> I have disabled OTA automatic checks, but i'm seeing these messages in the log:

The option does not prevent the device asking for an OTA, it only prevents z2m checking for on OTA on the internet.

Regarding the frontend issue; please create it here; https://github.com/nurikk/zigbee2mqtt-frontend/issues

DennisJohnsen, 2022-05-07:
> The option does not prevent the device asking for an OTA, it only prevents z2m checking for on OTA on the internet.
>
> Regarding the frontend issue; please create it here; [...]

Check, and check.
Any idea on why I have pairing issues with my GreenPower devices? There is lituraly nothing showing up in the logs when i look in a terminal using `sudo journalctl -u zigbee2mqtt.service -f`

I used to be able to pair these switches without any issues.

Koenkk, 2022-05-08:
Also nothing in the herdsman debug logging?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

DennisJohnsen, 2022-05-08:
> Also nothing in the herdsman debug logging?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Im not sure im seeing anything in relations to me hitting the paring sequence buttons on the switch. But there is a lot of things going on in the logs, so im propably not the one to judge.

This log i shared was from the beginning of me permitting new devices, until i had tried all 4 pairing options described in [the switch's documentation](https://www.zigbee2mqtt.io/devices/PTM_215Z.html#pairing)

```
zigbee2MQTT:info  2022-05-08 10:24:06: MQTT publish: topic 'z2m/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"sodmd-4"}'
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 143 +6s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +6s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +9ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +2s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +2s
  zigbee-herdsman:deconz:driver query aps data indication +14ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 144 +60ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 144 srcAddr: 0xe5d5 destAddr: 0x0 profile id: 0x104 cluster id: 0x19 lqi: 255 +92ms
  zigbee-herdsman:deconz:frameParser response payload: 1,97,1,0,11,16,29,1,0,29,0,1 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":97,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4107,"imageType":285,"fileVersion":16784640},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":58837,"endpoint":11,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}' +10s
Zigbee2MQTT:debug 2022-05-08 10:24:15: Received Zigbee message from 'Lily XL @spa @lille_plantekasse', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784640,"imageType":285,"manufacturerCode":4107}' from endpoint 11 with groupID null
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +24ms
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 145 +8s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +8s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +9ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +2s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +2s
  zigbee-herdsman:deconz:driver query aps data indication +37ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 146 +59ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 146 srcAddr: 0xe5d5 destAddr: 0x0 profile id: 0x104 cluster id: 0x19 lqi: 255 +114ms
  zigbee-herdsman:deconz:frameParser response payload: 1,98,1,0,11,16,29,1,0,29,0,1 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":98,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4107,"imageType":285,"fileVersion":16784640},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":58837,"endpoint":11,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}' +10s
Zigbee2MQTT:debug 2022-05-08 10:24:25: Received Zigbee message from 'Lily XL @spa @lille_plantekasse', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784640,"imageType":285,"manufacturerCode":4107}' from endpoint 11 with groupID null
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +22ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +2s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +2s
  zigbee-herdsman:deconz:driver query aps data indication +26ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 147 +61ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +94ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +7ms
[... cut ...]
Zigbee2MQTT:debug 2022-05-08 10:25:18: Received Zigbee message from 'Amarant @spa @lille_plantekasse', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784640,"imageType":285,"manufacturerCode":4107}' from endpoint 11 with groupID null
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +24ms
^[c  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 166 +5s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +5s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +9ms
```

DennisJohnsen, 2022-05-09:
> Could you check if it works on z2m 1.25.0?

Yeah, I first was on.. 1.23.0 I think.
Then I updated to 1.25.1 and could still not pair, and then for the same [issues as mentioned here](https://github.com/Koenkk/zigbee2mqtt/issues/12414) i then reverted back one release to 1.25.0, but still no luck with pairing.

I want to say though.. I brought out my old Philips Hue bridge and quickly set it up with a single Hue Bulb. And i could not pair the switches to that either - so I'm currently waiting for the seller to respond and send me a 3rd new one.

Regarding the Philips Hue paring issues, i'm not sure if its because the switches are both broken (unlikely) or that they're already "locked" to my Raspbee2 and i'm just having issues resetting them.

DennisJohnsen, 2022-05-12:
I just received a 3rd switch from the seller.
It paired instantly without any issues.

It seems like I had a switch break for me, and the replacement was DOA. The 2nd replacement works fine. 
Sorry for bothering you :)

closed 2022-05-12 by DennisJohnsen