Koenkk/zigbee2mqtt #12412: EcoDim Edison ST64 Lamp is wrongly recognized as Paulmann 50064 Lamp

Irru-, 2022-05-07:
### What happened?

After pairing the [Ecodim Edison ST64](https://www.ecodim.nl/ed-10041.html) (which is as of yet unsupported), the lamp was wrongly identified as [Paulmann 50064](https://www.zigbee2mqtt.io/devices/50064.html#paulmann-50064).

All functions do seem to work, however.

### What did you expect to happen?

I expected to go into the flow of adding support for new devices, but that did not happen.

### How to reproduce it (minimal and precise)

I've tried multiple times to reset/pair the lamp, but each time it gets recognized as the Paulmann 50064 lamp.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20190608

### Adapter

20190608

### Debug log

Info 2022-05-07 14:08:39MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"oepw0-2"}'
Info 2022-05-07 14:08:52Device '0x00158d0001273b2e' joined
Info 2022-05-07 14:08:52MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0001273b2e","ieee_address":"0x00158d0001273b2e"},"type":"device_joined"}'
Info 2022-05-07 14:08:52MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0001273b2e"},"type":"device_connected"}'
Info 2022-05-07 14:08:52MQTT publish: topic 'homeassistant/light/0x00158d0001273b2e/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0x00158d0001273b2e/set","device":{"identifiers":["zigbee2mqtt_0x00158d0001273b2e"],"manufacturer":"Paulmann","model":"SmartHome led spot (50064)","name":"0x00158d0001273b2e","sw_version":"00.00.01"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/0x00158d0001273b2e","max_mireds":500,"min_mireds":150,"name":"0x00158d0001273b2e","schema":"json","state_topic":"zigbee2mqtt/0x00158d0001273b2e","supported_color_modes":["color_temp"],"unique_id":"0x00158d0001273b2e_light_zigbee2mqtt"}'
Info 2022-05-07 14:08:52MQTT publish: topic 'homeassistant/sensor/0x00158d0001273b2e/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0001273b2e"],"manufacturer":"Paulmann","model":"SmartHome led spot (50064)","name":"0x00158d0001273b2e","sw_version":"00.00.01"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d0001273b2e","name":"0x00158d0001273b2e linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0001273b2e","unique_id":"0x00158d0001273b2e_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2022-05-07 14:08:52Starting interview of '0x00158d0001273b2e'
Info 2022-05-07 14:08:52MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0001273b2e","ieee_address":"0x00158d0001273b2e","status":"started"},"type":"device_interview"}'
Info 2022-05-07 14:08:52MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0001273b2e"},"type":"pairing"}'
Info 2022-05-07 14:08:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0001273b2e","ieee_address":"0x00158d0001273b2e"},"type":"device_announce"}'
Info 2022-05-07 14:08:55MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d0001273b2e"},"type":"device_announced"}'
Info 2022-05-07 14:09:02Successfully interviewed '0x00158d0001273b2e', device has successfully been paired
Info 2022-05-07 14:09:02Device '0x00158d0001273b2e' is supported, identified as: Paulmann SmartHome led spot (50064)
Info 2022-05-07 14:09:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"SmartHome led spot","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":150}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"50064","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Paulmann"},"friendly_name":"0x00158d0001273b2e","ieee_address":"0x00158d0001273b2e","status":"successful","supported":true},"type":"device_interview"}'
Info 2022-05-07 14:09:02Configuring '0x00158d0001273b2e'
Info 2022-05-07 14:09:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"SmartHome led spot","friendly_name":"0x00158d0001273b2e","model":"50064","supported":true,"vendor":"Paulmann"},"type":"pairing"}'
Info 2022-05-07 14:09:02MQTT publish: topic 'zigbee2mqtt/0x00158d0001273b2e', payload '{"color_mode":"xy","linkquality":31,"state":null}'
Info 2022-05-07 14:09:02MQTT publish: topic 'zigbee2mqtt/0x00158d0001273b2e', payload '{"color_mode":"xy","linkquality":31,"state":null}'
Info 2022-05-07 14:09:02Successfully configured '0x00158d0001273b2e'

Koenkk, 2022-05-07:
Could you provide the data/database.db entry of this device?

Irru-, 2022-05-12:
Apologies for the late reply

`{"id":15,"type":"Router","ieeeAddr":"0x00158d0001273b2e","nwkAddr":2259,"manufId":4137,"manufName":"ZigBee/CCT","powerSource":"Mains (single phase)","modelId":"CCT Light","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,4,3,6,8,5,768,65535,65535,4096,4111],"outClusterList":[25,4111],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":21,"dateCode":"20200505","swBuildId":"00.00.01","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1652368950490,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2022-05-12:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-19 by github-actions[bot]