Koenkk/zigbee2mqtt #12414: Hue Tap Green Power - Philips 8718696743133, slow respons after uppgradeing from 1.25.0 to 1.25.1

VesleSlotten, 2022-05-07:
### What happened?

After upgrading from 1.25.0 to 1.25.1 responstime on my two Hue Tap Green Power are very slow.  Takes like 2-4 sSeconds from I press a button to it respond (monitored in the Z2M log).

### What did you expect to happen?

More og less immediate response.

### How to reproduce it (minimal and precise)

open Z2M log.  Press any button on the Green Power Tap Switch and se how long time it takes befor you see a log entry apering.  Takes from 2-6 seconds.
Downgrade back to 1.25.0 and repeat the test and you get an immediate response.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x26720700

### Adapter

ConBee II

Koenkk, 2022-05-07:
@filips could this have something to do with https://github.com/Koenkk/zigbee-herdsman/commit/be8797f051af1d44d0a3704b616abec228d2e752 ?

DennisJohnsen, 2022-05-08:
Can confirm this issue. I just went back to 1.25.0 and my EnOcean PTM 215Z's are now again close to the instant speeds that it was before.

bambidotexe, 2022-05-08:
Same here, even worse, it completly crash my network (using a conbee 2), without any logs. Going back to 1.25.0 solve the issue.

filips, 2022-05-08:
> @filips could this have something to do with [Koenkk/zigbee-herdsman@be8797f](https://github.com/Koenkk/zigbee-herdsman/commit/be8797f051af1d44d0a3704b616abec228d2e752) ?

Hmm potentially.. 

It sounds like this is not happening during commissioning? There is some slight differences in the commissioning flow that could cause a different communication mode to be set on the selected Green Power Proxy. During normal communications I'm not completely sure what the difference would be. Is there any difference in the log output seen before/after 1.25.1?

Koenkk, 2022-05-09:
I tried to reproduce the issue but failed two do so; tried 2 scenarios:
- fresh network start on 1.25.1, pair hue bulb, pair hue tap -> actions are immediately sent to MQTT
- fresh network start on 1.25.0, pair hue bulb, pair hue tap, update to z2m 1.25.1, start z2m -> actions are immediately sent to MQTT

Can someone provide the herdsman debug logging starting 5 secs before pressing the button until the action is send to MQTT?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

VesleSlotten, 2022-05-09:
Here is herdsman debug logging.  Device in question is "GreenPower Switch 02"
[short_log_1.25.1.txt](https://github.com/Koenkk/zigbee2mqtt/files/8655041/short_log_1.25.1.txt)

tahoo777, 2022-05-10:
Can confirm this issue, I just went back to 1.25.0 and it works again.

Smart Switch Green Power Device
https://www.ledhome.dk/vare/smart-home/zigbee/zigbee-kontakter/styreprint-pcb-til-batteri-kontakt-tryk-hue-kompatibelt/

Koenkk, 2022-05-10:
@filips seems this only happens in combination with the Conbee II. I can also not pair any green power devices on z2m 1.25.1 when using the Conbee II. is there any chance you can look into this?

Pairing log on z2m 1.25.0:

```
  zigbee-herdsman:deconz:frameParser GP commissioning notification +8s
  zigbee-herdsman:deconz:frameParser GP_DATA_INDICATION - src id: 0x46f4fe cmd id: 0xe0 frameCounter: 0x75c +1ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":22,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":62081,"srcID":4650238,"frameCounter":1884,"commandID":224,"payloadSize":27,"commandFrame":{"deviceID":2,"options":129,"extendedOptions":242,"securityKey":{"type":"Buffer","data":[241,236,146,171,255,143,19,99,225,70,190,181,24,201,12,171]},"keyMic":3587458724,"outgoingCounter":1884,"applicationInfo":0,"numGdpCommands":0,"gpdCommandIdList":{"type":"Buffer","data":[]}}},"Command":{"ID":4,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":2048}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":2048}]}],"name":"commisioningNotification"}},"address":4650238,"endpoint":242,"linkquality":127,"groupID":2948,"wasBroadcast":false,"destinationEndpoint":1}' +18s
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '4650238', skipping... +2ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 22 paramId: 1 +8s
  zigbee-herdsman:deconz:frameParser GP commissioning notification +6ms
  zigbee-herdsman:deconz:frameParser MAC: 0x00212effff04d4e0 +1ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 23 paramId: 7 +6ms
  zigbee-herdsman:deconz:frameParser NWKADDR: 0 +6ms
  zigbee-herdsman:deconz:adapter no response expected +15s
  zigbee-herdsman:controller:log Green power device '{"sourceID":4650238,"deviceID":2,"networkAddress":62718}' joined +9ms
  zigbee-herdsman:controller:log New green power device '0x000000000046f4fe' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x000000000046f4fe' +0ms
Zigbee2MQTT:info  2022-05-10 19:55:51: Device '0x000000000046f4fe' joined
Zigbee2MQTT:info  2022-05-10 19:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000000000046f4fe","ieee_address":"0x000000000046f4fe"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-05-10 19:55:51: Successfully interviewed '0x000000000046f4fe', device has successfully been paired
Zigbee2MQTT:info  2022-05-10 19:55:51: Device '0x000000000046f4fe' is supported, identified as: Philips Hue tap (8718696743133)
Zigbee2MQTT:info  2022-05-10 19:55:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Hue tap","exposes":[{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["press_1","press_2","press_3","press_4"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"8718696743133","options":[],"supports_ota":false,"vendor":"Philips"},"friendly_name":"0x000000000046f4fe","ieee_address":"0x000000000046f4fe","status":"successful","supported":true},"type":"device_interview"}'
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x EP:242 SeqNr. 24 request id: 4 +15ms
```

z2m 1.25.1

```
  zigbee-herdsman:deconz:frameParser GP commissioning notification +7s
  zigbee-herdsman:deconz:frameParser GP_DATA_INDICATION - src id: 0x46f4fe cmd id: 0xe0 frameCounter: 0x75a +2ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":22,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":62081,"frameCounter":4650238,"commandID":90,"payloadSize":7,"commandFrame":{"raw":{"type":"Buffer","data":[0,0,224,27,2,129,242,241,236,146,171,255,143,19,99,225,70,190,181,24,201,12,171,164,70,212,213,90,7,0,0]}}},"Command":{"ID":4,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":2048}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":2048}]}],"name":"commisioningNotification"}},"address":4650238,"endpoint":242,"linkquality":127,"groupID":2948,"wasBroadcast":false,"destinationEndpoint":1}' +17s
[... cut ...]
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 27 +1ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x0 request id: 4 confirm status: 0 +6ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 22 +8ms
  zigbee-herdsman:deconz:adapter false, true, false +0ms
  zigbee-herdsman:deconz:adapter resolve request +0ms
```

filips, 2022-05-10:
I think there's some data parsing situation here.. Comparing 1.25.0 here:

```
"Payload": {
            "options": 62081,
            "srcID": 4650238,
            "frameCounter": 1884,
            "commandID": 224,
            "payloadSize": 27,
            "commandFrame": {
                "deviceID": 2,
                "options": 129,
                "extendedOptions": 242,
                "securityKey": {
                    "type": "Buffer",
                    "data": [
                        241,
                        236,
                        146,
                        171,
                        255,
                        143,
                        19,
                        99,
                        225,
                        70,
                        190,
                        181,
                        24,
                        201,
                        12,
                        171
                    ]
                },
                "keyMic": 3587458724,
                "outgoingCounter": 1884,
                "applicationInfo": 0,
                "numGdpCommands": 0,
                "gpdCommandIdList": {
                    "type": "Buffer",
                    "data": []
                }
            }
        }
```

With 1.25.1:

```
"Payload": {
            "options": 62081,
            "frameCounter": 4650238,
            "commandID": 90,
            "payloadSize": 7,
            "commandFrame": {
                "raw": {
                    "type": "Buffer",
                    "data": [
                        0,
                        0,
                        224,
                        27,
                        2,
                        129,
                        242,
                        241,
                        236,
                        146,
                        171,
                        255,
                        143,
                        19,
                        99,
                        225,
                        70,
                        190,
                        181,
                        24,
                        201,
                        12,
                        171,
                        164,
                        70,
                        212,
                        213,
                        90,
                        7,
                        0,
                        0
                    ]
                }
            }
        }
```

Looking into the latter, you can see byte 2 in the raw payload has the value 224, corresponding to the GP command ID seen parsed correctly in 1.25.0. I will look a bit more into this to see whether I can find the root cause of this, although I don't have a Conbee II to test with.

filips, 2022-05-10:
I observe that the option field has the value 62081, which in binary is

```
0b1111001010000001
               ^^^
```

I would expect the three least significant bits here to carry the information of Application ID, which has two valid values according to GP spec, `0b000` and `0b010`. If the byte order is messed up in deconzAdapter.ts:1051 in the function checkReceivedGreenPowerIndication(), that could explain the misparsing of the commissioning notification. If the bytes are swapped, this value would instead be interpreted as the Application ID:

```
0b1111001010000001
       ^^^
```

Which is indeed a valid value for that field. That would presumably make the command parse correctly as per the cluster definition.

Koenkk, 2022-05-11:
@filips could you create a pr to fix it? I can test it.

VesleSlotten, 2022-05-16:
unfortunately I am on holiday for a couple of days and not back until 18th of May. Will be testing as soon as I'm back.

VesleSlotten, 2022-05-18:
Log from latest test (1.25.1-dev):
[log_1.25.1_dev.txt](https://github.com/Koenkk/zigbee2mqtt/files/8718607/log_1.25.1_dev.txt)

Koenkk, 2022-05-21:
@VesleSlotten what log message immediately appears after pressing the button? I would expect something with `zigbee-herdsman:deconz:frameParser GP data notification`

VesleSlotten, 2022-05-21:
That's probably correct. The only thing that is a little strange is that the time from the log entry to the mqtt message is relatively short.
Did some tests today as well, and it is correct that it is "... data notification" which is the first command in the log after I have pressed the button. Sees that the time varies greatly from when I press the button to the mqtt message appears.

Koenkk, 2022-05-22:
@VesleSlotten does the message that is received immediately after pressing the button start with `GP` (so `GP data notification`)? Or is it just `Data notification`?

VesleSlotten, 2022-05-22:
Yes, "GP data notification",

Here's an example from yesterday’s log:

2022-05-18T17:02:02.754Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 246
2022-05-18T17:02:02.759Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-05-18T17:02:02.759Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-05-18T17:02:05.339Z zigbee-herdsman:deconz:frameParser **GP data notification**
2022-05-18T17:02:05.340Z zigbee-herdsman:deconz:frameParser GP_DATA_INDICATION - src id: 0x41d829 cmd id: 0x10 frameCounter: 0xddd
2022-05-18T17:02:05.344Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":247,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"options":0,"srcID":4315177,"frameCounter":3549,"commandID":16,"payloadSize":0,"commandFrame":{}},"Command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}},"address":4315177,"endpoint":242,"linkquality":127,"groupID":2948,"wasBroadcast":false,"destinationEndpoint":1}'
2022-05-18T17:02:05.347Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 247 paramId: 5
2022-05-18T17:02:05.348Z zigbee-herdsman:deconz:frameParser GP data notification
2022-05-18T17:02:05.349Z zigbee-herdsman:deconz:frameParser GP data notification
2022-05-18T17:02:05.353Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-05-18T17:02:05.353Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-05-18T17:02:05.354Z zigbee-herdsman:deconz:frameParser PANID: d96
2022-05-18T17:02:05.357Z zigbee-herdsman:deconz:driver query aps data indication
2022-05-18T17:02:05.358Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 248 paramId: 11
2022-05-18T17:02:05.362Z zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd
2022-05-18T17:02:05.363Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 249
2022-05-18T17:02:05.364Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 250 paramId: 28
2022-05-18T17:02:05.377Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 249 srcAddr: 0xd829 destAddr: 0xb84 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
2022-05-18T17:02:05.378Z zigbee-herdsman:deconz:frameParser response payload: 17,122,0,160,84,41,216,65,0,221,13,0,0,16,0,2,124,255
2022-05-18T17:02:05.383Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":122,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"options":21664,"srcID":4315177,"frameCounter":3549,"commandID":16,"payloadSize":0,"commandFrame":{"raw":{"type":"Buffer","data":[2,124,255]}},"gppNwkAddr":31746,"gppGddLink":255},"Command":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]},{"name":"gppGddLink","type":32,"conditions":[{"type":"bitMaskSet","param":"options","mask":16384}]}],"name":"notification"}},"address":55337,"endpoint":242,"linkquality":255,"groupID":2948,"wasBroadcast":true,"destinationEndpoint":0}'
2022-05-18T17:02:05.383Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-05-18T17:02:05.384Z zigbee-herdsman:deconz:frameParser CHANNEL: 25
2022-05-18T17:02:05.389Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 251 paramId: 5
2022-05-18T17:02:05.392Z zigbee-herdsman:deconz:frameParser PANID: d96
2022-05-18T17:02:05.395Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 252 paramId: 24
2022-05-18T17:02:05.399Z zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden
2022-05-18T17:02:05.399Z zigbee-herdsman:controller:greenpower Unhandled Zigbee GreenPower command: 0x10
[... cut ...] 05-18T19:02:05+02:00","linkquality":127}'
Zigbee2MQTT:info  2022-05-18 19:02:05: MQTT publish: topic 'conbee/GreenPower Switch 02', payload '{"action":"","device":{"friendlyName":"GreenPower Switch 02","ieeeAddr":"0x000000000041d829","manufacturerID":null,"model":"8718696743133","networkAddress":4315177,"type":"GreenPower"},"last_seen":"2022-05-18T19:02:05+02:00","linkquality":127}'
Zigbee2MQTT:info  2022-05-18 19:02:05: MQTT publish: topic 'conbee/GreenPower Switch 02/action', payload 'press_2'

Koenkk, 2022-05-22:
Does `2022-05-18T17:02:05.339Z zigbee-herdsman:deconz:frameParser GP data notification` directly appear after pressing the button? Because I see no delay with the actual MQTT publish later:

```
Zigbee2MQTT:info 2022-05-18 19:02:05: MQTT publish: topic 'conbee/GreenPower Switch 02/action', payload 'press_2'
```

VesleSlotten, 2022-05-22:
That's right, it varies whether there is a delay or not. With version 1.25.0, I never experience any delay.

Otherwise I see in the log from 1.25.1.dev two rows that do not come in version 1.25.0, 
`2022-05-18T17:02:05.399Z zigbee-herdsman:controller:greenpower Unhandled Zigbee GreenPower command: 0x10`
and 
`2022-05-18T17:02:05.448Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '55337', skipping...`
The latter often comes after the MQTT message.
The address in the message, '55337', can also be find in log entry:
`2022-05-18T17:02:05.383Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":122,"manufacturerCode":null,"commandIdentifier":0},"Payload":'...
`

VesleSlotten, 2022-05-23:
Hi, here is an example where there is like 6 seconds from I press the button to the mqtt massage appear.  I also included the log entries for a few seconds before I pressed the button, and a couple of seconds after the MQTT massage appear.
[log_1.25.1_dev_3.txt](https://github.com/Koenkk/zigbee2mqtt/files/8757619/log_1.25.1_dev_3.txt)

Koenkk, 2022-05-27:
@VesleSlotten pushed another possible fix, can you check the latest dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koenkk, 2022-05-27:
Both warnings can be ignored; zigbee-herdsman doesn't do anything with this message, it just passes it to z2m, previously this was not logged. Thanks for confirming, assuming this can be closed now 😄