Koenkk/zigbee2mqtt #12415: [New device support]: Clone of https://www.zigbee2mqtt.io/devices/ZVG1.html WAter Valve

frantzphilippe, 2022-05-07:
### Link

https://www.pearl.be/article/ZX7155/valve-darrosage-connectee-pour-passerelle-zigbee-avec-commandes-vocales

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x94deb8fffe24ff8c","nwkAddr":2218,"manufId":4098,"manufName":"_TZE200_2wg5qrjy","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"]�\t*\u0011","65506":31,"65508":0,"appVersion":72,"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1651948057349,"defaultSendRequestWhen":"immediate"}

### Comments

Looks like being exactly the same than this one : https://www.zigbee2mqtt.io/devices/ZVG1.html

But it's not recognized (not suported message once paired) by zigbee2mqtt.

I tried to use some of the converters that are available but nothing matches this model.
I tried to make my own external converter, but wihout success as I don't really what is exposed and the exact syntax.
Eventhough as it's the same than this one https://www.zigbee2mqtt.io/devices/ZVG1.html, but from another supplier, I really dont understand how to make it work.

Hope someone can help.

Thanks!

Koenkk, 2022-05-08:
`_TZE200_2wg5qrjy` is already supported in the latest dev

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-08 by Koenkk

frantzphilippe, 2022-05-08:
Thank you so much! Total efficiency and kindness.

SteCheFr, 2022-05-12:
Hi,

I am facing the same problem with _TZE200_81isopgh

Could you help me please ?

Regards

Koenkk, 2022-05-13:
@SteCheFr `_TZE200_81isopgh` is already supported in the latest dev (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SteCheFr, 2022-05-13:
Thanks you very much, it works !