Koenkk/zigbee2mqtt #12417: [New device support]: Mercator Ikuu Dual Gang Outdoor Power Outlet

llamalump, 2022-05-08:
### Link

https://www.ikuu.com.au/product/double-power-point-ip54/

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x84fd27fffece6888","nwkAddr":6108,"manufId":4098,"manufName":"_TZ3210_7jnk7l3k","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820,61184,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3210_7jnk7l3k","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":0,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,62]}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff07ab6f","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07ab6f","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff07ab6f","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff07ab6f","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6,61184,3],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07ab6f","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1504072437},"lastSeen":1645174607303,"defaultSendRequestWhen":"immediate"}

### Comments

I can't take credit for writing the converter, but I can't find the original thread where I found it to give the proper credit :(

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3210_7jnk7l3k'}],
    zigbeeModel: ['TS011F'],
    model: 'SPP02GIP',
    vendor: 'Mercator',
    description: 'Mercator Ikuu Double Outdoors Power Point',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.power().withEndpoint('left'), e.current().withEndpoint('left'), e.voltage().withEndpoint('left').withAccess(ea.STATE), e.energy()],
    endpoint: (device) => {
            return {left: 1, right: 2};
    },
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    meta: {multiEndpoint: true}, 
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint1);
        await reporting.rmsVoltage(endpoint1, {change: 5});
        await reporting.rmsCurrent(endpoint1, {change: 50});
        await reporting.activePower(endpoint1, {change: 1});
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint2);
        endpoint1.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint1.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
	device.save();
    },
};

module.exports = definition;
```

Koenkk, 2022-05-08:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-08 by Koenkk

llamalump, 2022-05-09:
Thanks @Koenkk! How long does it usually take for features to be propagated to master?

Koenkk, 2022-05-09:
@llamalump every 1st of the month.