Koenkk/zigbee2mqtt #12418: [Feature request]: Add batterylevel to BlitzWolf BW-IS3

Jens-Wymeersch, 2022-05-08:
### Is your feature request related to a problem? Please describe

Yes, as I never know if I need to charge my motion sensors.

### Describe the solution you'd like

I like to have the battery level visible

### Describe alternatives you've considered

The only alternative is not having this info. Else add another hub from blitzwolf which is not an option for me.

### Additional context

https://www.zigbee2mqtt.io/devices/BW-IS3.html
https://github.com/Koenkk/zigbee2mqtt/issues/10686

hungpr0, 2022-06-09:
> Could you provide the data/database.db entry of this device?

I have the device.
```
{"id":8,"type":"EndDevice","ieeeAddr":"0x847127fffe0d1db6","nwkAddr":1648,"manufId":0,"manufName":"_TYST11_i5j6ifxj","powerSource":"Battery","modelId":"5j6ifxj\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"5j6ifxj\u0000","manufacturerName":"_TYST11_i5j6ifxj","stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1654761628947,"defaultSendRequestWhen":"immediate"}
```

Jens-Wymeersch, 2022-06-24:
@Koenkk I'm happy to figure this out but i suppose this implies that I need to buy a Tuya Bridge or am I missing something ?

Koenkk, 2022-06-24:
@Jens-Wymeersch yes you need a TuYa bridge; [this once](https://www.aliexpress.com/item/3256802576084489.html?spm=a2g0o.productlist.0.0.2f643e99epqCca&algo_pvid=1f163717-812b-46b2-9a87-66ce50b7d0b3&algo_exp_id=1f163717-812b-46b2-9a87-66ce50b7d0b3-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000029089202550%22%7D&pdp_npi=2%40dis%21USD%21%2113.99%2113.58%21%21%21%21%400b0a050b16560826791826928e32f7%2112000029089202550%21sea) should be fine.

Jens-Wymeersch, 2022-07-14:
@Koenkk Just received the Tuya Gateway. Here is the result 

[image]

It looks like we can only get an indication of the battery level - High/Medium/Low.

Jens-Wymeersch, 2022-07-15:
@Koenkk Thank you. Can you please explain how I setup such an external converter? Thanks jens

Koenkk, 2022-07-15:
It is explained in https://github.com/Koenkk/zigbee2mqtt/issues/12418#issuecomment-1185633245  : 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Jens-Wymeersch, 2022-07-15:
@Koenkk I tried without the commented part which didn't do anything. So now I'm trying to uncomment, but I'm getting errors in the code. Should it be like this ?
```
const definition = {
    zigbeeModel: ['5j6ifxj'],
    model: 'BW-IS3',
    vendor: 'BlitzWolf',
    description: 'Rechargeable Zigbee PIR motion sensor',
    fromZigbee: [fz.blitzwolf_occupancy_with_timeout, fzLocal.BWIS3],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery(), e.battery_low(), 
    exposes.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};
```

Koenkk, 2022-07-17:
@Jens-Wymeersch can you check if the device configures successful? To reconfigure press the yellow refresh button on the device page, it may be necessary to wakeup the device right before doing this

Jens-Wymeersch, 2022-07-17:
@Koenkk Thank you for coming back to me. I'm running Z2M within HA on a RPI3 and updated the configuration file now directly in the directory which had an effect. Both with and without the commented text, all the motion sensors have became unsupported.

I get the following error message
`2022-07-17 23:23:03Received message from unsupported device with Zigbee model '5j6ifxj' and manufacturer name '_TYST11_i5j6ifxj'`

Pushing the the yellow refresh button, resulted in the following error message 
`device could not be configured`

Jens-Wymeersch, 2022-07-17:
Here you find my configuration.yaml file

`devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://XX.YY.ZZ.UU:1883
  user: User
  password: pw
serial:
  port: /dev/ttyUSB0
external_converters:
  - ext_converter.js
advanced:
  log_level: warn
  pan_id: 6754
  channel: 20
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  rtscts: false
device_options: {}
blocklist:
  - '0x0000000000000000'
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: false
zigbee_herdsman_debug: true
`

Koenkk, 2022-07-17:
I think the external converter is not loaded, can you change the description to `    description: 'Rechargeable Zigbee PIR motion sensor CUSTOM',`. restart z2m and verify the the description contains `CUSTOM` in the z2m frontend?

Jens-Wymeersch, 2022-07-17:
@Koenkk I'm sorry but i'm not following you. What do you want me to do exactly ?

Koenkk, 2022-07-17:
do the same steps as you did in https://github.com/Koenkk/zigbee2mqtt/issues/12418#issuecomment-1185837775 but change the line `    description: 'Rechargeable Zigbee PIR motion sensor',` to `    description: 'Rechargeable Zigbee PIR motion sensor CUSTOM',`. Then start z2m, open the web frontend and click on your device. The `description` should now end with `CUSTOM`.

Jens-Wymeersch, 2022-07-17:
@Koenkk The description is empty. I'm starting to think it has something to do with the configuration file of Z2M in Home Assistant.

Jens-Wymeersch, 2022-07-17:
In UI of Z2M of Home assistant, you aren't able to add this option (only socat, mqtt and serial). When you go to edit in Yaml, you see the configuration file, but this one is different from the one found in the directory.

Jens-Wymeersch, 2022-07-17:
Created an issue : https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/364

Koenkk, 2022-07-18:
> but this one is different from the one found in the directory.

Try editing the configuration.yaml directly, if that doesn't work use the frontend (settings -> external converters)

Jens-Wymeersch, 2022-08-01:
@Koenkk I've set up a clean new RPI HA development environment to test this with only 1 device. I've added the converter via the UI and looked at the configuration.yaml file (which was updated). Not sure what to do next ?

[image]

Koenkk, 2022-08-01:
@Jens-Wymeersch can you provide the debug log when starting z2m? 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Jens-Wymeersch, 2022-08-03:
@Koenkk Here you go

`Zigbee2MQTT:info  2022-08-03 19:22:06: Starting interview of '0xec1bbdfffe302ef3'
Zigbee2MQTT:info  2022-08-03 19:22:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe302ef3","ieee_address":"0xec1bbdfffe302ef3","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:debug 2022-08-03 19:22:07: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TYST11_i5j6ifxj","modelId":"5j6ifxj\u0000"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:07: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:09: Received Zigbee message from '0xec1bbdfffe302ef3', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":3}],"seq":1280}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:09: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:12: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:12: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:14: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:14: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:16: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"appVersion":73}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:16: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:18: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:18: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:20: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:20: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:22: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20180727"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:22: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-08-03 19:22:24: Received Zigbee message from '0xec1bbdfffe302ef3', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-03 19:22:24: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2022-08-03 19:22:24: Successfully interviewed '0xec1bbdfffe302ef3', device has successfully been paired
Zigbee2MQTT:warn  2022-08-03 19:22:24: Device '0xec1bbdfffe302ef3' with Zigbee model '5j6ifxj' and manufacturer name '_TYST11_i5j6ifxj' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2022-08-03 19:22:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xec1bbdfffe302ef3","ieee_address":"0xec1bbdfffe302ef3","status":"successful","supported":false},"type":"device_interview"}'`

Koenkk, 2022-08-04:
What is shown in the frontend under settings -> about?

Jens-Wymeersch, 2022-08-04:
HI @Koenkk, here you find the starting log 

`Zigbee2MQTT:debug 2022-08-05 00:26:32: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-08-05 00:26:32: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-05.00-26-32' filename: log.txt
Zigbee2MQTT:info  2022-08-05 00:26:32: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-05 00:26:32: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:debug 2022-08-05 00:26:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:info  2022-08-05 00:26:33: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-08-05 00:26:33: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:debug 2022-08-05 00:26:33: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b0021b951bd","panID":6754}
Zigbee2MQTT:info  2022-08-05 00:26:33: Currently 1 devices are joined:
Zigbee2MQTT:info  2022-08-05 00:26:33: 0xec1bbdfffe302ef3 (0xec1bbdfffe302ef3): Not supported (EndDevice)
Zigbee2MQTT:info  2022-08-05 00:26:33: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-05 00:26:34: Connecting to MQTT server at mqtt://XX.YY.ZZ.UU:1883
Zigbee2MQTT:debug 2022-08-05 00:26:34: Using MQTT login with username: Username
Zigbee2MQTT:info  2022-08-05 00:26:34: Connected to MQTT server
Zigbee2MQTT:info  2022-08-05 00:26:34: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2022-08-05 00:26:34: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info  2022-08-05 00:26:34: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2022-08-05 00:26:34: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe302ef3', payload '{"linkquality":171,"occupancy":true}'`

and they about section in the frontend :

[image]

Koenkk, 2022-08-05:
Found why it was not working, fixed it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-12 by github-actions[bot]