Koenkk/zigbee2mqtt #12421: Smart plug TS011F manufacturer _TZ3000_cehuw1lw not reporting power/current/energy

paradou, 2022-05-08:
### What happened?

Smart plug identified as TS011F _plug_1 using dev branch of zigbee2mqtt. Everything works fine but power, current and energy reporting. When i pair the device, power and current fields show N/A. Some times later, N/A becomes 0... and the device state changes to off.  

Manufacturer : _TZ3000_cehuw1lw .

 I see weird characters in database.db : (see full data in debug log field) 
{"genBasic":{"attributes":{"65503":"AV\n*\u0012","65506":53,"65508":0,"modelId":"TS011F","ma

MQTT debug message :
Zigbee2MQTT:debug 2022-05-08 11:37:19: Received Zigbee message from '0xa4c138750cc02a8b', type 'attributeReport', cluster 'genBasic', data '{"65506":53,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0

### What did you expect to happen?

Power, voltage and energy reporting.

### Zigbee2MQTT version

edge

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB dongle (latest firmware)

### Debug log

{"id":27,"type":"Router","ieeeAddr":"0xa4c138750cc02a8b","nwkAddr":14333,"manufId":4417,"manufName":"_TZ3000_cehuw1lw","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"AV\n*\u0012","65506":53,"65508":0,"modelId":"TS011F","manufacturerName":"_TZ3000_cehuw1lw","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":240}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,1]}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":362538,"53251":141,"53252":73660,"53253":9929,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c11517","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c11517","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c11517","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1651998304941,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-05-09:
- regarding power measurements; does it work correctly in z2m 1.25.0?
- the plug powering off seems to be an issue in the plug itself: https://github.com/Koenkk/zigbee2mqtt/issues/11648#issuecomment-1120991469

paradou, 2022-05-09:
No. The device was not recognized under 1.25.0. I tried to install an External converter but the power monitoring never worked.

Koenkk, 2022-05-09:
Could you check if it works now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

paradou, 2022-05-09:
Still no power and current monitoring (N/A or null). 

MQTT publish: topic 'zigbee2mqtt/0xa4c138750cc02a8b', payload '{"child_lock":"UNLOCK","current":null,"energy":0.01,"indicator_mode":"off/on","linkquality":163,"power":null,"power_outage_memory":"off","state":"ON","update":{"state":null},"update_available":null,"voltage":243}'

Data in database.db file look better :

{"id":27,"type":"Router","ieeeAddr":"0xa4c138750cc02a8b","nwkAddr":690,"manufId":4417,"manufName":"_TZ3000_cehuw1lw","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_cehuw1lw","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":243}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,1]}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":362538,"53251":141,"53252":73660,"53253":9929,"53264":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c11517","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c11517","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c11517","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1652130386743,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-05-10:
Is it recognised as TS011F_plug_3 now? (instead of TS011F_plug_1)

paradou, 2022-05-11:
Frontend version : 0.6.84

[image]

paradou, 2022-05-12:
Device recognized as TS011F_plug_1. I don't see any change : power and current = 0.

Koenkk, 2022-05-12:
@paradou could you try force removing the device, restart z2m and repair the device?

paradou, 2022-05-12:
Done. The device is instantly recognized during the pairing process and there is no change. I start to wonder if this plug is really working. I hope i don't waste your time.

Koenkk, 2022-05-13:
@[smulle48](https://github.com/smulle48) since you added the `_TZ3000_cehuw1lw`, can you confirm that power measurements are workign?

paradou, 2022-05-19:
@Koenkk I have some news about the faulty plug : i discovered that the power measurements are not working if the power is too low, ie 3W (light). Sorry i have missed that ! I couldn't guess as it was working fine with another plug with power monitoring. 

About _TZ3000_cehuw1lw , i checked that power monitoring is working for 13 watts device but there are some "holes" in the monitoring (not regular pattern) as you can see in the picture below (power is on but monitoring shows 0 watt). Same issue with a more powerful device (>60 watts) but the pattern seems to be more regular. I think that's all for now.
Thank's again for your time !
[image]

Koenkk, 2022-05-21:
@paradou I guess that is due to crappy hardware, there are more hardware related reports of this device: https://github.com/Koenkk/zigbee2mqtt/issues/11648

closed 2022-06-28 by github-actions[bot]

manny199, 2022-10-06:
New about NOT value below 3 Watt?

Aresitoo, 2023-02-18:
I have the same problem with a plug I just received it's showing up as [TS011F_plug_1] tried on a 1500w device and not reporting any watts in consumption