Koenkk/zigbee2mqtt #12422: New Device Pairing - Interview failed

h0mesmart, 2022-05-08:
### What happened?

Hello there, 
I need you help to pair this [this device](https://www.zigbee2mqtt.io/devices/TI0001-cover.html) please. 
When I paired  I got this as result only the IEEE address 0x00124b002336dda0 (0x92BC) and that's all. 

I've tried many times after rebooting the zigbee2mqtt server, and the device its self. Got the same result. 

Here is my config 

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt
serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  port: 8080
advanced:
  network_key:
       [HIDDEN]
  ext_pan_id:
    - 33
    - 117
    - 141
    - 25
    - 0
    - 75
    - 18
    - 0
  channel: 26

````

Please any help is welcome. Thank you 

### What did you expect to happen?

Able to pair and use the device

### How to reproduce it (minimal and precise)

Put the device in a pairing mode. Read the log

### Zigbee2MQTT version

    1.25.1 commit: 3f6a137

### Adapter firmware version

0x26720700

### Adapter

conbee II

### Debug log

```
zigbee2mqtt    | Zigbee2MQTT:info  2022-05-08 11:54:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002336dda0","ieee_address":"0x00124b002336dda0","status":"started"},"type":"device_interview"}'
zigbee2mqtt    | Zigbee2MQTT:info  2022-05-08 11:54:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b002336dda0"},"type":"pairing"}'
zigbee2mqtt    | 2022-05-08T09:54:16.991Z zigbee-herdsman:controller:device:log Interview - start device '0x00124b002336dda0'
zigbee2mqtt    | 2022-05-08T09:54:16.991Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0x00124b002336dda0', already got it
zigbee2mqtt    | 2022-05-08T09:54:16.992Z zigbee-herdsman:controller:device:log Interview - Detected potential TuYa end device, reading modelID and manufacturerName...
zigbee2mqtt    | 2022-05-08T09:54:16.995Z zigbee-herdsman:controller:endpoint Read 0x00124b002336dda0/1 genBasic(["modelId","manufacturerName"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt    | 2022-05-08T09:54:17.001Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x92bc EP:1 SeqNr. 78 request id: 11
zigbee2mqtt    | 2022-05-08T09:54:17.014Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 11 status: 0
zigbee2mqtt    | 2022-05-08T09:54:17.015Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt    | 2022-05-08T09:54:17.015Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
zigbee2mqtt    | 2022-05-08T09:54:17.016Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt    | 2022-05-08T09:54:17.018Z zigbee-herdsman:deconz:driver query aps data confirm
zigbee2mqtt    | 2022-05-08T09:54:17.019Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 79
zigbee2mqtt    | 2022-05-08T09:54:17.025Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x92bc request id: 11 confirm status: 0
zigbee2mqtt    | 2022-05-08T09:54:17.026Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt    | 2022-05-08T09:54:17.030Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 6
zigbee2mqtt    | 2022-05-08T09:54:17.030Z zigbee-herdsman:deconz:adapter true, true, false
zigbee2mqtt    | 2022-05-08T09:54:17.031Z zigbee-herdsman:deconz:adapter resolve request
......
zigbee2mqtt    | 2022-05-08T09:55:10.477Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt    | 2022-05-08T09:55:17.866Z zigbee-herdsman:controller:endpoint Read 0x00124b002336dda0/1 genBasic(["modelId","manufacturerName"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
zigbee2mqtt    | 2022-05-08T09:55:17.867Z zigbee-herdsman:controller:device:log Interview - TuYa read modelID and manufacturerName failed (Error: Read 0x00124b002336dda0/1 genBasic(["modelId","manufacturerName"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))
.......
zigbee2mqtt    | 2022-05-08T09:57:17.883Z zigbee-herdsman:controller:device:log Interview - active endpoints request failed for '0x00124b002336dda0', attempt 2
zigbee2mqtt    | 2022-05-08T09:57:17.884Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'EndDevice'
zigbee2mqtt    | 2022-05-08T09:57:17.887Z zigbee-herdsman:controller:device:log Interview - quirks did not match
zigbee2mqtt    | 2022-05-08T09:57:17.897Z zigbee-herdsman:controller:device:log Interview - failed for device '0x00124b002336dda0' with error 'Error: Interview failed because can not get active endpoints ('0x00124b002336dda0')
zigbee2mqtt    |     at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:592:19)
zigbee2mqtt    |     at runMicrotasks (<anonymous>)
zigbee2mqtt    |     at runNextTicks (node:internal/process/task_queues:61:5)
zigbee2mqtt    |     at processTimers (node:internal/timers:499:9)
zigbee2mqtt    |     at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:441:13)
zigbee2mqtt    |     at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:542:17)'
zigbee2mqtt    | 2022-05-08T09:57:17.903Z zigbee-herdsman:controller:error Interview failed for '0x00124b002336dda0 with error 'Error: Interview failed because can not get active endpoints ('0x00124b002336dda0')'
zigbee2mqtt    | Zigbee2MQTT:error 2022-05-08 11:57:17: Failed to interview '0x00124b002336dda0', device has not successfully been paired
zigbee2mqtt    | Zigbee2MQTT:info  2022-05-08 11:57:17: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002336dda0","ieee_address":"0x00124b002336dda0","status":"failed"},"type":"device_interview"}'
zigbee2mqtt    | Zigbee2MQTT:info  2022-05-08 11:57:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b002336dda0"},"type":"pairing"}'

```

h0mesmart, 2022-05-08:
OK I've managed to pair the device by downgrading Z2M to v 19.0. Now the issue is it's been recognized as a switch instead of cover. Here is the value in the database

``` 
{
    "id": 3,
    "type": "EndDevice",
    "ieeeAddr": "0x00124b002336dda0",
    "nwkAddr": 54829,
    "manufId": 0,
    "manufName": "LIVOLO\u0000\u0010TI0001  ",
    "powerSource": "Mains (single phase)",
    "modelId": "TI0001          ",
    "epList": [
        6
    ],
    "endpoints": {
        "6": {
            "profId": 260,
            "epId": 6,
            "devId": 0,
            "inClusterList": [
                0,
                3
            ],
            "outClusterList": [
                6
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TI0001          ",
                        "manufacturerName": "LIVOLO\u0000\u0010TI0001  ",
                        "powerSource": 1,
                        "zclVersion": 0,
                        "hwVersion": 0,
                        "dateCode": "20060831        "
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "hwVersion": 0,
    "dateCode": "20060831        ",
    "zclVersion": 0,
    "interviewCompleted": true,
    "meta": {
        "configured": 1
    },
    "lastSeen": 1652036170965,
    "defaultSendRequestWhen": "immediate"
}

```

Do I need to make a new converter to have it displayed as a cover ?  

Thaks for any help

Chippi, 2022-05-09:
Same issue, running same z2m version & conbee II.

IKEA E1603/E1702/E1708
https://www.zigbee2mqtt.io/devices/E1603_E1702_E1708.html#ikea-e1603%252Fe1702%252Fe1708

Interview failed, unsupported

Chippi, 2022-05-10:
Update, I managed to pair the device above after retrying a few times. I also moved my whole zigbee network to channel 25 cause I live in a pretty congested area. So far Z2M is running smoother than ever, seems to have solved most odd issues.

h0mesmart, 2022-05-14:
Unfortunately, I cannot change the channel, it's required to be on 26 to pair.

sztof, 2022-05-28:
i have similar problem with aqara sensors and conbee. After update I'm unable to pair them again

stskrp, 2022-05-31:
Same issue with Livolo TI0001-cover, firstly it is recognized as switch, after re-pairing, changes to cover, but it is accessible (online) only while in pairing mode, after pairing mode is ended, device goes offline and gets back online only if pairing mode enabled again

stskrp, 2022-05-31:
Later after pairing mode has ended, getting error: 
`Publish 'set' 'state' to '0x00124b00233ff31b' failed: 'Error: WriteStructured 0x00124b00233ff31b/6 genPowerCfg([{"attrId":0,"selector":null,"elementData":[4,0,0,0,0,0,0,0]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":true,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":3,"manufacturerCode":6866,"transactionSequenceNumber":233,"writeUndiv":true,"frameType":0}) failed (Data request failed with error: 'MAC transaction expired' (240))'`

closed 2022-07-15 by github-actions[bot]

stskrp, 2022-10-18:
Issue still in place for me

idelsink, 2024-06-19:
I had the same issue just now with my TRADFRI motion sensor [E1525/E1745](https://www.zigbee2mqtt.io/devices/E1525_E1745.html#ikea-e1525%252Fe1745). I'm not sure what caused it, but for some reason this started when the battery was empty. After replacing the batteries, it did not update its state anymore. So I thought it might be in some borked state.

I tried by pressing the button 4 times and it indeed left and rejoind but then I had the `Error: Interview failed because can not get active endpoints` error. I  fixed it by, 

1. removing the batteries of the device
2. Remove the device from zigbee2mqtt (with force: yes)
3. add the batteries again, and it just directly connected like nothing happened 😅

I think it might be related to the fact that that device had direct bindings with two lights. So I had to rebind those again after it was connected and all is running fine now