Koenkk/zigbee2mqtt #12424: TS0601 - 4 gang switch

RogerioVidigal, 2022-05-09:
### Link

R$ 96,24  Ganhe 35％ de desconto | Interruptor de parede de casa inteligente zigbee, interruptor de luz inteligente sem fio, 2 vias, controle por aplicativo, funciona com google home aleax, eua, ue, reino unido https://a.aliexpress.com/_m0nh1sA

### Database entry

":2,"type":"Router","ieeeAddr":"0x84fd27fffe80c59b","nwkAddr":2705,"manufId":4098,"manufName":"_TZE200_k6jhsr0q","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_k6jhsr0q","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd4ae04","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1652070512586,"defaultSendRequestWhen":"immediate"}

### Comments

I followed the guide to add new unsupported device, but when i reached the step 3, my results were different from the guide. i’m new in HA and understand almost nothing in codes, so, i couldn’t follow… i’ll apreciate if you could help me… I have 15 switchs like this one and can’t do nothing in my HA with them…

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_k6jhsr0q'
        },
    ],

    model: 'Interruptor 4 Vias',
    vendor: 'Xichen Sto',
    description: 'Interruptor 4 Vias',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

Koenkk, 2022-05-09:
Just added it: https://github.com/Koenkk/zigbee-herdsman-converters/commit/4f1de0a21add0db48bceb9d91fad0b5f18c5c12f

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RogerioVidigal, 2022-05-09:
Thank you so much… I’ll try the branch version tomorrow… 😉