Koenkk/zigbee2mqtt #12426: sendWhen 'active' only applied once

jaspernlp, 2022-05-09:
### What happened?

I was looking into the code how to have a request to a motion sensor (battery) device being processed when the device becomes active. I managed to get the call to the endpoint.command with {sendWhen:'active'}, the first call is with the value of 'active' the second call by the sendPendingRequest is with 'immediate'.

### What did you expect to happen?

I expected that the request would stay in the pending requests queue until the request would be successful.

### How to reproduce it (minimal and precise)

I added {sendWhen:'active'} in this call
https://github.com/Koenkk/zigbee-herdsman-converters/blob/ebc2e2451ce0ed6b2c7e422d1010883bbf203967/lib/tuya.js#L889
endpoints.command calls sendRequest
https://github.com/Koenkk/zigbee-herdsman/blob/cb8f2bb65ed1bdad62e762762ef5e546e5555291/src/controller/model/endpoint.ts#L772
sendRequest (https://github.com/Koenkk/zigbee-herdsman/blob/cb8f2bb65ed1bdad62e762762ef5e546e5555291/src/controller/model/endpoint.ts#L291) calls queueRequest which pushes the request on the queue. 
If the first call is not successful the request stayes on the queue.
sendPendingRequests (https://github.com/Koenkk/zigbee-herdsman/blob/cb8f2bb65ed1bdad62e762762ef5e546e5555291/src/controller/model/endpoint.ts#L267) is there to process the queue.
It calls the function that is pushed onto the queue (https://github.com/Koenkk/zigbee-herdsman/blob/cb8f2bb65ed1bdad62e762762ef5e546e5555291/src/controller/model/endpoint.ts#L272). The function is called without the sendWhen value as is defined in the request. 
During execution of the function the sendWhen is set to 'immediate' (because that is the defaultsendrequestWhen value).
With this configuration the requests won't stay on the queue forever (if the default sendWhen is 'active' then it will), but a single retry might be also insufficient.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-05-09:
Could you provide the herdsman debug logging of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

jaspernlp, 2022-05-09:
I spent an evening on this, starting with a fresh deployment of the zigbee2mqtt docker container in docker compose. I don't get the requests on the queue anymore.  Yesterdays' changes are deleted with this fresh deployment. Later this week I will try again to get the requests send at the time the sensor reports movement/light.

jaspernlp, 2022-05-10:
I figured out why the new deployment did not put requests sendWhen is active. Needed to remove the force of setting disableDefaultResponse (https://github.com/Koenkk/zigbee-herdsman-converters/blob/ebc2e2451ce0ed6b2c7e422d1010883bbf203967/lib/tuya.js#L889). If I don't disable then after 10 seconds an error is thrown on which the request is pushed to the queue of pending requests. 

Then it works nicely, the request that was pushed on the queue is send again when the sensor is detecting motion and is processed by the sensor.

I've added comments, instead of turning on the debug (that is so much). I'll try to get a nice log later this week.
Zigbee2MQTT:debug 2022-05-10 21:03:26: Received MQTT message on 'zigbee2mqtt/0xa4c1380278e6227b/set' with data '{"sensitivity":"high"}'
Zigbee2MQTT:debug 2022-05-10 21:03:26: Publishing 'set' 'sensitivity' to '0xa4c1380278e6227b'
Command 0xa4c1380278e6227b/1 manuSpecificTuya.dataRequest({"seq":2,"dpValues":[{"dp":9,"datatype":4,"data":[2]}]}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
**sendRequest sendWhen is active**
sendRequest caught error
queueRequest
sendPendingRequests
sendPendingRequests
Zigbee2MQTT:debug 2022-05-10 21:03:51: Received Zigbee message from 'lights/utilityRoom', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-05-10 21:03:51: Device 'lights/utilityRoom' requested OTA
sendRequest sendWhen is immediate
sendPendingRequests
Zigbee2MQTT:debug 2022-05-10 21:04:04: Received Zigbee message from '0xa4c1380278e6227b', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,130],"type":"Buffer"},"datatype":2,"dp":12}],"seq":11264}' from endpoint 1 with groupID 0
sendRequest sendWhen is immediate
Zigbee2MQTT:info  2022-05-10 21:04:04: MQTT publish: topic 'zigbee2mqtt/0xa4c1380278e6227b', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":80,"occupancy":false,"sensitivity":"low"}'
sendPendingRequests
{
  frame: ZclFrame {
    Header: {
      frameControl: [Object],
      transactionSequenceNumber: 92,
      manufacturerCode: null,
      commandIdentifier: 11
    },
    Payload: { cmdId: 0, statusCode: 0 },
    Cluster: {
      ID: 61184,
      attributes: {},
      manufacturerCode: undefined,
      name: 'manuSpecificTuya',
      commands: [Object],
      commandsResponse: [Object],
      getAttribute: [Function: getAttribute],
      hasAttribute: [Function: hasAttribute],
      getCommand: [Function: getCommand],
      getCommandResponse: [Function: getCommandResponse]
    },
    Command: { ID: 11, name: 'defaultRsp', parameters: [Array] }
  },
  address: 41505,
  endpoint: 1,
  linkquality: 80,
  groupID: 0,
  wasBroadcast: false,
  destinationEndpoint: 1
}
sendPendingRequests
sendPendingRequests
Zigbee2MQTT:debug 2022-05-10 21:04:04: Received Zigbee message from '0xa4c1380278e6227b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":9}],"seq":2}' from endpoint 1 with groupID 0
sendRequest sendWhen is immediate
Zigbee2MQTT:info  2022-05-10 21:04:04: MQTT publish: topic 'zigbee2mqtt/0xa4c1380278e6227b', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":80,"occupancy":false,"sensitivity":"high"}'
sendPendingRequests
Zigbee2MQTT:debug 2022-05-10 21:04:04: Received Zigbee message from '0xa4c1380278e6227b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":9}],"seq":2}' from endpoint 1 with groupID 0
sendRequest sendWhen is immediate
Zigbee2MQTT:info  2022-05-10 21:04:04: MQTT publish: topic 'zigbee2mqtt/0xa4c1380278e6227b', payload '{"battery":100,"illuminance":898,"keep_time":"10","linkquality":80,"occupancy":false,"sensitivity":"high"}'

jaspernlp, 2022-05-16:
I've managed to get around this. For the motion sensors I can send the request, getting queued and send at the time motion is reported. So the behavior now predictable. There is a bug, related to this issue and resolution. I will create an issue for that when I also have the logs and a possible resolution.

closed 2022-05-16 by jaspernlp