Koenkk/zigbee2mqtt #12432: Networkmap can no longer be created

habitoti, 2022-05-10:
### What happened?

When creating the network map, all goes well (according to debug log) until the final publishing to _zigbee2mqtt/bridge/response/networkmap_ takes place. It then fails with this error:

`
MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{},"error":"Cannot read property 'stack' of undefined","status":"error"}'
`

### What did you expect to happen?

Map should be published (worked before...)

### How to reproduce it (minimal and precise)

Initiate networkmap creation 

### Zigbee2MQTT version

1.25.1 

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

`
MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{},"error":"Cannot read property 'stack' of undefined","status":"error"}'
`

Homeassistant log:

```
2022-05-10 11:08:42 ERROR (MainThread) [homeassistant.util.logging] Exception in attributes_message_received when handling msg on 'zigbee2mqtt/bridge/response/networkmap': '{"data":{},"error":"Cannot read property 'stack' of undefined","status":"error"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 47, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 326, in attributes_message_received
    payload = attr_tpl(msg.payload)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/__init__.py", line 402, in async_render_with_possible_json_value
    return self._value_template.async_render_with_possible_json_value(
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 572, in async_render_with_possible_json_value
    return _render_with_context(
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1859, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.9/site-packages/jinja2/environment.py", line 1291, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.9/site-packages/jinja2/environment.py", line 926, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/local/lib/python3.9/site-packages/jinja2/filters.py", line 1688, in do_tojson
    return htmlsafe_json_dumps(value, dumps=dumps, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/jinja2/utils.py", line 658, in htmlsafe_json_dumps
    dumps(obj, **kwargs)
  File "/usr/local/lib/python3.9/json/__init__.py", line 234, in dumps
    return cls(
  File "/usr/local/lib/python3.9/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/lib/python3.9/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/lib/python3.9/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type LoggingUndefined is not JSON serializable

```

Koenkk, 2022-05-10:
Could you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

habitoti, 2022-05-10:
That one line was already from debug level logging. There‘s unfortunately nothing more. Here are some of the preceding activities while the map is being built:

```
Debug 2022-05-10 22:37:39Received Zigbee message from 'Klimadaten Außen', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1786}' from endpoint 1 with groupID null
Info 2022-05-10 22:37:39MQTT publish: topic 'zigbee2mqtt/Klimadaten Außen', payload '{"battery":99,"humidity":71,"last_seen":"2022-05-10T20:37:40.315Z","linkquality":255,"pressure":1003,"temperature":17.9,"voltage":2975}'
Debug 2022-05-10 22:37:40Received Zigbee message from 'Klimadaten Außen', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":7135}' from endpoint 1 with groupID null
Info 2022-05-10 22:37:40MQTT publish: topic 'zigbee2mqtt/Klimadaten Außen', payload '{"battery":99,"humidity":71,"last_seen":"2022-05-10T20:37:40.331Z","linkquality":232,"pressure":1003,"temperature":17.9,"voltage":2975}'
Debug 2022-05-10 22:37:40Retrieving state of 'Bodenfeuchtigkeit' after reconnect
Error 2022-05-10 22:37:48Failed to execute LQI for 'Stecklicht 1'
Info 2022-05-10 22:37:48MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{},"error":"Cannot read property 'stack' of undefined","status":"error"}'
```

Koenkk, 2022-05-11:
The logging is not included yet, try again within an hour, edge takes +- 1 hour to update after the commit.

habitoti, 2022-05-11:
Quick Update: HA Z2M Edge is still 1.25.1-dev commit: [147fe22] (complete uninstall and fresh install). Is there any way to enforce pulling the latest version?

Will try again tomorrow.

habitoti, 2022-05-12:
I am still getting the [May 8th commit](https://github.com/Koenkk/zigbee2mqtt/commit/147fe22f08edafa90edac594e30031a76e758dfc) only.
[image]

Are you sure, [920b492](https://github.com/Koenkk/zigbee2mqtt/commit/920b4924b0943e7e013726fedf0721f3c93e7736) has been made available for the HA Edge version already?

Koenkk, 2022-05-12:
Ah! The build was failing, fixed it now, new build should be up in an hour from now.

habitoti, 2022-05-13:
Here you are:

`
Info 2022-05-13 09:27:14MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{},"error":"Cannot read properties of undefined (reading 'stack')","status":"error"}'
Debug 2022-05-13 09:27:14TypeError: Cannot read properties of undefined (reading 'stack') at NetworkMap.networkScan (/app/lib/extension/networkMap.ts:248:36) at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.ts:63:34)
`

Koenkk, 2022-05-13:
Found the issue, fixed it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

habitoti, 2022-05-17:
Will for sure be in the next drop...

jocamane, 2022-05-17:
Ok, but what change was there from 2 days ago for it to have stopped working?
There is already a scheduled date for the next release

thanks

Koenkk, 2022-05-17:
The fix will be included in the 1 June release.

closed 2022-06-25 by github-actions[bot]