Koenkk/zigbee2mqtt #12434: Innr OPL 130 C Unsupported Attribute

addrum, 2022-05-10:
### What happened?

Publish 'get' 'color_temp_startup' to 'Garden Lights' failed: 'Error: Read 0x00158d0000f21bd7/1 lightingColorCtrl(["startUpColorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

### What did you expect to happen?

The effect either applies or isn't shown.

### How to reproduce it (minimal and precise)

Try to change `color_temp_startup` through the UI.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

Sonoff USB 3.0

addrum, 2022-05-23:
@Koenkk I assumed that was the case. This issue is for hiding that attribute from the UI as is seemingly the case with other devices I have that also don't support it.

Koenkk, 2022-05-23:
Done now, docs will be updated with the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-23 by Koenkk