Koenkk/zigbee2mqtt #12438: [New device support]:

RogerioVidigal, 2022-05-11:
### Link

R$ 96,24 Ganhe 35％ de desconto | Interruptor de parede de casa inteligente zigbee, interruptor de luz inteligente sem fio, 2 vias, controle por aplicativo, funciona com google home aleax, eua, ue, reino unido https://a.aliexpress.com/_m0nh1sA

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c13836ce672c84","nwkAddr":12739,"manufId":4417,"manufName":"_TZE200_di3tfv5b","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"O\u0011\u0000\u0000eO\u0011\u0000\u0000\u0012","65506":48,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_di3tfv5b","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1652228714456,"defaultSendRequestWhen":"immediate"}

### Comments

This is the same product as the previous request (and that you added), but it has a different Zigbee Manufacturer, so, please, could you include this one too??? Thank you so much...

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_di3tfv5b'
        },
    ],

    model: 'Interruptor 4 Vias',
    vendor: 'Xichen Sto',
    description: 'Interruptor 4 Vias',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

RogerioVidigal, 2022-05-29:
I've tried to use this External Converter including EXPOSES.
The switch is supported now, but when hit ON or OFF at the EXPOSES, it shows "No converter available for 'state_1' ("ON")" or "No converter available for 'state_1' ("OFF")" For each state it shows the same.
and if i use the switch, i receive this message "zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #1 from 0xa4c13836ce672c84 with raw data '{"dp":1,"datatype":1,"data":{"type":"Buffer","data":[1]}}': type='commandDataResponse', datatype='bool', value='true', known DP# usage: ["state","moesSsystemMode","tuyaSabCOalarm","moes105DimmerState1","trsPresenceState","trsfPresenceState","haozeeSystemMode","nousTemperature","wlsWaterLeak","AM02Control","garageDoorTrigger","connecteState","tshpsPresenceState"]" and the zigbee2mqtt stop to work and i need to wait a few seconds to start again.

This is my Converter code:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_di3tfv5b'
        },
    ],

    model: 'Interruptor 4 Vias',
    vendor: 'Xichen Sto',
    description: 'Interruptor 4 Vias',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().withEndpoint('1').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('2').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('3').setAccess('state', ea.STATE_SET),
            e.switch().withEndpoint('4').setAccess('state', ea.STATE_SET)
        // Here you should put all functionality that your device exposes
    ],
    
};

module.exports = definition;

closed 2022-07-06 by github-actions[bot]

brantmacga, 2023-01-16:
Did you ever solve this? I am facing the same issue on a Legrand device.