Koenkk/zigbee2mqtt #12439: Error: SREQ '--> ZDO - extRouteDisc - Zigbee2MQTT keeps crashing

Devon1984, 2022-05-11:
### What happened?

Since updating to 1.25.0 my Z2M container keeps crashing after a few hours with the following displayed in the logs.

Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":17327,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')
at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

If i restart the container it works fine again for another few hours and then crashes.
This has never happened with any other version previously.

I have now rolled back to 1.24.0 and my Z2M has been stable since the downgrade for weeks now.

Not sure if this makes a difference but im running this on my Unraid server version 6.9.2 with Linux kernal 5.10.28 and docker version 20.10.5

I do have a large Zigbee network with 77 Routers (IKEA lights) and 40 passive battery devices.

### How to reproduce it (minimal and precise)

after the update to 1.25.0 i expected it to be stable.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah! CC2652R

leroyloren, 2022-05-12:
I observe similar behavior, Z2M always drops + - at the same time (4am) then works for a while (6am) and in the morning I always have to restart, it continues to run all day.

There is nothing in the log, I only see "not connected to the mqtt server" messages.

Adapter)
zig-a-zig-ah! CC2652R

v)
1.25.1

FW)
Z-Stack_3.x.0_20220507,
before that
Z-Stack_3.x.0 20220219 (same issue)

Behaviour graph for idea
[image]

Devon1984, 2022-05-12:
Mine just crashed again now.
@Koenkk do you need me to enable the debug logs and upload them? I see some other people had the same issue in the past and you changed the handling of messages in Z2M of something and that seemed to fix it. This only started post 1.24.0

Koenkk, 2022-05-12:
@Devon1984 could you provide the herdsman debug log starting from the point where you start z2m until it fails? 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Devon1984, 2022-05-13:
@Koenkk Thank you so much for helping with this one. Hopefully this will help others too :-)
I have enabled the herdsman logging and its all running and working... Now we wait for the crash and ill upload the logs.

leroyloren, 2022-05-13:
@Koenkk I'm attaching a debug log all night when Z2M crashed several times.

downtime times
[image]

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8685331/log.txt)

Koenkk, 2022-05-13:
@leroyloren this is only the z2m debug logging, not the zigbee-herdsman one, could you provide that? Note that the zigbee-herdsman logging is logged to STDOUT/STDERR (not the log files!)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Devon1984, 2022-05-13:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8685558/log.txt)
@Koenkk my Z2M just crashed. Logs are attached.
I really hope you are able to find the cause of this one. Thanks again and if you need anymore logs or testing done please let me know :)

Devon1984, 2022-05-14:
@Koenkk Had another crash today so I captured those logs and attached here.
I think what ill do now is move to your dev branch of Z2M and capture another set of logs for you. If you don't need anymore logs just let me know otherwise ill keep capturing them for you.
Ill buy you a few coffees/beers for this :-) love your work! 
[log_14-05-22.txt](https://github.com/Koenkk/zigbee2mqtt/files/8692003/log_14-05-22.txt)

Koenkk, 2022-05-16:
@Devon1984 can you check if this issue still appears when you disable the availability feature?

Devon1984, 2022-05-16:
@Koenkk No problem. Ill revert back to the "latest" release and test with the Availability feature disabled.

Devon1984, 2022-05-17:
@Koenkk Just an update... Z2M 1.25.1 hasn't crashed yet with the "Availability Feature" dissabled. Normally by now it would have crashed at least once but lets see if the success continues.
Hopefully we can get the Availability option to work without crashing Z2M as its really useful with my IKEA lights and turning them off in Home assistant when the wall switch is turned off.

Koenkk, 2022-05-17:
@Devon1984 lets see if it keeps working for a few days.

Devon1984, 2022-05-19:
@Koenkk my Z2M has crashed again even with the "Availability Feature" disabled. Logs are attached.
Is there anything else that I can do on this side to help you troubleshoot the issue?
[log_19-05-22.zip](https://github.com/Koenkk/zigbee2mqtt/files/8730330/log_19-05-22.zip)

Koenkk, 2022-05-21:
I see a few issues in your log:

1. `0x847127fffebea814` is constantly sending state updates (not really an issue but this may reduce your network performance)
```
2022-05-18T22:16:53.278Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":86,"manufacturerCode":null,"commandIdentifier":66},"Payload":{"ctrlbits":0,"ontime":1800,"offwaittime":0},"Command":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"address":623,"endpoint":1,"linkquality":109,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-05-18T22:16:53.278Z zigbee-herdsman:controller:endpoint DefaultResponse 0x847127fffebea814/1 6(66, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-05-18T22:16:53.278Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x847127fffebea814:623/1 (0,0,2)
```
What device is this? 

2. MAC no ack errors
```
2022-05-18T22:29:29.562Z zigbee-herdsman:controller:endpoint Command 0xbc33acfffed654d1/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
Zigbee2MQTT:error 2022-05-19 06:29:29: Publish 'set' 'state' to 'Main Bathroom Light 2' failed: 'Error: Command 0xbc33acfffed654d1/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
```

Cannot be solved from z2m itself; this happens mostly due to interference (https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html#usb-based-adapter). Make sure to connect your dongle through a usb extension cable such that is not directly in range of other electrical equipment.

3. NWK_TABLE_FULL errors
```
2022-05-18T22:53:21.588Z zigbee-herdsman:controller:endpoint Command 0x804b50fffeef6ad5/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":25101,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))
Zigbee2MQTT:error 2022-05-19 06:53:21: Publish 'set' 'state' to 'Main Bedroom Light 6' failed: 'Error: Command 0x804b50fffeef6ad5/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":25101,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
```
Maybe this can be fixed by increasing the routing table sizes. Can you try with this fw?

[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8747181/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)

4. Then the error which lead to the z2m crash:
```
Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":623,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```
 
This error should be properly caught now (which prevents z2m from crashing)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Devon1984, 2022-05-21:
Hi @Koenkk,
Thank you so much for taking the time to go through my logs and help me with this issue I'm having! I REALLY do appreciate it as I know you are taking time out of your own day to do this work. To show my appreciation I will be sponsoring a one off half a hive to you as a thank you :-) 

To answer your Questions above...

1. This is strange. That sensor is an IKEA motion sensor that's in my dining room. I have about 15 of these sensors through the house. Ill try and reset the sensor, maybe also take out the battery for a while and try it again and see if i get the same issues in the log. If it keeps happening ill replace the sensor. Would you have any suggestions or is this the right approach?

2. My coordinator is connected to my Unraid Server with a 2m long USB extension cable and I have separated my zigbee channels form my wifi channel. My Zigbee channel is 11 and all of my wifi channels are channel 11,12 and 13 so i dont think that could be the cause. This device that has the Mac Ack errors are my Ikea downlights in the bathroom so perhaps there is local interference at the lights themselves. These lights are powered off at the wall a lot so could this cause issues perhaps? If you have any other suggestions im keen to hear as id like to make my network as solid as possible :-)

3. Great, thanks ill definetly give this firmware a try. i have a Electrolama zig-a-zig-ah! CC2652R Adapter, Will the CC26X2R1 fw be ok in mine? Just want to double check.

4. AWESOME!! Thank you so very much!! you are a legend @Koenkk  :-) Im switching to the dev branch now.

Ill continue to capture the Herdsman logs for a bit and will check on the points that you brought to my attention and see if i can at least solve points 1 and 2 and also let you know about the other points and if the fw solved the NWK_TABLE_FULL errors.

Is there anything else that I can provide you with or do you want anything more from my side?

Koenkk, 2022-05-21:
1. I see, if it is a motion sensor then it is fine (I thought it was a switch)
2. So there can be many reasons for this, thick walls, WiFi from neighbours, bluetooth devices, etc. Given that you've configured your wifi correctly + use a USB extension cable I wouldn't worry about too much if normally everything is working fine.
3. yes this one will work on the Electrolama zig-a-zig-ah! CC2652R Adapter
4. 👍 

> Is there anything else that I can provide you with or do you want anything more from my side?
No, lets see if the new firmware + zigbee-herdsman fix fixes your problems

Devon1984, 2022-05-21:
@Koenkk perfect! I'll get the new fw installed tonight and report back in a day or 2 then :-)

Devon1984, 2022-05-23:
Hi @Koenkk,
I have attached my latest log for you to have a quick look at if you have some time :) but Everything seems to be running a lot smoother with no crashes so far which is awesome. Ill keep monitoring for any crashes for the rest of the week as an extended test case. I am running the latest dev of Z2M which gets updated daily to all your pushes.

All the "NWK_TABLE_FULL" errors have been significantly reduced so it looks like the firmware mod that you did definetly did help a lot. There are still some NWK_TABLE_FULL errors being reported so not sure if its worth increasing the routing table sizes just a little bit more perhaps but im interested to see what you think.

I was also wondering, will you be incorporating the firmware mod that you did for me into the next release of the CC2652R_coordinator.hex firmware? Just wondering if when the new fw version is released whether or not i can upgrade to get back into the official released firmwares that you provide.

Thanks again for all you time and help with this.

[log_23-05-22.txt](https://github.com/Koenkk/zigbee2mqtt/files/8753137/log_23-05-22.txt)

Devon1984, 2022-05-24:
@Koenkk Thanks for the information and sharing the link. I see that TI have responded so lets see what their software development team come back with :-)

Devon1984, 2022-05-25:
Excellent, thanks for this Koen!
As soon as i get home today ill load the new fw and report back with logs. Fingers crossed this further reduces the NWK_TABLE_FULL errors. :-)

Devon1984, 2022-05-26:
@Koenkk Update on the new fw mod...
The additional routing table increase (150 to 255) has further decreased the NWK_TABLE_FULL errors by about half compared to the first firmware mod that you provided me so it definitely looks like we are on the right track :-)

Ill grab you the logs but i wanted to get a bigger sample size so ill upload it tomorrow afternoon (Perth, Australia) time but its looking good.

I'm still hoping that Ryan from TI gets back to us with a fix so we can continue with the tweaks and testing.

Koenkk, 2022-05-26:
> I'm still hoping that Ryan from TI gets back to us with a fix so we can continue with the tweaks and testing.

This can take quite some time, they release every quarter (last one was in april) and I'm not sure wether it will be fixed in the next SDK.

After https://github.com/Koenkk/zigbee2mqtt/issues/11374 is fixed I will publish the new firmwares

Devon1984, 2022-05-27:
No worries Koen. I understand that these things can take time to fix and get released and we will just have to wait until TI fixes it sometime this year.
On a positive note though, the fw changes that you have done have been a HUGE improvement from what it was before.

Once #11374 is fixed, will the fw changes that you have done for #12439 be included in the new official fw release? or are you planning on waiting for the "NWK_TABLE_FULL" issue to be completely fixed first before including the fix in the official fw?

Regarding the issue of Z2M crashing, im happy that this has now been fixed. Its been stable for about 1 week now which before your fix it was crashing almost daily. How would you prefer me to Close out the Z2M issue because we have 2 issues under 1 issue # ?
If there are no objections i was thinking of closing out #12439 and raising a new issue for the "NWK_TABLE_FULL" error with a link in the description to this thread for history and we can continue under the new issue # .

Devon1984, 2022-05-27:
@Koenkk For completeness, I have attached the log for the new fw you gave me. Still very good with only 4 instances of the "NWK_TABLE_FULL" error occurring over a 14 hour period.
[log_27-05-22.zip](https://github.com/Koenkk/zigbee2mqtt/files/8784984/log_27-05-22.zip)

Koenkk, 2022-05-27:
> On a positive note though, the fw changes that you have done have been a HUGE improvement from what it was before.

Besides less errors, do you see other improvements?

> Once https://github.com/Koenkk/zigbee2mqtt/issues/11374 is fixed, will the fw changes that you have done for https://github.com/Koenkk/zigbee2mqtt/issues/12439 be included in the new official fw release?

Yes I will release a new fw after this

> How would you prefer me to Close out the Z2M issue because we have 2 issues under 1 issue # ?

This one can be closed, I created a new one for the routing table sizes: https://github.com/Koenkk/Z-Stack-firmware/issues/375

closed 2022-05-27 by Koenkk

Devon1984, 2022-05-28:
> Besides less errors, do you see other improvements?

@Koenkk Yes, there is a huge improvement in the responsiveness of all my devices.

Before this change it was very hit and miss whether my lights would turn on when my motion sensors activate, now pretty much every time my lights turn on when motion is detected which is Great :-) Even my door sensors are working correctly now and not needing to be activated a few times before reporting correctly.

Koenkk, 2022-05-28:
@Devon1984 good to hear, and thanks for the donation, will ping you when a firmware with even greater routing tables is available.

Devon1984, 2022-05-28:
@Koenkk No worries at all :-) I wanted to give something as a token of my appreciation for all the hard work you put into maintaining this project, both the Z2M and coordinator firmware's. Also thanks so much again for helping me with these issues I have been having!

Yes, please ping me once TI have come back to you with an updated SDK and we are able to do some further tests :-)

timdonovanuk, 2022-12-10:
This is the only thread I can find that also mentions "extRouteDisc". My zigbee network was fine for a year but now has repeated issues like slow motion sensors, laggy light commands, and the occasional complete unavailability of all devices until I unplug my adapter or restart z2m.

Any idea what the source of `SRSP - ZDO - extRouteDisc after 6000ms` is?