Koenkk/zigbee2mqtt #12440: Can't upgrade to 1.25.1 from 1.22.0 docker container

fraddygil, 2022-05-11:
### What happened?

can't upgrade to 1.25.1 from 1.22.0 docker container

[image]
[image]

### What did you expect to happen?

When I upgrade the image I get the following message.

> Using '/app/data' as data directory
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbecd662c

### How to reproduce it (minimal and precise)

tengo imagen docker 1.22.0 y quiero actualizar hacia la 1.25.1 o latest y me devuelve este error. 

**Error no ocurre si tengo la version 1.22.0**

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Shred99, 2022-05-11:
Sounds a lot like this: #7662

Shred99, 2022-05-11:
[https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users](url)

fraddygil, 2022-05-11:
Thanks for your help @Shred99 , I managed to solve with the last link you sent me.

closed 2022-05-11 by fraddygil