Koenkk/zigbee2mqtt #12442: UseeLink SM-0306E-2W - Linked switches issue

thenexus7, 2022-05-11:
### What happened?

Device: UseeLink SM-0306E-2W
Device type: 4 gang switch, with USB (5 switches total)
Problem: When you switch on any of the 5 single switches, it switches on all 5 together

### What did you expect to happen?

When you switch "on" the endpoint 1 it only power on endpoint 1
When you switch "on" the endpoint 2 it only power on endpoint 2 
... and so on

### How to reproduce it (minimal and precise)

Just switch on any of the 5 switches and see the other 4 turning on, one by one... Same occurs when you turn off, they're all turning off at the same time. 

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

0x266b0700

### Adapter

ConBee2/RaspBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8668277/log.txt)

Koenkk, 2022-05-11:
Should be fixed now. (might require repairing)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

DVillagraG, 2022-05-11:
can I just replace useelink.js in my installation without going to the dev branch?

thenexus7, 2022-05-12:
I just switched to dev (edge) ... restarted, reconfigured the device, but the problem persist :( sorry ... 

No changes on log level, same thing:  

```
Info 2022-05-12 09:59:39MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"ON"}'
Info 2022-05-12 09:59:39MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"ON","state_l5":"ON"}'
Info 2022-05-12 09:59:39MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"OFF","state_l2":"OFF","state_l3":"ON","state_l4":"ON","state_l5":"ON"}'
Info 2022-05-12 09:59:39MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"ON","state_l5":"ON"}'
Info 2022-05-12 09:59:39MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"ON"}'
Info 2022-05-12 09:59:40MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}'
Info 2022-05-12 09:59:43MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}'
Info 2022-05-12 09:59:43MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}'
Info 2022-05-12 09:59:43MQTT publish: topic 'zigbee2mqtt/buanderie_multi', payload '{"linkquality":255,"state_l1":"ON","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}'
Info 2022-05-12 09:59:43MQTT publish: topic 'zigbee2mqtt/buanderie_firealarm', payload
```

Did I missed something?

thenexus7, 2022-05-13:
> @thenexus7 the edge build failed; fixed that now so try to uninstall and install the addon in an hour from now.

Morning @Koenkk , I'm delighted to announce you resolved the bug! Latest Edge release works perfectly now! A HUGE thanks for your work and implication on this project! I'm not coder, but through my work have access to a ton of hardware, if one day you need testers for something, I owe you one :) 

Thanks again and in advance, let me wish you a delightful weekend

closed 2022-05-13 by thenexus7