Koenkk/zigbee2mqtt #12444: DJT11LM adjustable vibration timeout vs fixed 60 sec lockout

JonSilver, 2022-05-11:
### What happened?

I've been trying to make sense of the Xiaomi Aqara DJT11LM vibration sensor, so I ran some structured experiments.
Somewhere behind the scenes, there appears to be a fixed 60 second lockout after a vibration detection event, during which time any amount of movement or vibration caused no extension to the period of `vibration = true` detection even though other properties (`angle` etc) are updated and sent.

### What did you expect to happen?

I expected that setting the timeout to a short period (e.g. 10 secs) would result in further detections upon timeout expiry. However all vibration events were ignored.

Setting the timeout to a longer period (e.g. 90 secs) resulted in further vibration events received during the first 60 secs being ignored, whereas any events after the first 60 secs (i.e. in the last 30 secs of the timeout period) extended the period of `vibration = true`.

If the 60 sec lockout is a driver issue, please could I request that someone takes a look at the code, identifies what's causing the lockout and makes the lockout configurable?

If it's a sensor issue, please could I request that this "feature" be added to the documentation?

Many thanks,
Jon

### How to reproduce it (minimal and precise)

### Establish test framework:

- DJT11LM sensor, sensitivity set to High
- client for observing sensor state
- timer

### Tests:

1. set vibration timeout to 90 secs
2. jog or tilt the sensor; simultaneously start the timer; observe `vibration = true`
3. during the first 60 secs, jog or tilt the sensor again; observe other properties (angle etc) changing
4. at the end of the 90 secs, observe `vibration = false`

5. jog or tilt the sensor; simultaneously start the timer; observe `vibration = true`
6. during the last 30 secs of the 90 sec period, jog or tilt the sensor again; observe other properties (`angle` etc) changing
7. at the end of the 90 secs, observe that `vibration = true`
8. `vibration = false` occurs at some time well after 90 secs has elapsed (time of last extension event + 90 secs)

### Conclusion:
there is a 60 sec fixed period of lockout after vibration is detected, during which further jog/tilt events are ignored; once this period has passed, within the variable (90 sec in this case) timeout, further jog/tilt events will extend the period of detection. However, jog/tilt events are clearly detected by the sensor and reported even during the lockout period.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20200417

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

lux73, 2022-05-13:
https://www.zigbee2mqtt.io/devices/DJT11LM.html#options

its adjustable

JonSilver, 2022-05-13:
> its adjustable

No, it's not.

The _timeout_ is adjustable/variable.

The _lockout_ is not. It is fixed at 60 secs. It is undocumented. It is at odds with the timeout. It makes the sensor timeout behave strangely.

I have documented all this in reproducible detail above, obtained using experimental observation.

closed 2022-06-25 by github-actions[bot]

JonSilver, 2022-06-25:
This is not completed and therefore should not be closed.

distante, 2023-02-12:
This is closed due inactivity but the issue still exists as per February 2023