Koenkk/zigbee2mqtt #12446: [New device support]: VS180

sebastien-rouen, 2022-05-11:
### Link

https://fr.aliexpress.com/item/1005002490419821.html?spm=a2g0o.order_list.0.0.21ef5e5bs2xN2D&gatewayAdapt=glo2fra

### Database entry

{"id":23,"type":"EndDevice","ieeeAddr":"0xb4e3f9fffeb457b6","nwkAddr":52155,"manufId":4098,"manufName":"_TYZB01_kulduhbj","powerSource":"Battery","modelId":"TS0210","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,10,1,1280],"outClusterList":[25],"clusters":{"ssIasZone":{"attributes":{"currentZoneSensitivityLevel":2}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"genBasic":{"attributes":{"65503":"��\u000e*\u0013","modelId":"TS0210","manufacturerName":"_TYZB01_kulduhbj","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1652291710682,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add with an external converter (like "TS0210.js", with some parameters, but I don't know to use correctly with exposes "sensitivity".
Thanks for your help.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VS180'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'VS180', // Vendor model number, look on the device for a model number
    vendor: 'TuYa', // Vendor of the device (only used for documentation and startup logging)
    description: 'Vibration sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [
        e.battery(),
        exposes.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('Vibration sensor sensitivity'),
        e.vibration(),
        e.linkquality(),
    ],
};

module.exports = definition;
```

waynen1967, 2022-06-06:
I am an extreme novice, however I did manage to get mine working somehow. I made the top of my Zigbee2MQTT's configuration.yaml say the following in Studio Code Editor between the *'s. In my case the first line was already in the file

******************************************************
external_converters:
  -  ts0210.js
*****************************************************
Then I created the ts0210.js file and put this in it
____________________________________________________________________________________________________________________________________
****************************************************
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0210'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'TS0210', // Vendor model number, look on the device for a model number
    vendor: 'TuYa', // Vendor of the device (only used for documentation and startup logging)
    description: 'TuYa Vibration sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.battery, fz.ias_vibration_alarm_1_with_timeout],
        toZigbee: [tz.TS0210_sensitivity],
        exposes: [e.battery(), e.vibration(), exposes.enum('sensitivity', exposes.access.STATE_SET, ['low', 'medium', 'high'])],
};

module.exports = definition;

********************************************************
____________________________________________________________________________________________________________________________________
Whether needed or not I don't know, but I rebooted everything and everything works now. I then added a second sensor and it was kind of slow to show up with all working but it did all work too. I have had some luck with pairing things a second time without removing them. Hope this helps someone

closed 2022-07-15 by github-actions[bot]

donster22, 2022-10-04:
> * ts0210.js

amazing. this sorted me out with the samotech vibration sensors which are cheap as chips. thanks very much.