Koenkk/zigbee2mqtt #12452: [Problem]: Aqara SSM-U01 not reporting power after update from version 1.23

inode64, 2022-05-12:
### What happened?

I have the same problem, after updating to any version higher than 1.23 the power is not shown
I have tried even with the latest version 1.25.1
But there is no way for it to work, I have erased the device and rejoined it but none of this has worked

Likewise, the "energy" and "voltage" parameters have never shown any value, always null, have these 2 parameters really ever worked?

### What did you expect to happen?

Show power consumption

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

Zigbee2MQTT:debug 2022-05-12 08:13:39: Loaded state from file /opt/zig/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-05-12 08:13:39: Logging to console and directory: '/var/log/zigbee2mqtt' filename: log.txt
Zigbee2MQTT:info  2022-05-12 08:13:39: Starting Zigbee2MQTT version 1.25.1 (commit #3f6a1378)
Zigbee2MQTT:info  2022-05-12 08:13:39: Starting zigbee-herdsman (0.14.27)
Zigbee2MQTT:debug 2022-05-12 08:13:39: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zig/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zig/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zig/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'DeconzAdapter' +0ms
  zigbee-herdsman:deconz:driver Set delay to 0 +0ms
CREATED DECONZ ADAPTER
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[19,70,162,177,83,152,1,119,175,125,115,55,155,237,19,96],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zig/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zig/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zig/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0 +1ms
  zigbee-herdsman:deconz:driver Serialport opened +3ms
  zigbee-herdsman:controller:log Started with result 'resumed' +4ms
  zigbee-herdsman:controller:log Injected database: true, adapter: true +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1 +3ms
  zigbee-herdsman:deconz:frameParser MAC: 0x00212effff0735fe +0ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7 +5ms
  zigbee-herdsman:deconz:frameParser NWKADDR: 0 +4ms
Zigbee2MQTT:info  2022-05-12 08:13:39: zigbee-herdsman started (resumed)
  zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 3 +5ms
  zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,88,38 +4ms
Zigbee2MQTT:info  2022-05-12 08:13:39: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 4 paramId: 5 +5ms
  zigbee-herdsman:deconz:frameParser PANID: 1a62 +6ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 5 paramId: 11 +6ms
  zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd +6ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 28 +5ms
  zigbee-herdsman:deconz:frameParser CHANNEL: 11 +5ms
  zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 7 paramId: 24 +5ms
  zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden +5ms
Zigbee2MQTT:debug 2022-05-12 08:13:39: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2022-05-12 08:13:39: Currently 1 devices are joined:
Zigbee2MQTT:info  2022-05-12 08:13:39: 0x04cf8cdf3c8e8762 (0x04cf8cdf3c8e8762): SSM-U01 - Xiaomi Aqara single switch module T1 (with neutral) (Router)
Zigbee2MQTT:warn  2022-05-12 08:13:39: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-05-12 08:13:39: Allowing new devices to join.
Zigbee2MQTT:warn  2022-05-12 08:13:39: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-05-12 08:13:39: Zigbee: allowing new devices to join.
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 8 request id: 1 +7ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 1 status: 0 +7ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +55ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +55ms
  zigbee-herdsman:deconz:driver query aps data confirm +5ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 9 +5ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 1 confirm status: 0 +12ms
[... 210 lines cut ...]
  zigbee-herdsman:controller:log Disable joining +4s
  zigbee-herdsman:deconz:adapter zclFrame to all - zclFrame.payload: +4s
  zigbee-herdsman:deconz:adapter { options: 10, commisioningWindow: 0 } +0ms
  zigbee-herdsman:deconz:adapter sendZclFrameToAll - message send +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 36 request id: 8 +2s
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 8 status: 0 +2s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +0ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +0ms
  zigbee-herdsman:deconz:driver query aps data confirm +2ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 37 +6ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 8 confirm status: 0 +9ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 38 request id: 9 +4ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 9 status: 0 +5ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +75ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +75ms
  zigbee-herdsman:deconz:driver query aps data confirm +5ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 39 +5ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 9 confirm status: 0 +11ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:adapter PERMIT_JOIN - 0 seconds +103ms
Zigbee2MQTT:info  2022-05-12 08:14:50: Stopped zigbee-herdsman
Zigbee2MQTT:info  2022-05-12 08:14:50: Stopped Zigbee2MQTT

Koenkk, 2022-05-12:
Try upgrading your conbee firmware to [0x26720700](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF); after this repair the Xiaomi device.

inode64, 2022-05-13:
thank you very much, now the power works

But still nothing is displayed in energy and current
And I have seen in the log some keys 8,9 and 154 that are not known

Thanks!!!
[debug_zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/8685433/debug_zigbee2mqtt.txt)

Koenkk, 2022-05-13:
@Otnow could you look into this?

inode64, 2022-05-13:
yes, but.
How?
There is guide or documentation

Otnow, 2022-05-13:
> debug_zigbee2mqtt.txt

Judging by the log and the converter, the switch quickly reports only power, and the remaining values are contained in the heartbeat-message, which the switch reports with a certain frequency of several minutes (possibly more).

@inode64, try to write a log from the moment the switch is turned on until the moment other values appear in addition to power.

inode64, 2022-05-14:
Analyzing the logs, power_outage_count always starts from 1, it should not be changed to subtract 1 and thus reflect the number of real power outages

--- zigbee-herdsman-converters/lib/xiaomi.js.old       2022-05-13 09:59:05.968693248 +0200
+++ zigbee-herdsman-converters/lib/xiaomi.js   2022-05-14 08:59:46.462776215 +0200
@@ -169,7 +169,7 @@
             break;
         case '5':
             if (['Mains (single phase)', 'DC Source'].includes(meta.device.powerSource)) {
-                payload.power_outage_count = value;
+                payload.power_outage_count = value - 1;
             }
             break;
         case '10':

debug 2022-05-14 08:43:33: lumi.switch.n0agl1: Processed buffer into data {"3":22,"5":1,"8":279,"9":256,"11":0,"100":0,"149":0,"150":2342.940673828125,"151":0,"152":0,"154":0}
debug 2022-05-14 08:43:33: lumi.switch.n0agl1: unknown key 8 with value 279
debug 2022-05-14 08:43:33: lumi.switch.n0agl1: unknown key 9 with value 256
debug 2022-05-14 08:43:33: lumi.switch.n0agl1: unknown key 154 with value 0

debug 2022-05-14 08:44:45: lumi.switch.n0agl1: Processed buffer into data {"3":22,"5":2,"8":279,"9":256,"11":0,"100":0,"149":0,"150":2342.132568359375,"151":0,"152":0,"154":0}
debug 2022-05-14 08:44:45: lumi.switch.n0agl1: unknown key 8 with value 279
debug 2022-05-14 08:44:45: lumi.switch.n0agl1: unknown key 9 with value 256
debug 2022-05-14 08:44:45: lumi.switch.n0agl1: unknown key 154 with value 0

debug 2022-05-14 08:54:54: lumi.switch.n0agl1: Processed buffer into data {"3":30,"5":3,"8":279,"9":256,"11":0,"100":0,"149":0,"150":2353.49658203125,"151":0,"152":0,"154":16}
debug 2022-05-14 08:54:54: lumi.switch.n0agl1: unknown key 8 with value 279
debug 2022-05-14 08:54:54: lumi.switch.n0agl1: unknown key 9 with value 256
debug 2022-05-14 08:54:54: lumi.switch.n0agl1: unknown key 154 with value 16
debug 2022-05-14 08:54:54: lumi.switch.n0agl1: Processed data into payload {"temperature":30,"power_outage_count":3,"illuminance":0,"illuminance_lux":0,"state":"OFF","energy":0,"consumption":0,"voltage":235.3,"current":0,"power":0}
debug 2022-05-14 08:54:54: lumi.switch.n0agl1: Processed data into payload {"temperature":30,"power_outage_count":3,"illuminance":0,"illuminance_lux":0,"state":"OFF","energy":0,"consumption":0,"voltage":235.3,"current":0,"power":0}
info  2022-05-14 08:54:54: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8e8762', payload '{"consumption":0,"current":0,"device_temperature":28,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":3,"power_outage_memory":null,"state":"OFF","switch_type":null,"temperature":30,"voltage":235.3}'
debug 2022-05-14 08:54:58: Received Zigbee message from '0x04cf8cdf3c8e8762', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"Dec 30 2021"}' from endpoint 1 with groupID null
debug 2022-05-14 08:54:59: Received Zigbee message from '0x04cf8cdf3c8e8762', type 'read', cluster 'genTime', data '["time","timeZone"]' from endpoint 1 with groupID null
debug 2022-05-14 08:55:00: Received Zigbee message from '0x04cf8cdf3c8e8762', type 'attributeReport', cluster 'genBasic', data '{"appVersion":23}' from endpoint 1 with groupID null
debug 2022-05-14 08:55:02: Received Zigbee message from '0x04cf8cdf3c8e8762', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,3,40,29,152,57,0,0,0,0,149,57,0,0,0,0,150,57,46,39,19,69,151,57,0,0,0,0,5,33,3,0,154,32,0,11,32,0,8,33,23,1,9,33,0,1],"type":"Buffer"}}' from endpoint 1 with groupID null
debug 2022-05-14 08:55:02: lumi.switch.n0agl1: Processed buffer into data {"3":29,"5":3,"8":279,"9":256,"11":0,"100":0,"149":0,"150":2354.44873046875,"151":0,"152":0,"154":0}
debug 2022-05-14 08:55:02: lumi.switch.n0agl1: unknown key 8 with value 279
debug 2022-05-14 08:55:02: lumi.switch.n0agl1: unknown key 9 with value 256
debug 2022-05-14 08:55:02: lumi.switch.n0agl1: unknown key 154 with value 0

Otnow, 2022-05-14:
> payload.power_outage_count = value - 1;

It seems to be more logical.

***

I still don’t understand, did you manage to wait after turning on the switch for other values (energy, voltage, current) in addition to power?

inode64, 2022-05-15:
Yes, all the values ​​are working, I have put a hot air fan and the values ​​have risen
Regarding the unknown values ​​8, 9 and 154, only 154 I have seen that when the device is started it is at value 16 and after a few seconds it goes to value 0, this may indicate that the device is starting and not that values ​​are being passed to it , but it is an assumption.
Attached log.

But well this would be to deal with it in another issue, we close task.
Thanks for all

<pre>
debug 2022-05-15 17:08:26: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genBasic', data '{"appVersion":23}' from endpoint 1 with groupID null
debug 2022-05-15 17:08:28: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,3,40,39,152,57,0,0,0,0,149,57,0,0,0,0,150,57,37,245,19,69,151,57,0,0,0,0,5,33,1,0,154,32,0,11,32,0,8,33,23,1,9,33,0,1],"type":"Buffer"}}' from endpoint 1 with groupID null
debug 2022-05-15 17:08:28: lumi.switch.n0agl1: Processed buffer into data {"3":39,"5":1,"8":279,"9":256,"11":0,"100":0,"149":0,"150":2367.321533203125,"151":0,"152":0,"154":0}
debug 2022-05-15 17:08:28: lumi.switch.n0agl1: Processed data into payload {"temperature":39,"power_outage_count":0,"unknown8":279,"unknown9":256,"illuminance":0,"illuminance_lux":0,"state":"OFF","energy":0,"consumption":0,"voltage":236.7,"current":0,"power":0,"unknown154":0}
debug 2022-05-15 17:08:28: lumi.switch.n0agl1: Processed data into payload {"temperature":39,"power_outage_count":0,"unknown8":279,"unknown9":256,"illuminance":0,"illuminance_lux":0,"state":"OFF","energy":0,"consumption":0,"voltage":236.7,"current":0,"power":0,"unknown154":0}
info  2022-05-15 17:08:28: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.7}'
debug 2022-05-15 17:08:29: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genOnOff', data '{"245":33554432,"onOff":0}' from endpoint 1 with groupID null
info  2022-05-15 17:08:29: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.7}'
debug 2022-05-15 17:08:46: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
info  2022-05-15 17:08:46: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.7}'
debug 2022-05-15 17:08:47: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genOnOff', data '{"245":56859392,"onOff":1}' from endpoint 1 with groupID null
info  2022-05-15 17:08:47: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.7}'
debug 2022-05-15 17:08:56: Saving state to file /var/lib/zigbee2mqtt/state.json
debug 2022-05-15 17:09:12: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genOnOff', data '{"245":56859392,"onOff":0}' from endpoint 1 with groupID null
info  2022-05-15 17:09:12: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.7}'
debug 2022-05-15 17:09:12: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2022-05-15 17:09:12: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.7}'
debug 2022-05-15 17:09:12: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 21 with groupID null
info  2022-05-15 17:09:12: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null [... cut ...]
debug 2022-05-15 17:11:56: Received Zigbee message from '0x54ef441000191d8c', type 'attributeReport', cluster 'genOnOff', data '{"245":56859392,"onOff":1}' from endpoint 1 with groupID null
info  2022-05-15 17:11:56: MQTT publish: topic 'zigbee2mqtt/0x54ef441000191d8c', payload '{"consumption":0,"current":0,"device_temperature":null,"energy":0,"illuminance":0,"illuminance_lux":0,"linkquality":255,"power":0,"power_outage_count":1,"power_outage_memory":null,"state":"ON","switch_type":null,"temperature":39,"unknown154":0,"unknown8":279,"unknown9":256,"voltage":236.2}'
</pre>

closed 2022-05-15 by inode64