Koenkk/zigbee2mqtt #12456: GL-D-007P  Gledopto is not recognized

xking3, 2022-05-12:
### What happened?

Gledopto GL-D-007P is not recognize. 

### What did you expect to happen?

It is in the device list, it should be recognized. 

### Zigbee2MQTT version

1.2.1

### Adapter firmware version

10551203

### Adapter

sonsoff 3.0

### Debug log

Warning Received message from unsupported device with Zigbee model 'GL-D-007P' and manufacturer name 'GLEDOPTO'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2022-05-12:
This device is supported in the latest dev.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-05-12 by Koenkk

xking3, 2022-05-12:
Thank you very much! Great work!