Koenkk/zigbee2mqtt #12457: [New device support]: Paulmann 501.34 Smart Switch 4 Buttons white

GerhardLang, 2022-05-12:
### Link

https://de.paulmann.com/p/wandschalter-smart-home-zigbee-on-off-dimm-weiss/50134

### Database entry

could not find the database.db but see below

### Comments

The Paulmann 501.34 works with all buttons with the converter below (I am using HA with zigbee2mqtt and this button):

@Koenkk maybe you can include in the next release if you like to check the code for obsolete parts, I am not a programmer :-)

Thank you.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['501.34'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '501.34', // Vendor model number, look on the device for a model number
    vendor: 'Paulmann', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart Switch 4 button white', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fz.command_off, fz.battery], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.action(['on_1', 'off_1','on_2', 'off_2', ])], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
	meta: {multiEndpoint: true},
    };

module.exports = definition;
```

Koenkk, 2022-05-12:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) (and next release on 1 June)

closed 2022-05-12 by Koenkk

IngoK1, 2023-01-25:
> Is there a way to support longpress?

Anyone found a solution to this? I can not get the actions "brightness_move_up_1, brightness_move_down_1, brightness_move_stop_1, brightness_move_up_2, brightness_move_down_2, brightness_move_stop_2" to be transferred. 
Anyone an idea how to get this to work?

Ingo

m-wo, 2023-02-20:
Long-press support would be awesome!

IngoK1, 2023-02-20:
This has been adressed and fixed in another issue. Please see #16348 Have fun!
Ingo

IngoK1, 2023-02-20:
This has been addressed in another issue #16348 . Have fun
Ingo